Koenkk/zigbee2mqtt #25454: Migration 1.42.0 to 2.0.0 - AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value

thierryprats, 2025-01-04:
### What happened?

Migration from 1.42.0 to 2.0.0 return this error :

```bash
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:

  (0, node_assert_1.default)(result instanceof Object)

    at read (/app/lib/util/yaml.ts:24:15)
    at Object.readIfExists (/app/lib/util/yaml.ts:36:34)
    at readDevicesOrGroups (/app/lib/util/settings.ts:317:38)
    at read (/app/lib/util/settings.ts:325:5)
    at getPersistedSettings (/app/lib/util/settings.ts:393:21)
    at Object.reRead (/app/lib/util/settings.ts:640:5)
    at start (/app/index.js:130:14)
```

zigbee2mqtt runs on docker engine with SMLIGHT SLZB-06 CC2652P Ethernet mode.

About the adapter :
|| Radio | Core |
|:-|:-|:-|
| CPU model | CC2652P | ESP32-D0WD-V3 |
| Firmware version | 20240710 | v2.3.6 |

Here is my `configuration.yaml` :

```yaml
homeassistant: true
advanced:
  # log_level: debug
  network_key:
    - 237
    - 55
    - 91
    - 2
    - 144
    - 251
    - 253
    - 199
    - 30
    - 235
    - 159
    - 138
    - 203
    - 43
    - 133
    - 113
  pan_id: 41531
  ext_pan_id:
    - 123
    - 104
    - 250
    - 62
    - 90
    - 148
    - 1
    - 118
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  channel: 25
  last_seen: ISO_8601
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt:1883
  user: fake_user
  password: fake_password
serial:
  port: tcp://192.168.1.104:6638
  adapter: zstack
  baudrate: 115200
frontend:
  port: 8080
device_options:
  legacy: false
devices: devices.yaml
groups: groups.yaml
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
```

Here is my `compose.yml` :

```yaml
services:
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:2.0.0
    container_name: z2m-sonoff
    hostname: z2m-sonoff
    restart: always
    networks:
      lan:
        ipv4_address: $IP_ADDR
    environment:
      - TZ=Europe/Paris
    volumes:
      - data:/app/data
    healthcheck:
      test: ["CMD", "nc", "-z", "localhost", "8080"]
      interval: 5s
      timeout: 10s
      retries: 5

volumes:
  data:
networks:
  lan:
    external: true
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

SMLIGHT SLZB-06 CC2652P Ethernet mode

### Setup

docker compose

keidischoi, 2025-01-04:
[image]

cmdshft, 2025-01-04:
It seems that I am seeing more and more reports of the SLZB-06 no longer working with Z2M as of the 2.0.0-1 release and none of the typical fixes are working for anyone.

I have `z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby` occurring now and nothing I do resolves it. I have tried different firmwares on the stick, I tried generating new IDs, adding new values, the legacy false settings in YAML, power cycling the stick... Nothing is working. This is very frustrating, was this release even tested on this stick by anyone?

Jaxel, 2025-01-04:
I am having the exact same error:
```
permit_join: false
serial:
  port: /dev/ttyUSB0
  adapter: ember
frontend:
  port: 8124
mqtt:
  server: mqtt://172.17.0.1:1883
  user: -----------------
  password: -----------------
  base_topic: zigbee2mqtt
  include_device_information: true
homeassistant:
  discovery_topic: homeassistant
  legacy_entity_attributes: false
  legacy_triggers: false
advanced:
  log_level: info
  network_key: '!secret.yaml network_key'
  channel: 25
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
devices: devices.yaml
groups: groups.yaml
```
I am using the Home Assistant Skyconnect; running on Ember.

jlmkerr, 2025-01-04:
Also having this issue. Has anyone got a workaround for now or way to rollback?

My startup logs are:

[16:41:56] INFO: Preparing to start...
[16:41:56] INFO: Socat not enabled
[16:41:56] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 16:41:57] error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-01-04 16:42:28] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-01-04 16:42:28] error: 	zh:ember: Adapter fatal error: ERROR_SERIAL_INIT
[2025-01-04 16:42:28] error: 	z2m: Adapter disconnected, stopping
[2025-01-04 16:42:28] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at Controller.onZigbeeAdapterDisconnected (/app/lib/controller.ts:260:9)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

Jaxel, 2025-01-05:
I figured out this issue... it comes from the following line in configuration.yaml
```
groups: groups.yaml
```

Because I have no groups, this file groups.yaml was EMPTY. So the declaration resolved to falsy. Just delete this line, and everything fixes itself.

XIIIX13, 2025-01-05:
> I figured out this issue... it comes from the following line in configuration.yaml
> 
> ```
> groups: groups.yaml
> ```
> 
> Because I have no groups, this file groups.yaml was EMPTY. So the declaration resolved to falsy. Just delete this line, and everything fixes itself.

It works! Thanks

thierryprats, 2025-01-05:
> I figured out this issue... it comes from the following line in configuration.yaml
> 
> ```
> groups: groups.yaml
> ```
> 
> Because I have no groups, this file groups.yaml was EMPTY. So the declaration resolved to falsy. Just delete this line, and everything fixes itself.

It works, thank you !

closed 2025-01-05 by thierryprats

ElStupid, 2025-01-06:
Confirmed, this is the workaround, thanks!

Beware to specify the extra settings fro HA, without that it still didn't work

`advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
`

levexis, 2025-01-11:
Should this have been closed, surely some error handling / type checking would be quick, easy and helpful?

lbird-mma, 2025-01-12:
> Should this have been closed, some error handling / type checking would be quick, easy and helpful?

Agree!

WilliLz, 2025-02-05:
> I figured out this issue... it comes from the following line in configuration.yaml
> 
> ```
> groups: groups.yaml
> ```
> 
> Because I have no groups, this file groups.yaml was EMPTY. So the declaration resolved to falsy. Just delete this line, and everything fixes itself.

just as me, many thanks, 
now upgrade works fine on homeassistant

JC38, 2025-02-24:
FYI, starting from a fresh install (2.11) but adding the line
devices: devices.yaml gave the same error

WilliLz, 2025-02-25:
is 'device.yaml' an empty file? if yes, I would first add a device and after working move it in 'device: device.yaml'

JC38, 2025-02-25:
Yes, device.yaml was there but empty. I've removed the line in my configuration.yaml as I am not on site for now for enrolling any devices. Affter I'll enroll one, I will split the devices into their dedicated files.
However, I think that the error that occurred with empty files like groups.yaml and devices.yaml could be handle with a piece of code. But unfortunately I don't have the skill for this.