Koenkk/zigbee2mqtt #25458: Fantem ZB003-X Illuminance not selectable as trigger or condition in automation

Dimmers81, 2025-01-04:
### What happened?

I use the Fantem ZB003-X https://www.zigbee2mqtt.io/devices/ZB003-X.html#fantem-zb003-x
Since updating to Zigbee2MQTT version 2.0.0, the Illuminance (numeric) is no longer a selectable option as a trigger for automation device trigger or numerous state and so on. EG lux is greater than 10.

That said, illuminance calibration is available in the dropdown
Since updating to Zigbee2MQTT version 2.0.0, the Illuminance (numeric) is no longer a selectable option as a trigger for automation device triggers or numerous states and so on.

Please note I can see it in the device exposes, but it's just not accessible outside of this.

### What did you expect to happen?

To have it as a trigger or part of a condition.

### How to reproduce it (minimal and precise)

Create an automation in this device and you will be unable to use it.

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Dell Wyse 5070 NUC

adminotaurus44, 2025-01-04:
Same here for the [Aqara T1](https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html#aqara-gzcgq11lm).

![1](https://github.com/user-attachm
[image]
ents/assets/ab18ac29-7ea5-4caa-88fe-4eae7b21a6b7)

mamema, 2025-01-04:
they are disabled now. you need to enable them first

adminotaurus44, 2025-01-04:
> they are disabled now. you need to enable them first

Thank you but how do I do it? There is nothing to enable in the device settings in z2m.

cedricbarbecue, 2025-01-04:
I did it directly on the device on home assistant interface
[image]

mamema, 2025-01-04:
read your own screenshot "This entity is disabled". It should be enabled (Button upper right corner "Enable"

cedricbarbecue, 2025-01-04:
you could also read my comment. I just wanted to show where to enable it... as it is asked by @adminotaurus44

Simonovsky, 2025-01-04:
> I did it directly on the device on home assistant interface <img alt="image" width="1557" src="https://private-user-images.githubusercontent.com/157848867/400130096-fc628e8b-3de8-4199-8eed-49cfd17c2b06.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5ODU4NDAsIm5iZiI6MTczNTk4NTU0MCwicGF0aCI6Ii8xNTc4NDg4NjcvNDAwMTMwMDk2LWZjNjI4ZThiLTNkZTgtNDE5OS04ZWVkLTQ5Y2ZkMTdjMmIwNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNFQxMDEyMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kM2NmYzVjOTQzNmE4OGIzMDBjZTgwZWE4MzMyOWE2YTc2NjU5ZWJmMGM3MzFmYWFjOWFhY2RjNjRlOTA0NDZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ibY2sMLejBsk-c6tdb8ILML_aM7ouxX25KxgSMoekcA">

Thank you so much, this wasn't really obvious from the release notes for me as well. If i may ask does anyone know why is it disabled by default? To me it seems like unnecessary obfuscation especially for new/unexperienced users. Why not just show all available sensors? I dont mean to be rude, just curious :)

Koenkk, 2025-01-05:
It is disabled by because it was initially discovered as disabled (since `illuminance_lux` was used back then). Unfortunately Home Assistant doesn't automatically enable it when discovered as "enabled by default". Users have to manually enable it, I updated https://github.com/Koenkk/zigbee2mqtt/discussions/24198 with this.

closed 2025-01-05 by Koenkk