Koenkk/zigbee2mqtt #25459: Z2M V2.00 will not upgrade/compile on RPi Linux

3pcedev, 2025-01-04:
### What happened?

Previously had Z2M V1.4.2 running on simple Linux (Debian based) installation. This has run successfully for many months.

Attempted to upgrade to V2.0.0 via ./upgrade.sh however this failed.

Read the breaking changes, along with the Linux install instructions (using pnpm rather than npm etc).

Attempted to rebuild using pnpm vs npm. This also failed.

Finally backed up the data folder (moved to separate location) and deleted the entire /opt/zigbee2mqtt directory. From there I followed installation instructions from the Z2M website for a Linux install. Install failed at the same point (and error) as when I attempted to rebuild earlier. I have not copied the data folder back into /opt/zigbee2mqtt (i.e. currently a vanilla install).

Error message is as follows:

/opt/zigbee2mqtt/pnpm run build
> zigbee2mqtt@2.0.0 build /opt/zigbee2mqtt
> tsc && node index.js writehash

Error: ENOENT: no such file or directory, open '/opt/zigbee2mqtt/dist/.hash'
    at Object.writeFileSync (node:fs:2367:20)
    at writeHash (/opt/zigbee2mqtt/index.js:71:8)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
 ELIFECYCLE  Command failed with exit code 1.

This error is similar to another error (previously logged as an issue on GitHub) however following the same steps for resolution (deleting entire install after backing up data) did not make any difference.

I can confirm there is no folder /opt/zigbee2mqtt/dist/ present. Not sure if there is supposed to be or not?

I can confirm the permissions for the Z2M folder are correct as per the install instructions (see below):

ls-l of /opt/zigbee2mqtt:

total 492
-rw-r--r--  1 matthew matthew 187271 Jan  4 03:33 CHANGELOG.md
-rw-r--r--  1 matthew matthew   5447 Jan  4 03:33 CODE_OF_CONDUCT.md
-rw-r--r--  1 matthew matthew   1355 Jan  4 03:33 CONTRIBUTING.md
-rw-r--r--  1 matthew matthew  35141 Jan  4 03:33 LICENSE
-rw-r--r--  1 matthew matthew   7502 Jan  4 03:33 README.md
-rwxr-xr-x  1 matthew matthew    179 Jan  4 03:33 cli.js
drwxr-xr-x  2 matthew matthew   4096 Jan  4 03:33 data
drwxr-xr-x  2 matthew matthew   4096 Jan  4 03:33 docker
-rw-r--r--  1 matthew matthew   1048 Jan  4 03:33 eslint.config.mjs
drwxr-xr-x  2 matthew matthew   4096 Jan  4 03:33 images
-rw-r--r--  1 matthew matthew   5782 Jan  4 03:33 index.js
drwxr-xr-x  6 matthew matthew   4096 Jan  4 03:33 lib
drwxr-xr-x 10 matthew matthew   4096 Jan  4 03:34 node_modules
-rw-r--r--  1 matthew matthew   3240 Jan  4 03:33 package.json
-rw-r--r--  1 matthew matthew 111610 Jan  4 03:33 pnpm-lock.yaml
-rw-r--r--  1 matthew matthew    375 Jan  4 03:33 release-please-config.json
drwxr-xr-x  2 matthew matthew   4096 Jan  4 03:33 scripts
drwxr-xr-x  5 matthew matthew   4096 Jan  4 03:33 test
-rw-r--r--  1 matthew matthew    659 Jan  4 03:33 tsconfig.json
-rw-r--r--  1 matthew matthew  78787 Jan  4 03:35 tsconfig.tsbuildinfo
-rwxr-xr-x  1 matthew matthew    634 Jan  4 03:33 update.sh

ALSO:
ls -l /opt/zigbee2mqtt
total 4
drwxr-xr-x 11 matthew matthew 4096 Jan  4 03:35 zigbee2mqtt

Also note versions below:

NPM version 10.8.2
pnpm version 9.15.2
node version 20.18.1

### What did you expect to happen?

Install to have completed successfully

### How to reproduce it (minimal and precise)

As per description above

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

Didn't even get this far

### Adapter

Did not get this far

### Setup

Debian linux, Rpi zero W

### Debug log

No log file generated as install was never completed.

Koenkk, 2025-01-05:
Could you try again after removing `tsconfig.tsbuildinfo`?

3pcedev, 2025-01-05:
I deleted tsconfig.tsbuildinfo from the /opt/zigbee2mqtt folder. This solved the issue.

'pnpm run build' succeeded and Z2M started normally.

Thanks for your help!

closed 2025-01-05 by 3pcedev