Koenkk/zigbee2mqtt #25460: Philips 324131092621 - z2m 2.0 does not expose "action" in HASS integration

tomtom13, 2025-01-04:
### What happened?

After upgrading z2m to 2.0 via hass update process, for all philips hue dimmer switches:
Philips 324131092621
"action" have disappeared from the hass. Luckilly for me, I backup whole VM so a full restore fixed the problem. 

### Zigbee2MQTT version

2.0

### Adapter firmware version

20240710

### Adapter

SLZB06

### Setup

add-on to home assistant inside of VM.

Drafteed, 2025-01-04:
Please always read the [release notes](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0), especially the breaking changes section, before updating.

psybernoid, 2025-01-04:
I too am having the exact same issue. Have reverted back to 1.42.0 for now and restore the configuration.yaml

Reading the release notes, I'm unsure what to do to reasolve it in version 2

I added
`advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false`

as described in the docs, but that was _after_ the upgrade happened. Should I add that before the upgrade?

tomtom13, 2025-01-04:
Ok, so I see that “_action” got removed and triggers are supposed to be used. However I’m a bit confused here because:
Philips 8718699693985
_action still exists and acts how it’s supposed to.
Additionally if anybody would point me toward how to use triggers along with node-red it would be fantastic.
(It’s going to be long few evenings to reconfigure all automations)

KennethLavrsen, 2025-01-04:
I actually installed previous version and moved all switches from Deconz to Zigbee2MQTT because the new action events were comming. The new way is pretty cool unlike the horrible device triggers

Here is an example of a full HA automation for a Philips dimmer controlling lights incl up and down dimming

```
alias: Bedroom Dimmer
triggers:
  - trigger: state
    entity_id: event.bedroom_switch_action
    to: null
conditions:
  - condition: template
    value_template: "{{trigger.from_state.state != 'unavailable'}}"
actions:
  - choose:
      - conditions:
          - "{{trigger.to_state.attributes.button == 'on'}}"
          - "{{trigger.to_state.attributes.event_type == 'press'}}"
        sequence:
          - data:
              entity_id:
                - light.bedroom
              brightness_pct: 50
            action: light.turn_on
      - conditions:
          - "{{trigger.to_state.attributes.button == 'on'}}"
          - "{{trigger.to_state.attributes.event_type == 'hold'}}"
        sequence:
          - data:
              entity_id:
                - light.bedroom
                - light.bedroom_desk
                - light.background
              brightness_pct: 100
            action: light.turn_on
      - conditions:
          - "{{trigger.to_state.attributes.button == 'up'}}"
          - "{{trigger.to_state.attributes.event_type == 'hold'}}"
        sequence:
          - action: mqtt.publish
            metadata: {}
            data:
              topic: zigbee2mqtt/Bedroom/set
              payload: "{\"brightness_move\": 50 }"
      - conditions:
          - "{{trigger.to_state.attributes.button == 'down'}}"
          - "{{trigger.to_state.attributes.event_type == 'hold'}}"
        sequence:
          - action: mqtt.publish
            metadata: {}
            data:
              topic: zigbee2mqtt/Bedroom/set
              payload: "{\"brightness_move\": -50 }"
      - conditions:
          - "{{trigger.to_state.attributes.button in ('up', 'down')}}"
          - "{{trigger.to_state.attributes.event_type == 'hold_release'}}"
        sequence:
          - action: mqtt.publish
            metadata: {}
            data:
              topic: zigbee2mqtt/Bedroom/set
              payload: "{\"brightness_move\": 0 }"
      - conditions:
          - "{{trigger.to_state.attributes.button == 'off'}}"
          - "{{trigger.to_state.attributes.event_type == 'press'}}"
        sequence:
          - data:
              entity_id:
                - light.bedroom
            action: light.turn_off
      - conditions:
          - condition: template
            value_template: "{{trigger.to_state.attributes.button == 'off'}}"
          - condition: template
            value_template: "{{trigger.to_state.attributes.event_type == 'hold'}}"
        sequence:
          - data:
              entity_id:
                - light.bedroom
                - light.bedroom_desk
                - light.background
            action: light.turn_off
initial_state: "on"
mode: queued
max_exceeded: silent
```

skerbl, 2025-01-04:
> Additionally if anybody would point me toward how to use triggers along with node-red it would be fantastic. (It’s going to be long few evenings to reconfigure all automations)

It's actually fairly simple, there's a "device trigger" node in Node Red in the category home assistant. The node color should be green instead of the usual blue. However, I don't think it's possible anymore to use a "switch" node for the various actions. Not that big of a deal for simple switches, but one of my devices has 5 buttons and exposes a total number of 13 button actions. Not a fun time.

gerrytb, 2025-01-04:
I have the same problem, after migration to mqtt 2.0.0 and enabling Home Assistant legacy entity attributes, I have the action but when I create an automation to control the remote, it does nothing....

tomtom13, 2025-01-05:
In node-red I can see that there is a "device" which when placed on the canvas expands to "device trigger" - great, however it seems that I need to setup a input node for each trigger. No biggy, but a bit cumbersome.

Koenkk, 2025-01-05:
This is one of the 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

>  - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>    ```yaml
>    homeassistant:
>      legacy_action_sensor: true
>    ```

closed 2025-01-05 by Koenkk