Koenkk/zigbee2mqtt #25461: Migration 1.42.0 to 2.0.0 - Missing action in home assistant

kleymanm1, 2025-01-04:
### What happened?

aftre migration to z2m 2.0.0-1 from 1.42 missing action for multiple svitches such as sonoff and moes, in mqtt explorer can for device under z2m can see action but in homeassistant screenshot in 1.42 action is present after update will be missing.
[image]
[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

 20240315

### Adapter

zStack3x0

### Setup

Add-on on Home assistant NUC

anttirokka, 2025-01-04:
i had the same issue and searched for reasons, for me it looks like the action is disabled by default and should be replaced by device trigger.  Device trigger works for my case. Didnt test enabling the legacy action sensor

https://github.com/Koenkk/zigbee2mqtt/discussions/24198

All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml.
homeassistant:
  legacy_action_sensor: true

mstuij, 2025-01-04:
I have the same issues and I have set the legacy_action_sensor to true and with that it works again.

blind-oracle, 2025-01-04:
For me setting the `legacy_action_sensor: true` didn't help, the action entities are not created.

blind-oracle, 2025-01-04:
Update: it seems that UI settings override the ones from the `yaml`, so one has to set this in the UI.

KennethLavrsen, 2025-01-04:
I wrote this posting for HA that can help. https://community.home-assistant.io/t/using-the-new-action-events-in-zigbee2mqtt-2-0/821709

dpgh947, 2025-01-04:
> Update: it seems that UI settings override the ones from the `yaml`, so one has to set this in the UI.

Can you show how you set this? I cannot get it working.

m0wlheld, 2025-01-04:
> For me setting the `legacy_action_sensor: true` didn't help, the action entities are not created.

Maybe you have automatic activation of new entities disabled in HA's system options for the MQTT integration?

blind-oracle, 2025-01-04:
> Maybe you have automatic activation of new entities disabled in HA's system options for the MQTT integration?

No no, I was just updating yaml, but UI overrides it.

@dpgh947 just open z2mqtt UI in homeassistant or wherever it is and go to settings there to set this option.

dpgh947, 2025-01-04:
> just open z2mqtt UI in homeassistant or wherever it is and go to settings there to set this option.

I could not get it working via the ui no matter what I tried. I have now edited the z2m config yaml file directly and added it to the homeassistant: section there and it is now working.

dpgh947, 2025-01-04:
AHA! I now see past the confusion I think is affecting many people. When folks talk about changing the option in the UI, I thought this was referring to the configuration tab of the addon. But if you go to the addon page (clicking z2m in the lefthand menu), then settings (under the cog wheel in the top menu), then the home assistant tab, there is a tickbox there for this option:

[image]

dmosquera87, 2025-01-04:
was anyone able to make it work with Node Red?
many automatons in my Node Red got broken due to "Actions" not being populated anymore. i added the flag legacy_action_sensor: true but no dice. had to rollback. also i dont see how to use the new function "trigger" using Node Red

dpgh947, 2025-01-04:
I had the same problem, broke node red. See my post above yours, mine is working again. Short term fix as this is also going to be removed, but it will get you going for now. I think listening to the mqtt traffic may be the way to go in node red. I think these automation triggers that rely on the unreadable device name is a huge backward step, as this will change if the device is ever removed and added back.

dmosquera87, 2025-01-04:
Thanks! it worked, as you said i was editing the .yml in the addon instead.

Hopefully we find a long-term solution as this will be deprecated as you said. i use Node Red for everything

getget1980, 2025-01-05:
Hello, just to follow the node-red long-term solution.

As no more action is triggered I don't see how we can manage to use node-red events like before ?

Koenkk, 2025-01-05:
This is one of the 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

>  - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>    ```yaml
>    homeassistant:
>      legacy_action_sensor: true
>    ```

closed 2025-01-05 by Koenkk

AndreKR, 2025-03-19:
I think I'm running into #25614 as my E1524 show pretty much the same behavior as the E2002.