Koenkk/zigbee2mqtt #25462: UZG-01 on lan failed to start since 2.0.0 upgrade

artoodetoo3301, 2025-01-04:
### What happened?

Log:

[13:10:19] INFO: Preparing to start...
[13:10:19] INFO: Socat not enabled
[13:10:19] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 13:10:21] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 13:10:21] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 13:10:21] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 13:10:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 13:10:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 13:10:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 13:10:21] error: 	z2m: Exiting...
[2025-01-04 13:10:21] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

The XZG was connected through Ethernet.
I am sure the ports and IP are correct.

### How to reproduce it (minimal and precise)

Happens since upgrade,

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240909

### Adapter

XZG-01

### Setup

Proxmox HA, hosted on a Ryzen PC

### Debug log

[13:10:19] INFO: Preparing to start...
[13:10:19] INFO: Socat not enabled
[13:10:19] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 13:10:21] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 13:10:21] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 13:10:21] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 13:10:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 13:10:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 13:10:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 13:10:21] error: 	z2m: Exiting...
[2025-01-04 13:10:21] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

AJ-Reckless, 2025-01-04:
My TubesZB CC2652P2 based LAN coordinator is having the same issue as of the 2.0.0 update.

Log:

[2025-01-04 00:01:51] info:     z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-04 00:01:51] info:     z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 00:01:51] error:    z2m: Error while starting zigbee-herdsman
[2025-01-04 00:01:51] error:    z2m: Failed to start zigbee-herdsman
[2025-01-04 00:01:51] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 00:01:51] error:    z2m: Exiting...
[2025-01-04 00:01:51] error:    z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

ZHonry, 2025-01-04:
#25443

CVKBaca, 2025-01-04:
I have the same issue with my UZG-01. Based on the recommendations, I modified the configuration as follows:
port: tcp://192.168.1.205:6638
baudrate: 115200
adapter: zstack
disable_led: false

However, after applying these changes, I encountered the following error in the logs:

[10:04:42] INFO: Preparing to start...
[10:04:42] INFO: Socat not enabled
[10:04:42] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 10:04:45] info: z2m: Logging to console, file (filename: log.log)
[2025-01-04 10:04:45] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 10:04:45] info: z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 10:04:45] info: zh:zstack:znp: Opening TCP socket with 192.168.1.205:6638
[2025-01-04 10:04:45] info: zh:zstack:znp: Socket connected
[2025-01-04 10:04:45] info: zh:zstack:znp: Socket ready
[2025-01-04 10:04:45] info: zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-04 10:04:46] info: zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-04 10:05:04] error: z2m: Error while starting zigbee-herdsman
[2025-01-04 10:05:04] error: z2m: Failed to start zigbee-herdsman
[2025-01-04 10:05:04] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 10:05:04] error: z2m: Exiting...
[2025-01-04 10:05:04] error: z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

iroger, 2025-01-04:
Same issue here after update to 2.0.0-1 on XZG connected via ethernet

serial:
  port: tcp://192.168.2.94:6638

`Starting Zigbee2MQTT without watchdog.
[2025-01-04 12:36:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 12:36:49] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 12:36:49] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 12:36:49] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 12:36:49] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 12:36:49] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 12:36:49] error: 	z2m: Exiting...
[2025-01-04 12:36:49] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)`

m0wlheld, 2025-01-04:
> Specify valid 'adapter' and 'port' in your configuration.

Did you do this?

AJ-Reckless, 2025-01-04:
> > Specify valid 'adapter' and 'port' in your configuration.
> 
> Did you do this? 

In case anyone else winds up here this fixed it for me.  Needed to add 'adapter: zstack' to my config. It's documented in the notes for this release, but I wasn't expecting a version update to break existing adapter configuration settings.

m0wlheld, 2025-01-04:
> It's documented in the notes for this release, but I wasn't expecting a version update to break existing adapter configuration settings.

C'mon, be fair: You did not read the releases notes, this is a major release 1.x to 2.x and the notes start with:

[image]

AJ-Reckless, 2025-01-04:
I'm not sure how you interpreted my response but it's not the dig you think it was.  You're correct, and that's exactly what I'm saying.  I didn't read the release notes until afterwards and just updated my Docker container when I was notified one was available.  It's completely fair to state that I didn't think it would break existing adapter configs (because it's true, I didn't, and hadn't encountered similar issues applying previous updates), but the release notes state otherwise.  My response was posted to confirm your solution for those who inevitably do the same as me (and others apparently).  Thanks for your help.

> > It's documented in the notes for this release, but I wasn't expecting a version update to break existing adapter configuration settings.
> 
> C'mon, be fair: You did not read the releases notes, this is a major release 1.x to 2.x and the notes start with:
> 
> ![grafik](https://private-user-images.githubusercontent.com/23745709/400152954-ff8ceeb6-215c-4383-8180-bd3fa50a08b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMTQ3MDUsIm5iZiI6MTczNjAxNDQwNSwicGF0aCI6Ii8yMzc0NTcwOS80MDAxNTI5NTQtZmY4Y2VlYjYtMjE1Yy00MzgzLTgxODAtYmQzZmE1MGEwOGI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDE4MTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzNmU2MWQwYzM1MTMxMzIyNzZiYTY2YmUyZjUyM2E4OWJiYjMzYTBhYWU1OWI0MTQ1ZjE2NDI1NDRmOGFlYWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yENN66RRvlx6-ZaQOWD68OhkO0t1muvthURxZKm28_8)

iroger, 2025-01-04:
> In case anyone else winds up here this fixed it for me. Needed to add 'adapter: zstack' to my config. It's documented in the notes for this release, but I wasn't expecting a version update to break [...]

Thanks, it was the “zstack” config in my case. I have a very recent z2m setup and all the other “legacy” stuff was not relevant but I missed the zstack change.

coasttech, 2025-01-04:
> C'mon, be fair: You did not read the releases notes, this is a major release 1.x to 2.x and the notes start with: [...]

Thank you and you don't have to be so mean :P

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 on how to fix this.

closed 2025-01-05 by Koenkk