Koenkk/zigbee2mqtt #25463: 2.0 cannot load external converters

ZHonry, 2025-01-04:
### What happened?

Previously, an external converter was used for normal parsing. After upgrading to 2.0, the converter file was moved to the data/external_extensions/ folder. An error occurred when restarting z2m.

`[2025-01-04 13:38:54] error: 	z2m: Failed to call 'ExternalExtensions' 'start' (TypeError: module is not a constructor

    at ExternalExtensions.loadJS (/app/lib/extension/externalExtensions.ts:43:13)

    at ExternalExtensions.loadFiles (/app/lib/extension/externalJS.ts:156:13)

    at ExternalExtensions.start (/app/lib/extension/externalJS.ts:45:9)

    at Controller.callExtensions (/app/lib/controller.ts:368:17)

    at Controller.start (/app/lib/controller.ts:186:9)

    at start (/app/index.js:161:5))`

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

// 常量定义
const DEVICE_MODEL = 'TS0026';
const MANUFACTURER_NAME = '_TYZB01_gffjp6f2';

// 本地转换器定义
const fzLocal = {
    action: {
        cluster: 'genOnOff',
        type: ['commandTuyaAction'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.endpoint.ID;
            return { action: `scene_${dp}` };
        },
    },
};

// 设备定义
const definition = {
    fingerprint: [
        {
            modelID: DEVICE_MODEL,
            manufacturerName: MANUFACTURER_NAME,
        },
    ],
    model: DEVICE_MODEL,
    vendor: 'Tuya',
    description: '6 button scene wall switch (AC powered)', // 明确说明是 AC 供电
    fromZigbee: [fzLocal.action, tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // 处理时间同步事件
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1); // 假设设备使用 endpoint 1
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
    exposes: [
        e.action(['scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'action', tuya.valueConverter.static('scene_1')],
            [2, 'action', tuya.valueConverter.static('scene_2')],
            [3, 'action', tuya.valueConverter.static('scene_3')],
            [4, 'action', tuya.valueConverter.static('scene_4')],
            [5, 'action', tuya.valueConverter.static('scene_5')],
            [6, 'action', tuya.valueConverter.static('scene_6')],
        ],
    },
    extend: [], // 扩展功能（可选）
};

module.exports = definition;
`

### Zigbee2MQTT version

 2.0.0 commit: 060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a

### Adapter firmware version

20190619

### Adapter

zStack12

### Setup

docker

### Debug log

[2025-01-04 13:38:40] info: 	z2m: Logging to console, file (filename: log.log)

[2025-01-04 13:38:40] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)

[2025-01-04 13:38:40] info: 	z2m: Starting zigbee-herdsman (3.2.1)

[2025-01-04 13:38:40] info: 	zh:zstack:znp: Opening TCP socket with 192.168.2.120:8880

[2025-01-04 13:38:40] info: 	zh:zstack:znp: Socket connected

[2025-01-04 13:38:40] info: 	zh:zstack:znp: Socket ready

[2025-01-04 13:38:40] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload

[2025-01-04 13:38:41] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352

[2025-01-04 13:38:54] info: 	z2m: zigbee-herdsman started (resumed)

[2025-01-04 13:38:54] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'

[2025-01-04 13:38:54] info: 	z2m: 0xb4e3f9fffe90a03f (0xb4e3f9fffe90a03f): TS0026 - Tuya 6 button scene wall switch (Router)

[2025-01-04 13:38:54] info: 	z2m: Currently 1 devices are joined.

[2025-01-04 13:38:54] info: 	z2m: Connecting to MQTT server at mqtt://192.168.2.6:1883

[2025-01-04 13:38:54] info: 	z2m: Connected to MQTT server

[2025-01-04 13:38:54] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

[2025-01-04 13:38:54] error: 	z2m: Failed to call 'ExternalExtensions' 'start' (TypeError: module is not a constructor

    at ExternalExtensions.loadJS (/app/lib/extension/externalExtensions.ts:43:13)

    at ExternalExtensions.loadFiles (/app/lib/extension/externalJS.ts:156:13)

    at ExternalExtensions.start (/app/lib/extension/externalJS.ts:45:9)

    at Controller.callExtensions (/app/lib/controller.ts:368:17)

    at Controller.start (/app/lib/controller.ts:186:9)

    at start (/app/index.js:161:5))

[2025-01-04 13:38:54] info: 	z2m: Started frontend on port 8080

[2025-01-04 13:38:54] info: 	z2m: Zigbee2MQTT started!

rp2019433, 2025-01-04:
There is an issue open: [25447](https://github.com/Koenkk/zigbee2mqtt/issues/25447).

Gexwh, 2025-01-04:
I have the same: z2m: Failed to call 'ExternalExtensions' 'start' (TypeError: module is not a constructor

m0wlheld, 2025-01-04:
> After upgrading to 2.0, the converter file was moved to the data/external_extensions/ folder.

If I read the instructions correctly, the folder is `data/external_converters` (not _extensions).

> * The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.

aberenguel, 2025-01-04:
The absolute path is `/data/external_converters` or `/config/zigbee2mqtt/data/external_converters` ? 
No success to load my external converters yet.

ZHonry, 2025-01-04:
> If I read the instructions correctly, the folder is `data/external_converters` (not _extensions).

Thank you very much for your reminder. I created a new "external_converters" folder and moved the converter files into it. Now they are loaded normally.

closed 2025-01-04 by ZHonry

arthurperezbessa, 2025-01-06:
Can't figure where to put the external converters .js files. I've tried:

- /config/zigbee2mqtt
- /config/zigbee2mqtt/external_converters
- /config/zigbee2mqtt/data/external_converters

aberenguel, 2025-01-06:
I've got a solution in https://community.home-assistant.io/t/z2mqtt-external-converter/822123/9

- open Terminal
```sh
cd homeassistant/
cd zigbee2mqtt/
mkdir external_converters
cd external_converters
touch convertor.js
```
- open File editor
paste your convertor code

arthurperezbessa, 2025-01-06:
Still no luck

[image]

Restarted YAML, Zigbee and the entire home assistant. Still unsupported. It was working fine before the upgrade.

arthurperezbessa, 2025-01-06:
So I used the command with sudo before and workes.
sudo touch converter.js

Timieh, 2025-01-11:
I edited the doc's to correct. 
https://github.com/Koenkk/zigbee2mqtt.io/pull/3418

lattice0, 2025-02-22:
can someone please explain if external converters are for zigbee2mqtt or home assistant? where are they stored? I need to know for which project first, itś confusing cause Iḿ using docker

arthurperezbessa, 2025-02-22:
> can someone please explain if external converters are for zigbee2mqtt or home assistant? where are they stored? I need to know for which project first, itś confusing cause Iḿ using docker

Put the external converters folder inside de Zigbee2mqtt folder

lattice0, 2025-02-22:
for docker users it s a bit confusing what is a zigbee2mqtt folder, what is homeassistant, etc. So I was able to do it my mounting like this:

```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
        - /run/udev:/run/udev:ro
        - ./zigbee2mqtt-data:/app/data
        - ./external_converters:/app/data/external_converters
```
the external_converters folder has nothing to do with homeassistant but some users might find zigbee2mqtt inside homeassistant depending on the installation.

I also confused external_converters with external-converters multiple times. Please remember the correct is

`external_converters`.

I wanted to keep the external-converters out of zigbee2mqtt-data for other reasons (backup)