Koenkk/zigbee2mqtt #25464: Zigbee2MQTT won't start after 2.0.0-1 update with Sonoff ZBDongle-E w/ Ember firmware

lbovay, 2025-01-04:
### What happened?

After updating to Zigbee2MQTT 2.0.0-1, the add-on won't start.

I'm using Home Assistant with a Sonoff ZBDongle-E running Ember 7.4.5.0 firmware. Addon configuration:
`adapter: ember`

I cannot find anything usefull in the Addon logs.

### How to reproduce it (minimal and precise)

Start Zigbee2MMQTT Addon in Home Assistant.

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

Ember: ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl

### Adapter

Sonoff ZBDongle-E

### Setup

Homea Assistant

### Debug log

```
[07:25:37] INFO: Preparing to start...
[07:25:39] INFO: Socat not enabled
[07:25:49] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 07:26:14] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 07:26:14] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 07:26:14] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 07:26:15] info: 	zh:ember: Using default stack config.
[2025-01-04 07:26:15] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-04 07:26:15] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-04 07:26:15] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-04 07:26:15] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
```

swat33333, 2025-01-04:
Same Problem with Sonoff-P USB Stick.

Mark-Muc, 2025-01-04:
same here 
usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V

Mark-Muc

uniquePWD, 2025-01-04:
I've got the same problem with Dongle-E.

Tried to switch to zstack even though ember was working fine pre-update, that didn't work. Switched back to ember and set rtscts to true, that didn't work. Set that to false, that didn't work. UArt just refuses to start up.

CVKBaca, 2025-01-04:
I have the same issue with my UZG-01. Based on the recommendations, I modified the configuration as follows:
`port: tcp://192.168.1.205:6638
baudrate: 115200
adapter: zstack
disable_led: false`

However, after applying these changes, I encountered the following error in the logs:
`[10:04:42] INFO: Preparing to start...
[10:04:42] INFO: Socat not enabled
[10:04:42] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 10:04:45] info: z2m: Logging to console, file (filename: log.log)
[2025-01-04 10:04:45] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 10:04:45] info: z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 10:04:45] info: zh:zstack:znp: Opening TCP socket with 192.168.1.205:6638
[2025-01-04 10:04:45] info: zh:zstack:znp: Socket connected
[2025-01-04 10:04:45] info: zh:zstack:znp: Socket ready
[2025-01-04 10:04:45] info: zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-04 10:04:46] info: zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-04 10:05:04] error: z2m: Error while starting zigbee-herdsman
[2025-01-04 10:05:04] error: z2m: Failed to start zigbee-herdsman
[2025-01-04 10:05:04] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 10:05:04] error: z2m: Exiting...
[2025-01-04 10:05:04] error: z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
`

uniquePWD, 2025-01-04:
> I have the same issue with my UZG-01. Based on the recommendations, I modified the configuration as follows:
>
> port: tcp://192.168.1.205:6638 baudrate: 115200 adapter: zstack disable_led: false [...]

That's a different adapter that uses a different software stack. You use zstack, we use ember. I would suggest starting a thread for your specific adapter

uniquePWD, 2025-01-04:
I'm seeing this in my portainer logs
```
[2025-01-04 09:19:27] error: 	z2m: Error while starting zigbee-herdsman

[2025-01-04 09:19:27] error: 	z2m: Failed to start zigbee-herdsman

[2025-01-04 09:19:27] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions

[2025-01-04 09:19:27] error: 	z2m: Exiting...

[2025-01-04 09:19:27] error: 	z2m: Error: Adapter EZSP protocol version (12) is not supported by Host [13-16].

    at EmberAdapter.emberVersion (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1403:19)

    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:674:9)

    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)

    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)

    at Zigbee.start (/app/lib/zigbee.ts:69:27)

    at Controller.start (/app/lib/controller.ts:142:13)

    at start (/app/index.js:161:5)

      
      ```

plaiddk, 2025-01-04:
> same here usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V
> 
> ```
> port: >-
>   /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231101112552-if00
> adapter: ember
> baudrate: 115200
> rtscts: false
> ```
> 
> Mark-Muc

Same here get this error: 

[2025-01-04 10:41:34] error: z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-16].

uniquePWD, 2025-01-04:
Looks like Dongle-E users need to update their firmware: https://github.com/Koenkk/zigbee2mqtt/issues/24417#issuecomment-2425239430

lbovay, 2025-01-04:
> ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl

I'm running "ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl" as stated in the original post to no avail. Did you have success with a Dongle-E on a different firmware ?

ExquisiteMetropolis, 2025-01-04:
Update went flawless on my end. 
My Sonoff Zigbee Dongle-E was already on this firmware:

https://github.com/darkxst/silabs-firmware-builder/releases/download/20241124/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl

In my configuration.yaml I had the following added:

`homeassistant:
  legacy_entity_attributes: false
  legacy_triggers: false`
  homeassistant_legacy_triggers: false
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  channel: 25
  output: attribute_and_json
  last_seen: ISO_8601
  log_level: error
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230509114723-if00
  adapter: ember
frontend:
  port: 8099
device_options:
  legacy: false
permit_join: false`

Mark-Muc, 2025-01-04:
> Looks like Dongle-E users need to update their firmware: [#24417 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/24417#issuecomment-2425239430)

I am runnig 
```
Zigbee (EZSP) 7.4.4.0 build 0
Sonoff ZBDongle-E

```
and
[(https://darkxst.github.io/silabs-firmware-builder/)]
tells me, that's there isn't any newer!!

Mark_Muc

plaiddk, 2025-01-04:
> I am runnig
>
> ```
> Zigbee (EZSP) 7.4.4.0 build 0
> Sonoff ZBDongle-E
> ```
>
> and [(https://darkxst.github.io/silabs-firmware-builder/)] tells me, that's there isn't any newer!!
>
> Mark_Muc

I got it to work with updating to 7.4.4.0 - So maybe your missing something?

Mark-Muc, 2025-01-04:
i now have updated to:
```
Zigbee (EZSP) 8.0.2.0 build 397
Sonoff ZBDongle-E
```
and z2m won't start..
Log:

```
[11:41:37] INFO: Preparing to start...
[11:41:37] INFO: Socat not enabled
[11:41:37] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:
  (0, node_assert_1.default)(result instanceof Object)
    at read (/app/lib/util/yaml.ts:24:15)
    at Object.readIfExists (/app/lib/util/yaml.ts:36:34)
    at readDevicesOrGroups (/app/lib/util/settings.ts:317:38)
    at read (/app/lib/util/settings.ts:325:5)
    at getPersistedSettings (/app/lib/util/settings.ts:393:21)
    at Object.reRead (/app/lib/util/settings.ts:640:5)
    at start (/app/index.js:130:14)
```

uniquePWD, 2025-01-04:
#How To Fix It:

1. In Chrome goto https://darkxst.github.io/silabs-firmware-builder/
2. Press Connect under your device 
3. Press upgrade 
4. When that's finished, it will probably say it failed, don't worry
5. Restart Z2M
6. Restart Home Assistant 
7. Wait for things to finish booting up as it may be a little slow.
8. Smile

Mark-Muc, 2025-01-04:
THX uniquePWD.
So I did now the same . switched back the firmware to 7.4.4.0
and still got the same error in the log:

```
[11:57:56] INFO: Preparing to start...
[11:57:57] INFO: Socat not enabled
[11:57:57] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:
  (0, node_assert_1.default)(result instanceof Object)
    at read (/app/lib/util/yaml.ts:24:15)
    at Object.readIfExists (/app/lib/util/yaml.ts:36:34)
    at readDevicesOrGroups (/app/lib/util/settings.ts:317:38)
    at read (/app/lib/util/settings.ts:325:5)
    at getPersistedSettings (/app/lib/util/settings.ts:393:21)
    at Object.reRead (/app/lib/util/settings.ts:640:5)
    at start (/app/index.js:130:14)
```
I think i will switch back to the last working version and WAIT...

Regards
Mark-Muc

uniquePWD, 2025-01-04:
Yours is looking like a different issue Mark and reading through your logs, it's looking like an issue with your configuration.yaml

lbovay, 2025-01-04:
i tried with this firmware and still the startup hangs at the same step as originally posted.

`Zigbee (EZSP) 8.0.2.0 build 397
Sonoff ZBDongle-E`

svde, 2025-01-04:
reverted to koenkk/zigbee2mqtt:1.42.0, will wait for this to be fixed in v2

plaiddk, 2025-01-04:
> reverted to koenkk/zigbee2mqtt:1.42.0, will wait for this to be fixed in v2

How do you revert the version if I may ask?

svde, 2025-01-04:
> > reverted to koenkk/zigbee2mqtt:1.42.0, will wait for this to be fixed in v2
> 
> How do you revert the version if I may ask?

changed in tag of the docker image from koenkk/zigbee2mqtt:latest to koenkk/zigbee2mqtt:1.42.0

(I run Zigbee2MQTT in docker)

uniquePWD, 2025-01-04:
> i tried with this firmware and still the startup hangs at the same step as originally posted.
> 
> [Zigbee (EZSP) 8.0.2.0 build 397 Sonoff ZBDongle-E](https://github.com/darkxst/silabs-firmware-builder/releases/download/20241124/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl)
> 
> from this repo: [silabs-firmware-builder](https://github.com/darkxst/silabs-firmware-builder)
> 
> Also tried to uninstall and delete the Addon data and re-install it, but nothing seems to work. It still hangs at startup.

Use version 7.4.4.0 based on the post above. Anyone using version 8 and finding things aren't working, your inability to follow instructions will be your bane.

ExquisiteMetropolis, 2025-01-04:
@lbovay & @svde maybe you can post your configuration.yaml files so we can look what's the different from succesfull updates?
Must be something that can clairfy the issue, if it works for others, there should be a way that it can work for you as well. :)

Did you try this repo/firmware?
https://github.com/darkxst/silabs-firmware-builder/releases/ zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl

lbovay, 2025-01-04:
> @lbovay & @svde maybe you can post your configuration.yaml files so we can look what's the different from succesfull updates? Must be something that can clairfy the issue, if it works for others, [...]

Here is my /homeassistant/zigbee2mqtt/configuration.yaml:

`version: 4
homeassistant:
  enabled: true
advanced:
  network_key:
    - 66
    - ...
  pan_id: 11090
  ext_pan_id:
    - 107
    - ...
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ...
serial:
  port: >-
    /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_e44dfe02b639ef119cf057f454516304-if00-port0
  adapter: ember
frontend:
  enabled: true
  port: 8099`

uniquePWD, 2025-01-04:
Here's my configuration.yaml

```
homeassistant:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.164
  user: [redacted]
  password: [redacted]
serial:
  port: /dev/ttyACM0
  adapter: ember
  rtscts: false
frontend:
  enabled: true
```

svde, 2025-01-04:
I noticed that watchman updated the Zigbee2MQTT image. This is the only things which changed, so I decided to revert that change, going to the latest release of the previous major (1.42). I'm going to run this until there is a clear path forwards. Not going to waste time by trying different firmware versions, etc.

```
serial:
  port: /dev/ttyACM0
```

I tried adding
  adapter: ember
  baudrate: 115200
which didn't help.

lbovay, 2025-01-04:
> Use version 7.4.4.0 based on the post above. Anyone using version 8 and finding things aren't working, your inability to follow instructions will be your bane.

Tried with this: https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

Same behavior.

uniquePWD, 2025-01-04:
> I noticed that watchman updated the Zigbee2MQTT image. This is the only things which changed, so I decided to revert that change, going to the latest release of the previous major (1.42). I'm going [...]

The only working firmware version is 7.4.4.0

> Tried with this: https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
>
> Same behavior.

I told you how to get it working. Why are you trying alternative ways? Follow https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943

Don't try and remix it or be smart. Just follow the simple steps.

ExquisiteMetropolis, 2025-01-04:
your issues might be related to the port: /dev/ttyACM0 entry.
You should not use /dev/ttyACM0 (as that can randomly change).
It's best practice to use the /dev/serial/by-id/usb.... path for your adapter.

Obtain it via: 
`ls -l /dev/serial/by-id`

Should return something like:
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230509114723-if00 

replace your /dev/ttyACM0 with the /dev/serial/etc....

svde, 2025-01-04:
Don't think that's the issue. /dev/ttyACM0 is used inside of the container. Docker links the actual device to that like this:

        "Devices": [
            {
                "CgroupPermissions": "rwm",
                "PathInContainer": "/dev/ttyACM0",
                "PathOnHost": "/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221031184751-if00"
            }
        ],

Also 1.42 works fine.... For me this is a 2.0.0 issue.

lbovay, 2025-01-04:
> > I noticed that watchman updated the Zigbee2MQTT image. This is the only things which changed, so I decided to revert that change, going to the latest release of the previous major (1.42). I'm going to run this until there is a clear path forwards. Not going to waste time by trying different firmware versions, etc.
> > ```
> > serial:
> >   port: /dev/ttyACM0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I tried adding adapter: ember baudrate: 115200 which didn't help.
> 
> The only working firmware version is 7.4.4.0
> 
> > > > i tried with this firmware and still the startup hangs at the same step as originally posted.
> > > > [Zigbee (EZSP) 8.0.2.0 build 397 Sonoff ZBDongle-E](https://github.com/darkxst/silabs-firmware-builder/releases/download/20241124/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl)
> > > > from this repo: [silabs-firmware-builder](https://github.com/darkxst/silabs-firmware-builder)
> > > > Also tried to uninstall and delete the Addon data and re-install it, but nothing seems to work. It still hangs at startup.
> > > 
> > > 
> > > Use version 7.4.4.0 based on the post above. Anyone using version 8 and finding things aren't working, your inability to follow instructions will be your bane.
> > 
> > 
> > Tried with this: https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
> > Same behavior.
> 
> I told you how to get it working. Why are you trying alternative ways? Follow [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943)
> 
> Don't try and remix it or be smart. Just follow the simple steps.

So you're saying to use the version that is offered in the chrome firmware tool:
[image]

I thought you meant 7.4.4 from the ember repo.

Anyway I tried with your steps and it still hangs at the same step.

Additional warning about EZSP being deprecated but nothign new in the logs.

`[2025-01-04 12:59:16] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`

uniquePWD, 2025-01-04:
> > I told you how to get it working. Why are you trying alternative ways? Follow [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943)
> > Don't try and remix it or be smart. Just follow the simple steps.
> 
> So you're saying to use the version that is offered in the chrome firmware tool: ![{99AFFF89-6B9C-43FB-A973-9606AABD064B}](https://private-user-images.githubusercontent.com/17215057/400137015-76afdfea-6af2-4854-b055-98b4105bfed2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5OTI0ODksIm5iZiI6MTczNTk5MjE4OSwicGF0aCI6Ii8xNzIxNTA1Ny80MDAxMzcwMTUtNzZhZmRmZWEtNmFmMi00ODU0LWIwNTUtOThiNDEwNWJmZWQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDEyMDMwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiYjY0NzQyMmJmY2M5NWY0Mzg4YTZmZjVjYzgzZWZjOTIyY2NiYTE2MjdiMGY0NmY1NWE3OTI0NmYwYWZjZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-jgqQ39qC9sdT2beUkLBVmzhVS7P86-o0uVPMwj9B4I)
> 
> I thought you meant 7.4.4 from the ember repo.
> 
> Anyway I tried with your steps and it still hangs at boot.
> 
> Additional warning about EZSP being deprecated but nothign new in the logs.
> 
> `[2025-01-04 12:59:16] warning: zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`

It's saying that your new problem is that you have the wrong adapter set in your configuration.yaml

```
serial:
  port: /dev/ttyACM0
  adapter: ember
  rtscts: false
```

lbovay, 2025-01-04:
> It's saying that your new problem is that you have the wrong adapter set in your configuration.yaml
> 
> ```
> serial:
>   port: /dev/ttyACM0
>   adapter: ember
>   rtscts: false
> ```
> 
> Specifically the last two lines, the adapter and rtscts

I did change the adapter from ember to ezsp to match the firmware. I think the log message just says to use ember instead of the ezsp firmware.

Anyway, looks like a rollback is the only option for now.

svde, 2025-01-04:
ada

> It's saying that your new problem is that you have the wrong adapter set in your configuration.yaml
> 
> ```
> serial:
>   port: /dev/ttyACM0
>   adapter: ember
>   rtscts: false
> ```
> 
> Specifically the last two lines, the adapter and rtscts

Doesn't work (with 1.42 or 2.0.0).

uniquePWD, 2025-01-04:
> I did change the adapter from ember to ezsp to match the firmware. I think the log message just says to use ember instead of the ezsp firmware.
>
> Anyway, looks like a rollback is the only option for [...]

So changing back to ember doesn't fix it despite your logs requesting that you do that specifically? What's the new error message?

uniquePWD, 2025-01-04:
> ada
>
>
> Doesn't work (with 1.42 or 2.0.0).

Did you update your firmware as per post https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943

svde, 2025-01-04:
> > ada
> > > It's saying that your new problem is that you have the wrong adapter set in your configuration.yaml
> > > ```
> > > serial:
> > >   port: /dev/ttyACM0
> > >   adapter: ember
> > >   rtscts: false
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Specifically the last two lines, the adapter and rtscts
> > 
> > 
> > Doesn't work (with 1.42 or 2.0.0).
> 
> Did you update your firmware as per post [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943)

No. Will wait for others facing the same issue to report that it works.

lbovay, 2025-01-04:
> So changing back to ember doesn't fix it despite your logs requesting that you do that specifically? What's the new error message?

It's here: https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571267090

uniquePWD, 2025-01-04:
> No. Will wait for others facing the same issue to report that it works.

😂 You can

> > So changing back to ember doesn't fix it despite your logs requesting that you do that specifically? What's the new error message?
> 
> It's here: [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571267090)

That doesn't answer my question. Does changing to ember in your configuration.yaml display a different error?

Please note, I'm just a fellow community member trying to assist you. I'm not a developer or paid support, I'm just trying to get you over the line so we all win together.

uniquePWD, 2025-01-04:
> > > ada
> > > > It's saying that your new problem is that you have the wrong adapter set in your configuration.yaml
> > > > ```
> > > > serial:
> > > >   port: /dev/ttyACM0
> > > >   adapter: ember
> > > >   rtscts: false
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > Specifically the last two lines, the adapter and rtscts
> > > 
> > > 
> > > Doesn't work (with 1.42 or 2.0.0).
> > 
> > 
> > Did you update your firmware as per post [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943)
> 
> No. Will wait for others facing the same issue to report that it works.
> 
> EDIT: updated the firmware from 6.10.3.0 to 7.4.4.0, added adapter: ember + rtscts: false to the config, changed the tag back to latest and it works. Case closed for me.

Glad you're sorted.

lbovay, 2025-01-04:
There is some useful info here: https://github.com/Koenkk/zigbee2mqtt/issues/25495

But I am still stuck with the symptoms & log entries described in the first post.

TLDR: Z2M hangs on startup at the step where it is opening the serial connection without throwing errors.

Dazzel, 2025-01-04:
I have the same issue. Tried multiple firmware versions but without success.
Rebooted the whole system, used different ports etc.

Also using the home assistant add-on (fresh install of the add-on).

I've tried multiple firmware from [here](https://github.com/darkxst/silabs-firmware-builder):

* 7.4.1
* 7.4.4
* 7.4.5
* 8.0.2

And also the "original" firmware from [here](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master).

uniquePWD, 2025-01-04:
> There is some useful info here: #25495
> 
> But I am still stuck with the symptoms & log entries described in the first post.
> 
> TLDR: Z2M hangs on startup at the step where it is opening the serial connection without throwing errors.

That person didn't add the adapter. Not sure how that's helpful. That's something everyone has been told to do.

As for your problems, I can't help you if you refuse to follow the steps to fix the problems or answer questions. It's like asking why your ice-cream doesn't taste like chocolate even though you keep buying strawberry.

Peixeiro, 2025-01-04:
Did you follow these steps #25464 (comment) exactly?

> Did you follow these steps [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943) exactly?

I followed, but keep not working...

uniquePWD, 2025-01-04:
> Did you follow these steps #25464 (comment) exactly?
> 
> > Did you follow these steps [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571251943) exactly?
> 
> I followed, but keep not working...

Did you make the relevant changes to your configuration.yaml as per 

```
serial:
  port: /dev/ttyACM0
  adapter: ember
  rtscts: false
```

Peixeiro, 2025-01-04:
Yes 
[image]

uniquePWD, 2025-01-04:
> Yes ![imagem](https://private-user-images.githubusercontent.com/55285974/400157876-476017c2-66fa-464f-b729-f8cd33de8c6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMTUyMTMsIm5iZiI6MTczNjAxNDkxMywicGF0aCI6Ii81NTI4NTk3NC80MDAxNTc4NzYtNDc2MDE3YzItNjZmYS00NjRmLWI3MjktZjhjZDMzZGU4YzZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDE4MjE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViMjQ5YzVhMTkxNTRjYmRkYTdjYjcwNWUzOWNkMzI4ZjVkYjA5NjdmOGJjMTlkNGNiNTA3ZTQxOThiNzFkNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ym5xntBoUFPWTqNCx_gskviL326smELh6TdkDlCRzr8)

Post your logs

Peixeiro, 2025-01-04:
`[18:22:49] INFO: Preparing to start...
[18:22:50] INFO: Socat not enabled
[18:22:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 18:23:06] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.18-05-42'
[2025-01-04 18:23:06] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 18:23:06] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 18:23:06] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 18:23:06] debug: 	z2m: sd-notify loaded
[2025-01-04 18:23:06] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 18:23:06] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"/dev/ttyACM0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-04 18:23:06] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20220816092006","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816092006-if00","vendorId":"1a86","productId":"55d4"}]
[2025-01-04 18:23:06] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20220816092006","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816092006-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-01-04 18:23:07] info: 	zh:ember: Using default stack config.
[2025-01-04 18:23:07] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyACM0","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-04 18:23:07] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-04 18:23:07] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-04 18:23:07] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-04 18:23:07] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-04 18:23:07] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}`

uniquePWD, 2025-01-04:
It looks like there's some extra stuff in your configuration.yaml. Post your first 20 lines

Dazzel, 2025-01-04:
I've followed the steps exactly, yes.

But it seems the firmware was not the problem. I had a 32bit installation of home assistant. Since I've wanted to upgrade to the 64bit version a long time ago, I figured this might be worth a shot. And now it works without any problems.

Simply created a full backup of my installation, re-flashed the 64bit home assistant OS, imported the backup and re-installed the z2m-add-on.

uniquePWD, 2025-01-04:
> I've followed the steps exactly, yes.
>
> But it seems the firmware was not the problem. I had a 32bit installation of home assistant. Since I've wanted to upgrade to the 64bit version a long time ago, [...]

Nice one. Glad you're sorted.

Peixeiro, 2025-01-04:
> It looks like there's some extra stuff in your configuration.yaml. Post your first 20 lines

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://1xxxxxxxxxxx:1883
  user: mqttuser
  password: xxxxxxxxxxxxx
serial:
  port: /dev/ttyACM0
  adapter: ember
  rtscts: false
advanced:
  log_level: debug

swat33333, 2025-01-04:
Is There a work around for the sunoff p Stick ?

uniquePWD, 2025-01-04:
> > It looks like there's some extra stuff in your configuration.yaml. Post your first 20 lines
> 
> homeassistant: enabled: true mqtt: server: mqtt://xxxxxxxxxxxxx:1883 user: mqttuser password: xxxxxxxxxxxxxxx keepalive: 60 reject_unauthorized: true version: 4 force_disable_retain: false base_topic: zigbee2mqtt serial: port: /dev/ttyACM0 adapter: ember baudrate: 115200 frontend: enabled: true port: 8099 advanced: log_syslog: app_name: Zigbee2MQTT eol: /n host: localhost localhost: localhost path: /dev/log pid: process.pid port: 123 protocol: tcp4 type: '5424' channel: 25 network_key: - 1 - 3 - 5 - 7 - 9 - 11 - 13 - 15 - 0 - 2 - 4 - 6 - 8 - 10 - 12 - 13 cache_state: true cache_state_persistent: false cache_state_send_on_startup: false log_level: debug last_seen: ISO_8601 device_options: {} availability: enabled: true active: timeout: 30 passive: timeout: 300 devices: .......

There's a bunch of extra stuff in your config that could be causing your issue. Remove everything but the essentials and try a restart.

Peixeiro, 2025-01-04:
> There's a bunch of extra stuff in your config that could be causing your issue. Remove everything but the essentials and try a restart.

Updates, updates, ... I don't know what is the essential!!! I don't add anything manually.

uniquePWD, 2025-01-04:
> Is There a work around for the sunoff p Stick ?

That should be as simple as adding zstack to your config as the adapter. I haven't followed the issues for that device since I don't have it.

uniquePWD, 2025-01-04:
> > There's a bunch of extra stuff in your config that could be causing your issue. Remove everything but the essentials and try a restart.
> 
> Updates, updates, ... I don't know what is the essential!!! I don't add anything manually.

In this post https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571263586, you can see mine. Yours is adding things like network keys, cache, baudrate, etc. All unnecessary things that could be causing your problems.

Peixeiro, 2025-01-04:
Still not working

New configuration.yaml

homeassistant:
  enabled: true
mqtt:
  server: mqtt://xxxxxxxxxxx:1883
  user: mqttuser
  password: xxxxxxxxx
  base_topic: zigbee2mqtt
serial:
  port: /dev/ttyACM0
  adapter: ember
  rtscts: false
frontend:
  enabled: true
  port: 8099
device_options: {}
devices:
  '0x5c0272fffe7f76f0':
    friendly_name: Router
.....

LOG: 

[18:52:18] INFO: Preparing to start...
[18:52:20] INFO: Socat not enabled
[18:52:25] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 18:52:38] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 18:52:38] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 18:52:38] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 18:52:38] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20220816092006","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816092006-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-01-04 18:52:38] info: 	zh:ember: Using default stack config.
[2025-01-04 18:52:38] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-04 18:52:38] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-04 18:52:38] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-04 18:52:38] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.

uniquePWD, 2025-01-04:
> Still not working
>
> New configuration.yaml
>
> homeassistant: enabled: true mqtt: server: mqtt://xxxxxxxxxxx:1883 user: mqttuser password: xxxxxxxxx base_topic: zigbee2mqtt serial: port: /dev/ttyACM0 [...]

Your firmware is wrong

Peixeiro, 2025-01-04:
> Your firmware is wrong

I update with the link you share... It say "Successful"

I restore the z2m backup and all working on version  1.42.0-2

uniquePWD, 2025-01-04:
> > Your firmware is wrong
> 
> I update with the link you share... It say "Successful"
> 
> I restore the z2m backup and all working on version 1.42.0-2

Other than the firmware I can't think of another reason. Your adapter is booting up fine and the fact you can roll back shows there's no issues with the adapter itself. The only other thing I can suggest is that you install Z2M fresh. That said, you can pin your version to 1.4.2 and stay there. I don't think Home Assistant will stop supporting that version any time soon.

[49 more replies]

closed 2025-01-05 by Koenkk