Koenkk/zigbee2mqtt #25465: Upgrade to 2.0 with legacy and deconz conbee iii

talondnb, 2025-01-04:
### What happened?

Upgraded to 2.0 after putting legacy config as suggested [here](https://github.com/Koenkk/zigbee2mqtt/discussions/24198). Now addon won't start.

log.log:
```
[2025-01-04 14:20:09] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 14:20:09] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 14:20:09] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 14:20:10] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"dresden elektronik","serialNumber":"DE03219176","pnpId":"usb-dresden_elektronik_ConBee_III_DE03219176-if00-port0","vendorId":"0403","productId":"6015"} => deconz: 3
```

migration-1-to-2.log:
```
[ADDITION] Migrated settings to version 2

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```

migration-2-to-3.log:
```
[CHANGE] Migrated settings to version 3

[SPECIAL] Property 'homeassistant' is now always an object.

[SPECIAL] Property 'frontend' is now always an object.

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```

migration-3-to-4.log:
```
[CHANGE] Migrated settings to version 4

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```

Main log repeats after a few times then stops.

### What did you expect to happen?

Normal loading of addon.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

Unable to obtain as addon can't start.

### Adapter

deconz

### Setup

Home Assistant 'hassio' on Pi4 as an addon.

cmdshft, 2025-01-04:
Post your z2m YAML

talondnb, 2025-01-04:
> Post your z2m YAML

```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://homeassistant:1883
  user: "!secret mqtt_user"
  password: "!secret mqtt_pass"
serial:
  port: /dev/ttyUSB0
  adapter: deconz
  baudrate: 115200
experimental:
  new_api: false
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

Der-WU, 2025-01-04:
Try to remove the 

legacy: false

line.
By me it caused zigbee2mqtt won´t start.

z-master42, 2025-01-04:
After the upgrade to 2.0.0 there should no longer be any legacy flags in the config. In my case they were removed automatically.

Nerivec, 2025-01-04:
The yaml configuration you are showing seems to be from HA addon config page?
The settings available in that page are [limited](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/15b7393e7648f5256cd5ce0228ea858b2509ecf2/zigbee2mqtt/config.json#L49-L60) compared to what is available in the actual file. Most of these should effectively have no effect at all:
```yaml
experimental:
  new_api: false
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
```
Note: The _valid_ settings in the addon configuration page, if set, take precedence over the yaml file.
https://www.zigbee2mqtt.io/guide/configuration/configuration-update.html

Can you post the actual configuration.yaml file?
Can you also re-try with `log_level: debug` so we can get more details (your logs only show an abrupt stop, no error)?

itsaints, 2025-01-04:
I have exactly the same issue with same logs config files though using Combee II in PI 5. With debug on I get:- 

[2025-01-04 15:46:48] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.14-35-42'
[2025-01-04 15:46:48] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 15:46:48] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 15:46:48] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 15:46:48] debug: 	z2m: sd-notify loaded
[2025-01-04 15:46:48] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 15:46:48] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":1074,\"extendedPanID\":[90,171,16,156,125,85,204,156],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2459153-if00\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-04 15:46:48] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2459153","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2459153-if00","vendorId":"1cf1","productId":"0030"}]

I assume it crashes at that point. As a combee II I have not set the baudrate in config as per documentation though.

itsaints, 2025-01-04:
My z2m config is :-

`homeassistant:
  enabled: true
advanced:
  network_key:
    - 142
    - 82
    - 171
    - 226
    - 164
    - 64
    - 20
    - 41
    - 4
    - 245
    - 11
    - 249
    - 18
    - 42
    - 235
    - 130
  pan_id: 1074
  ext_pan_id:
    - 90
    - 171
    - 16
    - 156
    - 125
    - 85
    - 204
    - 156
  log_level: debug
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: "!secret mosq_username"
  password: "!secret mosq_password"

serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  enabled: true
  port: 8099
devices:
  '0x680ae2fffe3f176c':
    friendly_name: Repeater 1 - Upper Hall
  '0x00158d0008321ba4':
    friendly_name: Postbox Sensor
  '0x00158d000894baa3':
    friendly_name: Ensuite Shower Door
  '0xccccccfffe893214':
    friendly_name: Family Side Light
  '0x5c0272fffef5bee0':
    friendly_name: Parent Left Light
  '0x804b50fffe421c45':
    friendly_name: Parent Right Light
  '0x14b457fffec69662':
    friendly_name: Hall Light 3
  '0x680ae2fffe80c0ff':
    friendly_name: Hall Light 4
  '0x680ae2fffe32dfc7':
    friendly_name: Hall Light 1
  '0x680ae2fffe83a58e':
    friendly_name: Hall Light 2
  '0xcc86ecfffed842ed':
    friendly_name: Ensuite Left Light
  '0x84fd27fffeae609a':
    friendly_name: Ensuite Right Light
  '0x842e14fffea02c0f':
    friendly_name: Parent Side Light
  '0x000b57fffed51447':
    friendly_name: Outside Lower Light
  '0x90fd9ffffe13d351':
    friendly_name: Outside Upper Light
  '0xccccccfffe8f803b':
    friendly_name: Family Side Light Switch
  '0x842e14fffeb2b274':
    friendly_name: Parents Side Light Switch
  '0x54ef441000781cda':
    friendly_name: SH Blind Center
  '0x54ef441000781f98':
    friendly_name: SH Blind Right
  '0x54ef4410006bc17e':
    friendly_name: SH Blind Left
  '0x54ef4410006bc9a1':
    friendly_name: SH Blind Side
  '0x00158d0007c58100':
    friendly_name: Sam Window Sensor
  '0x00158d0007c5b8e4':
    friendly_name: Joes Window Sensor
  '0x588e81fffe5edfce':
    friendly_name: Sam Right Light
  '0x588e81fffe421113':
    friendly_name: Sam Left Light
  '0x680ae2fffea7f53a':
    friendly_name: Joe Main Light
  '0x84ba20fffe34bac5':
    friendly_name: Parents Wall Switch
  '0x14b457fffe6713d3':
    friendly_name: Joe Main Light Switch
  '0x3425b4fffe55f858':
    friendly_name: SH Motion Sensor
  '0x680ae2fffe6ab2c7':
    friendly_name: Parent Blind Switch
  '0xd0cf5efffe4399cc':
    friendly_name: Repeater 2 - Fam Room
  '0x000b3cfffef6c8e9':
    friendly_name: Amp Power
  '0x000d6ffffe2ba11c':
    friendly_name: HATablet Power
  '0x842e14fffe68615c':
    friendly_name: Livingroom Patio Left
  '0x588e81fffe59219e':
    friendly_name: Livingroom Patio Right
  '0x000b57fffe3130db':
    friendly_name: Livingroom Light Stand
  '0x680ae2fffe4322c2':
    friendly_name: Kitchen Lights 2
  '0x680ae2fffe3fb8bd':
    friendly_name: Kitchen Lights 1
  '0x680ae2fffe38c6b7':
    friendly_name: Parent Blind
  '0xcc86ecfffe7627f5':
    friendly_name: Livingroom Air Purifier
  '0x001788010248de73':
    friendly_name: SH Spotlight
  '0x00158d000832d316':
    friendly_name: Electric Gate Sensor
  '0x003c84fffe14f44a':
    friendly_name: xmas star lamp
  '0x00124b002a575921':
    friendly_name: SH Temp Sensor
  '0x680ae2fffe30c384':
    friendly_name: Kitchen Motion Sensor
  '0x84ba20fffec509b7':
    friendly_name: xmas star lamp 2
groups:
  '1':
    friendly_name: Family Side Light Group
  '2':
    friendly_name: Kitchen Lights
  '3':
    friendly_name: Parent Side Lights
device_options:
  legacy: false
version: 4
`

z-master42, 2025-01-04:
There is no error in your log.

itsaints, 2025-01-04:
My z2m config is :-
```
homeassistant:
  enabled: true
advanced:
  network_key:
    - 142
    - 82
    - 171
    - 226
    - 164
    - 64
    - 20
    - 41
    - 4
    - 245
    - 11
    - 249
    - 18
    - 42
    - 235
    - 130
  pan_id: 1074
  ext_pan_id:
    - 90
    - 171
    - 16
    - 156
    - 125
    - 85
    - 204
    - 156
  log_level: debug
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: "!secret mosq_username"
  password: "!secret mosq_password"

serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  enabled: true
  port: 8099
devices:
  '0x680ae2fffe3f176c':
    friendly_name: Repeater 1 - Upper Hall
  '0x00158d0008321ba4':
    friendly_name: Postbox Sensor
  '0x00158d000894baa3':
    friendly_name: Ensuite Shower Door
  '0xccccccfffe893214':
    friendly_name: Family Side Light
  '0x5c0272fffef5bee0':
    friendly_name: Parent Left Light
  '0x804b50fffe421c45':
    friendly_name: Parent Right Light
  '0x14b457fffec69662':
    friendly_name: Hall Light 3
  '0x680ae2fffe80c0ff':
    friendly_name: Hall Light 4
  '0x680ae2fffe32dfc7':
    friendly_name: Hall Light 1
  '0x680ae2fffe83a58e':
    friendly_name: Hall Light 2
  '0xcc86ecfffed842ed':
    friendly_name: Ensuite Left Light
  '0x84fd27fffeae609a':
    friendly_name: Ensuite Right Light
  '0x842e14fffea02c0f':
    friendly_name: Parent Side Light
  '0x000b57fffed51447':
    friendly_name: Outside Lower Light
  '0x90fd9ffffe13d351':
    friendly_name: Outside Upper Light
  '0xccccccfffe8f803b':
    friendly_name: Family Side Light Switch
  '0x842e14fffeb2b274':
    friendly_name: Parents Side Light Switch
  '0x54ef441000781cda':
    friendly_name: SH Blind Center
  '0x54ef441000781f98':
    friendly_name: SH Blind Right
  '0x54ef4410006bc17e':
    friendly_name: SH Blind Left
  '0x54ef4410006bc9a1':
    friendly_name: SH Blind Side
  '0x00158d0007c58100':
    friendly_name: Sam Window Sensor
  '0x00158d0007c5b8e4':
    friendly_name: Joes Window Sensor
  '0x588e81fffe5edfce':
    friendly_name: Sam Right Light
  '0x588e81fffe421113':
    friendly_name: Sam Left Light
  '0x680ae2fffea7f53a':
    friendly_name: Joe Main Light
  '0x84ba20fffe34bac5':
    friendly_name: Parents Wall Switch
  '0x14b457fffe6713d3':
    friendly_name: Joe Main Light Switch
  '0x3425b4fffe55f858':
    friendly_name: SH Motion Sensor
  '0x680ae2fffe6ab2c7':
    friendly_name: Parent Blind Switch
  '0xd0cf5efffe4399cc':
    friendly_name: Repeater 2 - Fam Room
  '0x000b3cfffef6c8e9':
    friendly_name: Amp Power
  '0x000d6ffffe2ba11c':
    friendly_name: HATablet Power
  '0x842e14fffe68615c':
    friendly_name: Livingroom Patio Left
  '0x588e81fffe59219e':
    friendly_name: Livingroom Patio Right
  '0x000b57fffe3130db':
    friendly_name: Livingroom Light Stand
  '0x680ae2fffe4322c2':
    friendly_name: Kitchen Lights 2
  '0x680ae2fffe3fb8bd':
    friendly_name: Kitchen Lights 1
  '0x680ae2fffe38c6b7':
    friendly_name: Parent Blind
  '0xcc86ecfffe7627f5':
    friendly_name: Livingroom Air Purifier
  '0x001788010248de73':
    friendly_name: SH Spotlight
  '0x00158d000832d316':
    friendly_name: Electric Gate Sensor
  '0x003c84fffe14f44a':
    friendly_name: xmas star lamp
  '0x00124b002a575921':
    friendly_name: SH Temp Sensor
  '0x680ae2fffe30c384':
    friendly_name: Kitchen Motion Sensor
  '0x84ba20fffec509b7':
    friendly_name: xmas star lamp 2
groups:
  '1':
    friendly_name: Family Side Light Group
  '2':
    friendly_name: Kitchen Lights
  '3':
    friendly_name: Parent Side Lights
device_options:
  legacy: false
version: 4
```

itsaints, 2025-01-04:
> There is no error in your log.

Yeah annoying isn't it :-) That's the last log entry and the addon is no longer running. What else can I do to help debug the issue ?

itsaints, 2025-01-04:
My addon yaml is:-

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: "!secret mosq_username"
  password: "!secret mosq_password"
serial:
  port: /dev/ttyACM0
  adapter: deconz
```

z-master42, 2025-01-04:
Where do you see that your addon is not running? Have you ever left the addon page?

itsaints, 2025-01-04:
If I go to the menu for z2m I get this :- 

[image]

itsaints, 2025-01-04:
Also this repair task after a reboot:- 

[image]

gecastan, 2025-01-05:
I have the same problem with Deconz Conbee 2 on RPI 4
Auto update to 2.0.0 broke my config and the container just stops after starting booting up .

I tried to reinstall from fresh zigbee2mqtt install (removed the addon, removed the zigbee2mqtt folder and start again)

I get : 
`[10:14:39] INFO: Preparing to start...
[10:14:40] INFO: Socat not enabled
[10:14:43] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 10:14:49] info:     z2m: Logging to console, file (filename: log.log)
[2025-01-05 10:14:49] debug:    z2m: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
[2025-01-05 10:14:49] info:     z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 10:14:49] debug:    z2m: sd-notify loaded
[2025-01-05 10:14:49] info:     z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 10:14:49] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":46837,\"extendedPanID\":[187,225,154,126,165,217,184,104],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 10:14:49] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE24XXXX","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436551-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyS0"}]
[2025-01-05 10:14:49] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE24XXXX","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436551-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
[2025-01-05 10:14:50] debug:    zh:deconz:driver: Set delay to 0
[2025-01-05 10:14:50] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":46837,"extendedPanID":[187,225,154,126,165,217,184,104],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 10:14:50] debug:    zh:deconz:driver: Opening with /dev/ttyACM0`

Config file auto generated from HA Zigbee2MQTT Module page (but showing the true configuration.yaml file with all options)

`version: 4
homeassistant:
  enabled: true
advanced:
  log_level: debug
  network_key:
    - 80
    - 196
    - 237
    - 176
    - 113
    - 194
    - 253
    - 209
    - 168
    - 166
    - 63
    - 219
    - 138
    - 146
    - 31
    - 180
  pan_id: 46837
  ext_pan_id:
    - 187
    - 225
    - 154
    - 126
    - 165
    - 217
    - 184
    - 104
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: XXXX
  password: XXXXXXX
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  enabled: true
  port: 8099
`

I only added debug log_level to the config

No error shown in any log

talondnb, 2025-01-05:
> Can you check the HA system logs to see if there is any indication of what is happening around the time of the crash?

That's the thing, there are no errors in the HA system logs for this. Nor anything useful in the addon logs to indicate why the addon isn't loading up, even with debug enabled.

itsaints, 2025-01-05:
I restored from backup and all working ok again. Attaching the current log on 1.42 with the legacy settings on in case it highlights anything to the trained eye that I can't see.

[log.log](https://github.com/user-attachments/files/18310377/log.log)

Koenkk, 2025-01-05:
This should be fixed with the addon v2.0.0-2 (same as https://github.com/Koenkk/zigbee2mqtt/issues/25464)

closed 2025-01-05 by Koenkk