Koenkk/zigbee2mqtt #25466: Zigbee2MQTT has suddenly stopped connecting to my Zigbee coordinator

DaveTSG, 2025-01-04:
### What happened?

Zigbee2MQTT add-on fails to start with "Error: Adapter 'auto' does not exists" in log. It has worked perfectly fine for the last year or so. I haven't changed anything relating to the Z2M add-on.

I am also using the "Mosquitto broker" Home Assistant add-on (again, not HACS), and that is working fine (I also have Frigate running on another computer, sending data to the Mosquitto broker add-on, so I can be confident that's not the issue).

Any suggestions on a solution please?
And yes, I have checked the URL provided in the logs. It doesn't have any obvious solution...

### What did you expect to happen?

Z2M add-on to start.

### How to reproduce it (minimal and precise)

I haven't changed any settings in the Z2M add-on, but here they are as of now:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://192.168.3.15:6638
```

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

Can't access it

### Adapter

TubesZB CC2652P2 Based Zigbee to PoE Coordinator 2023

### Setup

Add-on on Home Assistant Yellow

### Debug log

[17:08:47] INFO: Preparing to start...
[17:08:47] INFO: Socat not enabled
[17:08:47] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 17:08:53] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.16-57-44'
[2025-01-04 17:08:53] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 17:08:53] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 17:08:53] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 17:08:53] debug: 	z2m: sd-notify loaded
[2025-01-04 17:08:53] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 17:08:53] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":17504,\"extendedPanID\":[114,89,105,13,62,213,25,198],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://192.168.3.15:6638\",\"adapter\":\"auto\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-04 17:08:53] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 17:08:53] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 17:08:53] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 17:08:53] error: 	z2m: Exiting...
[2025-01-04 17:08:53] error: 	z2m: Error: Adapter 'auto' does not exists, possible options: deconz, ember, ezsp, zstack, zboss, zigate
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:80:19)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

DaveTSG, 2025-01-04:
Ah, I found a discussion about the update to v2. Apparently there was a warning message in the previous version explaining what steps would be required before updating to v2, but I never saw that, as Home Assistant updated the add-on automatically and I had no idea it was even an issue.

In any case, I had to add in one line into the configuration:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://192.168.3.15:6638
  adapter: zstack
```

(The _adapter: zstack_ line, as per [this documentation](https://www.zigbee2mqtt.io/guide/adapters/).)

All working now. Thanks.

closed 2025-01-04 by DaveTSG

Stefan300381, 2025-01-04:
thanks, faced the same issue. my docker container was auto updated to the latest version and then z2m stopped working :-)

jwall87, 2025-01-04:
Thanks, same problem here. 

Added the one line as mentioned.
serial:
  port: tcp://192.168.3.15:6638
  **adapter: zstack**

Note you do have to confirm what adapter your Zigbee dongle uses here - https://www.zigbee2mqtt.io/guide/adapters/