Koenkk/zigbee2mqtt #25467: Does not start after update 2.0.0

thueske, 2025-01-04:
### What happened?

```
Refusing to start because configuration is not valid, found the following errors:
- homeassistant must be object
- availability must be object
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

-

### Adapter

CONBEE II

### Setup

Docker

### Debug log

```
homeassistant:
  enabled: true
frontend:
  enabled: true
  port: 8080
device_options:
  legacy: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://broker:1883
serial:
  port: /dev/ttyACM0
  adapter: deconz
devices:
  '0x60a423fffef82db4':
    friendly_name: Nachttischschalter Larissa
    legacy: false
    icon: device_icons/7146bc3333829dd31b1a37b1f83f6e01.png
  '0x0017880108ca46ed':
    friendly_name: Flurlampe (Hinten)
    icon: device_icons/4f622d30f1367a043961de0d6632676d.png
  '0x0017880108ca3d09':
    friendly_name: Flurlampe (Vorne)
    icon: device_icons/4f622d30f1367a043961de0d6632676d.png
  '0x00158d00063db2e9':
    friendly_name: Flurbewegungsmelder (Hinten)
    icon: device_icons/2e7485d54446698ed62425812fe12303.png
  '0x00158d00046554ce':
    friendly_name: Abstellraumtürsensor
    icon: device_icons/95eff091034c37f5a8e34e03c82cff29.png
  '0x00158d00044ea422':
    friendly_name: Wohnungstürsensor
    icon: device_icons/95eff091034c37f5a8e34e03c82cff29.png
  '0x00158d0005bd56a3':
    friendly_name: Flurbewegungsmelder (Vorne)
    icon: device_icons/2e7485d54446698ed62425812fe12303.png
  '0x00158d000633621e':
    friendly_name: Badezimmerbewegungsmelder
    no_occupancy_since: []
    icon: device_icons/2e7485d54446698ed62425812fe12303.png
  '0x00158d000449a2ab':
    friendly_name: Wohnzimmertemperatursensor
    icon: device_icons/8da97afa2d57d9d2e86756a4aa2a6072.png
  '0x00158d0006b63438':
    friendly_name: Waschmaschinensensor
    icon: device_icons/2d19c1b8e82d9b72921bbd1fcbde4541.png
  '0x00158d00047706eb':
    friendly_name: Wohnzimmerdeckenlampe
    icon: device_icons/9b3d5ce25b5583be19620c7b236e01db.png
  '0x5c0272fffe3e63da':
    friendly_name: Esszimmerschalter
    icon: device_icons/7146bc3333829dd31b1a37b1f83f6e01.png
  '0x001788010b81e0ec':
    friendly_name: Esszimmerlampe (4)
    icon: device_icons/726d27967c7389b823e559d3c34b5d0e.png
  '0x001788010b82309c':
    friendly_name: Esszimmerlampe (1)
    icon: device_icons/726d27967c7389b823e559d3c34b5d0e.png
  '0x001788010b08671b':
    friendly_name: Esszimmerlampe (3)
    icon: device_icons/726d27967c7389b823e559d3c34b5d0e.png
  '0x001788010b823a18':
    friendly_name: Esszimmerlampe (2)
    icon: device_icons/726d27967c7389b823e559d3c34b5d0e.png
  '0x54ef441000035a16':
    friendly_name: Lichtsensor
    icon: device_icons/956bdca587d0b5a2cd6200ffdcb3e3d1.png
  '0x001788010b856bda':
    friendly_name: Regallampe
    icon: device_icons/726d27967c7389b823e559d3c34b5d0e.png
  '0x001788010b849658':
    friendly_name: Betonlampe
    icon: device_icons/5fecacbceb69a0bf5d30772d9504680b.png
  '0x00158d00044a1ed9':
    friendly_name: Schlafzimmertemperatursensor
    icon: device_icons/8da97afa2d57d9d2e86756a4aa2a6072.png
  '0x00158d000533ccea':
    friendly_name: Badezimmertemperatursensor
    icon: device_icons/8da97afa2d57d9d2e86756a4aa2a6072.png
  '0x00158d0004496060':
    friendly_name: Bürotemperatursensor
    icon: device_icons/8da97afa2d57d9d2e86756a4aa2a6072.png
  '0x842e14fffe155260':
    friendly_name: Schreibtischschalter
    icon: device_icons/7146bc3333829dd31b1a37b1f83f6e01.png
  '0xe0798dfffeb9176e':
    friendly_name: Schreibtischsteckdose
    icon: device_icons/08a9016bbc0657cf5f581ae9c19c31a5.png
  '0xe0798dfffeb91738':
    friendly_name: Balkonlichterkette
    measurement_poll_interval: 30
    icon: device_icons/08a9016bbc0657cf5f581ae9c19c31a5.png
  '0xe0798dfffeb90db6':
    friendly_name: Weihnachtsbaum
    icon: device_icons/08a9016bbc0657cf5f581ae9c19c31a5.png
  '0x001788010d30a151':
    friendly_name: Wohnzimmerstehlampe
    icon: device_icons/a81e0ced22a229e63816d1d857487988.png
  '0x001788010d30b3b3':
    friendly_name: Küchenlampe
    icon: device_icons/a81e0ced22a229e63816d1d857487988.png
  '0xa4c1381a80ca588a':
    friendly_name: Esszimmerbewegungsmelder
    icon: device_icons/59552f193e6d33d9679c39dc74313ce6.png
  '0xa4c1388ea63cc169':
    friendly_name: Badezimmerfenstersensor
    icon: device_icons/67d8a38ea355da6e701908018fd64f87.png
  '0xa4c13837705a0712':
    friendly_name: Balkontürsensor
    icon: device_icons/67d8a38ea355da6e701908018fd64f87.png
  '0x70ac08fffe72b2bb':
    friendly_name: Autoklimatisierung
    icon: device_icons/caccf240693b810cc1b9ef99b41aea54.png
  '0x001788010c8eab0d':
    friendly_name: Flurlampe (3)
    icon: device_icons/4f622d30f1367a043961de0d6632676d.png
  '0x001788010c958a2d':
    friendly_name: Flurlampe (4)
    icon: device_icons/4f622d30f1367a043961de0d6632676d.png
  '0x001788010c34bd84':
    friendly_name: Schlafzimmerstehlampe
    icon: device_icons/f951d495ed40ce8ace4861e502327720.png
  '0x5c0272fffe57ab7c':
    friendly_name: Nachttischschalter Tobi
    icon: device_icons/7146bc3333829dd31b1a37b1f83f6e01.png
  '0x842e14fffe2c4d65':
    friendly_name: Bürofensterlampe
    icon: device_icons/fc0b006315bcd935d334e97735f8bf71.png
  '0x60a423fffe8e72ef':
    friendly_name: Laterne
    icon: device_icons/fc0b006315bcd935d334e97735f8bf71.png
  '0x60a423fffe8e7326':
    friendly_name: Weihnachtsstern
    icon: device_icons/fc0b006315bcd935d334e97735f8bf71.png
advanced:
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  log_level: warn
  log_output: []
  adapter_delay: 50
  last_seen: ISO_8601
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
groups:
  '1':
    friendly_name: Flurlampen
  '2':
    friendly_name: Esszimmerlampen
  '3':
    friendly_name: Wohn- und Esszimmerlampen
  '4':
    friendly_name: Alle Zigbee-Lampen
blocklist: []
availability:
  enabled: true
version: 4
```

thebatfink, 2025-01-04:
Add your logs like it asks, not your interpretation of them.

Der-WU, 2025-01-04:
for test:

remove   legacy: false line

without mine z2mqtt starts

thueske, 2025-01-04:
> Add your logs like it asks, not your interpretation of them.

[image]

I already did in the first post ;-)

thueske, 2025-01-04:
> for test:
> 
> remove legacy: false line
> 
> without mine z2mqtt starts

Does not work.

z-master42, 2025-01-04:
Do you have migration logs in your Z2M directory?

thueske, 2025-01-04:
> Do you have migration logs in your Z2M directory?

**migration-1-to-2.log**

```
[CHANGE] Log level 'warn' has been renamed to 'warning'.

[ADDITION] Migrated settings to version 2

[REMOVAL] Action and click sensors have been removed (homeassistant.legacy_triggers setting). This means all sensor.*_action and sensor.*_click entities are removed. Use the MQTT device trigger instead.

[REMOVAL] The permit_join setting has been removed, use the frontend or MQTT to permit joining.

[REMOVAL] Due to the OTA rework, the ota.ikea_ota_use_test_url option has been removed.

[REMOVAL] Removed configuring group members through configuration.yaml (groups.xyz.devices setting). This will not impact current group members; however, you will no longer be able to add or remove devices from a group through the configuration.yaml.
```

**migration-2-to-3.log**

```
[CHANGE] Migrated settings to version 3

[SPECIAL] Property 'homeassistant' is now always an object.

[SPECIAL] Property 'frontend' is now always an object.

[SPECIAL] Property 'availability' is now always an object.
```

**migration-3-to-4.log**

```
[CHANGE] Migrated settings to version 4

[SPECIAL] Device icons are now saved as images.
```

z-master42, 2025-01-04:
> **migration-1-to-2.log**
>
> ```
> [CHANGE] Log level 'warn' has been renamed to 'warning'.
>
> [ADDITION] Migrated settings to version 2
>
> [REMOVAL] Action and click sensors have been removed [...]

Locks good so far.

shadowpanther, 2025-01-04:
I have the same issue here.

Migration logs:
```
# more migration-1-to-2.log 
[ADDITION] Migrated settings to version 2

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

# more migration-2-to-3.log 
[CHANGE] Migrated settings to version 3

[SPECIAL] Property 'homeassistant' is now always an object.

[SPECIAL] Property 'frontend' is now always an object.

[SPECIAL] Property 'availability' is now always an object.

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

# more migration-3-to-4.log 
[CHANGE] Migrated settings to version 4

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```

```
# cat configuration.yaml
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: password
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - (HIDDEN)
  pan_id: 45894
  ext_pan_id:
    - (HIDDEN)
  log_level: warning
device_options: {}
availability:
  enabled: true
devices:
 (LIST OF DEVICES)
version: 4
homeassistant:
  enabled: true
```

I've added these last two lines and removed the legacy `homeassistant: true` in an effort to fix the config, but it didn't help
```
homeassistant:
  enabled: true
```

It also doesn't start without these two and without the legacy one (no `homeassistant` entries at all).

z-master42, 2025-01-04:
Remove the homeassistant... completely.

shadowpanther, 2025-01-04:
Judging by the automatic config backups, the `homeassistant` entry was made into an object with v2 to v3 migration, but reverted back to a single-line entry in v3 to v4. Still, neither of the variants (or lack of the entry altogether) work properly.

> Remove the homeassistant... completely.

Doesn't help at all, same error from z2m.

m0wlheld, 2025-01-04:
Okay, this might help: 

I was facing the same issue on one of two nodes. The second, now failing node uses my own "quickstart" project at https://github.com/m0wlheld/homeassistant-zigbee2mqtt/

Turned out, the environment variable `ZIGBEE2MQTT_CONFIG_HOMEASSISTANT=true` as mentioned in the docker compose's `compose.yaml` file injects "homeassistant: true" into the generated `configuration.yaml` . This lead to an invalid configuration.yaml every time I used docker compose to start "the stack".

After removing the appropriate line from the compose.file, Z2M 2.0.0 now starts.

bart1, 2025-01-04:
I'm quickly tagging along in this issue as my docker setup seems to fail with a somewhat similar issue giving the message
```
- frontend must be object
```
Playing around with the configuration file does not seem to resolve it

shadowpanther, 2025-01-04:
> Turned out, the environment variable `ZIGBEE2MQTT_CONFIG_HOMEASSISTANT=true` as mentioned in the docker compose's `compose.yaml` file injects "homeassistant: true" into the generated [...]

That was exactly the case, thank you! My configuration is in Kubernetes, but it's the same principle.

bart1, 2025-01-04:
I had a similar issue with `ZIGBEE2MQTT_CONFIG_FRONTEND=true` in my compose file, thanks!

thueske, 2025-01-04:
> Okay, this might help:
>
> I was facing the same issue on one of two nodes. The second, now failing node uses my own "quickstart" project at https://github.com/m0wlheld/homeassistant-zigbee2mqtt/ [...]

omg. fixed my issue. one instance is running 👏🏻 another having some other errors:

```
bridge-1 exited with code 0
bridge-1  | 2025-01-04T19:08:37.245884390Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:08:44.140025196Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:08:51.116657257Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:08:58.528443126Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:09:06.526969702Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:09:16.157476541Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:09:29.073147502Z Starting Zigbee2MQTT without watchdog.
bridge-1 exited with code 1
bridge-1  | 2025-01-04T19:09:48.400749334Z Starting Zigbee2MQTT without watchdog.
```

Koenkk, 2025-01-05:
As @bart1 mentioned, removing the env vars should fix your issue.

closed 2025-01-05 by Koenkk

marekorok, 2025-01-05:
> Okay, this might help:
>
> I was facing the same issue on one of two nodes. The second, now failing node uses my own "quickstart" project at https://github.com/m0wlheld/homeassistant-zigbee2mqtt/ [...]

That was my case as well, compose config lines preventing new z2m to start
[image]

Thank you for the tip

Hadatko, 2025-01-18:
It would be great to mentioned that in https://github.com/Koenkk/zigbee2mqtt/discussions/24198 or fix behavior :/ But anyway thank you

tzago, 2025-03-31:
> That was exactly the case, thank you! My configuration is in Kubernetes, but it's the same principle.

same here thanks!