Koenkk/zigbee2mqtt #25468: Installation on macOS fails after 2.0.0

dawiinci, 2025-01-04:
### What happened?

Installation fails.

### What did you expect to happen?

Installation works as before.

### How to reproduce it (minimal and precise)

https://kinori.tech/blog/en/2021/11/19/install-and-run-zigbee2mqtt-on-your-mac/

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

-

### Adapter

-

### Setup

macOS venv

### Debug log

```
Successfully installed nodeenv-1.9.1
 * Install prebuilt node (22.12.0) ..... done.
 * Appending data to /Users/server/Steuerung/zigbee2mqtt/bin/activate
 * Appending data to /Users/server/Steuerung/zigbee2mqtt/bin/activate.fish
npm error code EUSAGE
npm error
npm error The `npm ci` command can only install with an existing package-lock.json or
npm error npm-shrinkwrap.json with lockfileVersion >= 1. Run an install with npm@5 or
npm error later to generate a package-lock.json file, then try again.
npm error
npm error Clean install a project
npm error
npm error Usage:
npm error npm ci
npm error
npm error Options:
npm error [--install-strategy <hoisted|nested|shallow|linked>] [--legacy-bundling]
npm error [--global-style] [--omit <dev|optional|peer> [--omit <dev|optional|peer> ...]]
npm error [--include <prod|dev|optional|peer> [--include <prod|dev|optional|peer> ...]]
npm error [--strict-peer-deps] [--foreground-scripts] [--ignore-scripts] [--no-audit]
npm error [--no-bin-links] [--no-fund] [--dry-run]
npm error [-w|--workspace <workspace-name> [-w|--workspace <workspace-name> ...]]
npm error [-ws|--workspaces] [--include-workspace-root] [--install-links]
npm error
npm error aliases: clean-install, ic, install-clean, isntall-clean
npm error
npm error Run "npm help ci" for more info
npm error A complete log of this run can be found in: /Users/server/.npm/_logs/2025-01-04T07_18_00_781Z-debug-0.log
```

mentioned log not found...

m0wlheld, 2025-01-04:
> pnpm is now used instead of npm, the [installation](https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/installation/01_linux.md#installing) instructions and update.sh script has been updated accordingly. For existing installations you need to execute npm install -g pnpm before updating.

See https://github.com/Koenkk/zigbee2mqtt/discussions/24198

dawiinci, 2025-01-04:
I did install that, but still it failed.

82sevo, 2025-01-04:
I have the same issue. Even with a fresh install --not updating an existing-- I get the same error message.

Koenkk, 2025-01-05:
Instead of `npm ci` you have to use `pnpm install --frozen-lockfile` (https://kinori.tech/blog/en/2021/11/19/install-and-run-zigbee2mqtt-on-your-mac/ is outdated)

dawiinci, 2025-01-05:
Thanks for the clarification. It works.

closed 2025-01-05 by dawiinci