Koenkk/zigbee2mqtt #25469: Problem to Zigbee2MQTT Update

xnetworker, 2025-01-04:
### What happened?

After the update from Zigbee2MQTT to 2.0.0-1, Zigbee2MQTT no longer works.

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-On on Home Assistant on MiniPC

### Debug log

`[08:05:30] INFO: Preparing to start...
[08:05:30] INFO: Socat not enabled
[08:05:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 08:05:32] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 08:05:32] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 08:05:32] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 08:05:32] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 08:05:32] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 08:05:32] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 08:05:32] error: 	z2m: Exiting...
[2025-01-04 08:05:32] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
`

K3vb3rt, 2025-01-04:
+1 
Same problem with 2.0.0-1 and HA 2025.1.0

> [2025-01-04 09:01:56] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 09:01:56] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 09:01:56] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 09:01:56] error: 	z2m: Exiting...
[2025-01-04 09:01:56] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

goldrapper19872504, 2025-01-04:
> +1 Gleiches Problem mit 2.0.0-1 und HA 2025.1.0. Bin gerade zu meinem Backup 1.42.0-2 zurückgekehrt; funktioniert im Moment. Warte auf die Lösung, danke!
> 
> > [2025-01-04 09:01:56] Fehler: z2m: Fehler beim Starten von zigbee-herdsman 
> > [2025-01-04 09:01:56] Fehler: z2m: Starten von zigbee-herdsman fehlgeschlagen 
> > [2025-01-04 09:01:56] Fehler: z2m: Prüfen Sie https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html auf mögliche Lösungen 
> > [2025-01-04 09:01:56] Fehler: z2m: Beenden... 
> > [2025-01-04 09:01:56] Fehler: z2m: Fehler: TCP-Adapter können derzeit nicht erkannt werden. Geben Sie in Ihrer Konfiguration einen gültigen „Adapter“ und „Port“ an. 
> > bei findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15) 
> > bei discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26) 
> > bei Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54) 
> > bei Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38) 
> > bei Zigbee.start (/app/lib/zigbee.ts:69:47) 
> > bei Controller.start (/app/lib/controller.ts:142:13) 
> > beim Start (/app/index.js:161:5)

Wie genau hast du das Backup eingespielt? Habe auch die Probleme möchte auch wieder zurück! Danke

frittenlab, 2025-01-04:
You need to specify a valid adapter for your setup to work:

[upgrade guide](https://github.com/Koenkk/zigbee2mqtt/discussions/24364)

Try adding this line to your serial config:
 
adapter: zstack # <-- Add this line

xnetworker, 2025-01-04:
Ich habe es direkt über HA/Backups eingespielt. Vor dem Update wurde ein Backup angelegt

xnetworker, 2025-01-04:
> You need to specify a valid adapter for your setup to work:
> 
> [upgrade guide](https://github.com/Koenkk/zigbee2mqtt/discussions/24364)
> 
> Try adding this line to your serial config:
> 
> adapter: zstack # <-- Add this line

Where's the Serial Config?

frittenlab, 2025-01-04:
https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/configuration/adapter-settings.md#basic-configuration

K3vb3rt, 2025-01-04:
> You need to specify a valid adapter for your setup to work:
> 
> [upgrade guide](https://github.com/Koenkk/zigbee2mqtt/discussions/24364)
> 
> Try adding this line to your serial config:
> 
> adapter: zstack # <-- Add this line

That did the trick for me, too. Thanks! Now on 2.0.0-1!

xnetworker, 2025-01-04:
> https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/configuration/adapter-settings.md#basic-configuration

The SLZB-06 is connected via POE in the network and not with USB.

goldrapper19872504, 2025-01-04:
Bei mir gehen die Automationen nicht mehr bei Philips Hue Lampen in der 2er Version

xnetworker, 2025-01-04:
[image]

Eintrag "adapter: zstack" ergänzt und es läuft wieder

closed 2025-01-04 by xnetworker

bigbeka, 2025-01-05:
> You need to specify a valid adapter for your setup to work:
> 
> [upgrade guide](https://github.com/Koenkk/zigbee2mqtt/discussions/24364)
> 
> Try adding this line to your serial config:
> 
> adapter: zstack # <-- Add this line

@frittenlab life saver! This worked for me too. 

I am using SMLIGHT SLZM-06 via LAN PoE and adding `adapter: zstack` to `serial` part of the Z2M Add-on `Configuration` solved the problem!

It might help someone else: I was getting the following error without the `adapter: zstack` part:

```
[13:30:22] INFO: Preparing to start...
[13:30:22] INFO: Socat not enabled
[13:30:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 13:30:27] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 13:30:27] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 13:30:27] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 13:30:27] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 13:30:27] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 13:30:27] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 13:30:27] error: 	z2m: Exiting...
[2025-01-05 13:30:27] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

brandon-dacrib, 2025-01-05:
+1 for the tubeszb-cc2652-poe-2022

zeus86, 2025-01-05:
the `zstack`-entry fixed it for me as well. i just wonder: why? so what was the problem in the first place?

richieh92, 2025-01-06:
> @frittenlab life saver! This worked for me too.
>
> I am using SMLIGHT SLZM-06 via LAN PoE and adding `adapter: zstack` to `serial` part of the Z2M Add-on `Configuration` solved the problem!
>
> It might [...]

Life saver, thank you! Worked for me too!

richieh92, 2025-01-06:
> You need to specify a valid adapter for your setup to work:
> 
> [upgrade guide](https://github.com/Koenkk/zigbee2mqtt/discussions/24364)
> 
> Try adding this line to your serial config:
> 
> adapter: zstack # <-- Add this line

Honestly, this problem was starting to stress me out. What a life saver, adding `adapter: zstack` in the serial section resolved this problem for me. Thank you!

elco1965, 2025-01-06:
This worked for me as well. Running Zigbee2MQTT Docker under unRAID. Changed \appdata\zigbee2mqtt\configuration.yaml  to include : adapter: zstack

serial:
  adapter: zstack
  port: tcp://x.x.x.x:6638
  baudrate: xxxxxx
  disable_led: false

Thanks all

Smith-Happens, 2025-01-07:
Yes! Adding in "adapter: zstack" worked for me. I had to add this to the configuration on the plugin page.

djurny, 2025-01-07:
> I added `adapter: zstack` but it's not working for me
> 
> ```yaml
> serial:
>   adapter: zstack
>   port: >-
>     /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123170056-if00
> ```
> 
> Anything else I could try?

Hi there,
Can you show your `configuration.yaml` and log/output that shows it's not working?
Thx,

gr2m, 2025-01-08:
Thanks @djurny for responding! 

I figured it out now. I had to update the firmware of my stick to the latest and I had to set `adapter: ember` in my case.

My full config looks like this

```yml
homeassistant:
  enabled: true
serial:
  adapter: ember
  rtscts: true
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123170056-if00
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: # redacted
frontend:
  enabled: true
  port: 8099
device_options: {}
devices:
  # ... device friendly_name configruations
version: 4
```

I tried several things like adding `rtscts: true`, not sure if that is needed or what it is even doing 😅

heejew, 2025-01-25:
> i just wonder: why? so what was the problem in the first place? If new config-parameters were required, than a note in the inline-changelogs would be nice (haven't seen one).

Agree. It is unacceptable to make so breaking changes. At the very least, this can be noted in warn logs, and with a transition period. But no break everyone..

fireheadman, 2025-06-14:
ZSTACK for the win!  Glad I stumbled upon this thread

Very strange issue indeed!   It has worked perfect without this setting until I started down my path to migrate from core/supervised on debian to Home Assistant OS this evening.  I run 3 instances of Z2M.

My primary House instance, then 2 additional Edge addons (connecting to SLZB-06 ethernet modules)
The primary came online since it connect via USB, but the other 2 would not connect until I applied the new setting.

[image]

[image]