Koenkk/zigbee2mqtt #25470: v2.0.0-1 won’t start

derekoharrow, 2025-01-04:
### What happened?

I was running v2.0.0 just fine. Upgraded to v2.0.0-1 and Z2M no longer starts. 

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

Latest

### Adapter

SLZB06p

### Setup

Container

### Debug log

derek@docker:~$ docker logs -f zigbee2mqtt
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
TypeError: Cannot create property 'status_topic' on boolean 'true'
    at setValue (/app/lib/util/settingsMigration.ts:55:33)
    at transferValue (/app/lib/util/settingsMigration.ts:180:13)
    at Object.migrateIfNecessary (/app/lib/util/settingsMigration.ts:535:60)
    at start (/app/index.js:136:27)
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.

platonastza, 2025-01-04:
Just happened to me also.
Add the following line below the port section in your zigbee2mqtt/configuration.yaml

adapter: zstack

derekoharrow, 2025-01-04:
> Just happened to me also.
> 
> Add the following line below the port section in your zigbee2mqtt/configuration.yaml
> 
> 
> 
> adapter: zstack 

I already have this:

serial:
  port: tcp://192.168.1.29:6638
  baudrate: 115200
  adapter: zstack

derekoharrow, 2025-01-04:
Any ideas how to get back to the v2 release before yesterday’s? Can’t seem to find the appropriate tags to use and my network is down currently.

derekoharrow, 2025-01-04:
I’ve had to regress back to 1.42.0 using my backed up config to get up and running again

krzysiek2788, 2025-01-04:
Go into your configuration.yaml and find:
```
advanced:
  homeassistant_status_topic: hass/status
  homeassistant_discovery_topic: homeassistant
```

and change it to:
```
advanced:
  homeassistant_status_topic: homeassistant/status
```

this solved this issue for me ;)

dcampogiani, 2025-01-04:
Same issue here, logs:

[2025-01-04 11:00:22] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 11:00:22] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 11:00:23] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_4c7d4d1fd61bef11aee9a4d94909ffd0-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[11:00:32] INFO: Preparing to start...
[11:00:33] INFO: Socat not enabled
[11:00:40] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

krzysiek2788, 2025-01-04:
> Same issue here, logs:
>
> [2025-01-04 11:00:22] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown) [2025-01-04 11:00:22] info: z2m: Starting zigbee-herdsman (3.2.1) [2025-01-04 11:00:23] [...]

This is not the same issue. Remove all regarding adapter in your configuration and leave only path.

derekoharrow, 2025-01-04:
> Go into your zigbee2mqtt configuration.yaml and find:
> 
> ```
> 
> advanced:
> 
>   homeassistant_status_topic: hass/status
> 
>   homeassistant_discovery_topic: homeassistant
> 
> ```
> 
> 
> 
> and change it to:
> 
> ```
> 
> advanced:
> 
>   homeassistant_status_topic: homeassistant/status
> 
> ```
> 
> 
> 
> line`
> 
> homeassistant_discovery_topic: homeassistant`
> 
> need to be removed
> 
> 
> 
> this solved this issue for me ;)

Mine was already set like that and happily running in the early 2.0 release. Did you upgrade from 1.42 or from 2.0?

krzysiek2788, 2025-01-04:
@derekoharrow I updated just now from 1.42 to 2.0.0-1 and had this issue and resolve it like mentioned above 😉

derekoharrow, 2025-01-04:
> @derekoharrow I updated just now from 1.42 to 2.0.0-1 and had this issue and resolve it like mentioned above 😉

Thanks. Good to know. I was already on 2.0.0 and then upgraded to 2.0.0-1. Might try upgrading again now I'm back at 1.42 straight to 2.0.0-1 and see if that makes a difference.

geofffarmer, 2025-01-04:
I have this problem too. I looked in I looked in `configuration.yaml` in `config/zigbee2mqtt` but I have:

`advanced:
  log_level: info`

I'm not sure what to do.

z-master42, 2025-01-04:
2.0.0-1 ist the HA addon version of 2.0.0.

derekoharrow, 2025-01-04:
> 2.0.0-1 ist the HA addon version of 2.0.0.

Thanks

Then, technically, I upgraded from the pre Jan 3rd version of 2.0.0 to the Jan 3rd version of 2.0.0 and got the issue.

adorobis, 2025-01-04:
> and change it to:
> 
> ```
> advanced:
>   homeassistant_status_topic: homeassistant/status
> ```

I have it like that and still face the same issue. Docker install of `latest` release.
```
2025-01-04T10:16:14.634368437Z Starting Zigbee2MQTT without watchdog.
2025-01-04T10:16:14.949185662Z TypeError: Cannot create property 'status_topic' on boolean 'true'
2025-01-04T10:16:14.949215204Z     at setValue (/app/lib/util/settingsMigration.ts:55:33)
2025-01-04T10:16:14.949221196Z     at transferValue (/app/lib/util/settingsMigration.ts:180:13)
2025-01-04T10:16:14.949224668Z     at Object.migrateIfNecessary (/app/lib/util/settingsMigration.ts:535:60)
2025-01-04T10:16:14.949227905Z     at start (/app/index.js:136:27)
```

krzysiek2788, 2025-01-04:
Did you check your mqtt configuration as koneki described in post regarding update to 2.0?

derekoharrow, 2025-01-04:
> Did you check your mqtt configuration as koneki described in post regarding update to 2.0?

Yes, as above - I was already running the pre Jan 3 version of 2.0.0 with that exact config.

adorobis, 2025-01-04:
Do you mean this section:
"By default, Zigbee2MQTT does not listen to the hass/status Home Assistant status topic anymore, homeassistant/status is used instead (if homeassistant.status_topic is not set in your Z2M configuration.yaml). Double check that Home Assistant is using this status topic by going to: HA -> Settings -> Devices & services -> MQTT -> Configure -> Re-configure MQTT -> Next; the Birth message topic should be homeassistant/status."
If so, I have it configured as described. Is there anything else I should check/change?

krzysiek2788, 2025-01-04:
The best to provide your configuration. Yaml fine so we can see where might be the issue

derekoharrow, 2025-01-04:
> The best to provide your configuration. Yaml fine so we can see where might be the issue

Here it is, but as I say this config worked fine in the pre-release of v2

```
homeassistant: true
permit_join: false
frontend:
  port: 9122
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  transmit_power: 20
  channel: 11
  rtscts: false
  pan_id: 6755
  network_key:
    - 42
    - 134
    - 159
    - 11
    - 78
    - 112
    - 61
    - etc...
  last_seen: ISO_8601_local
  log_level: warning
  log_output:
    - console
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  homeassistant_status_topic: homeassistant/status
  timestamp_format: DD-MM-YYYY HH:mm:ss
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt.lan
  discovery: true
serial:
  port: tcp://192.168.1.29:6638
  baudrate: 115200
  adapter: zstack
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
devices: devices.yaml
groups: groups.yaml
device_options:
  legacy: false
```

krzysiek2788, 2025-01-04:
Did you tell to remove: 
```
baudrate: 115200
  adapter: zstack
``??

adorobis, 2025-01-04:
> The best to provide your configuration. Yaml fine so we can see where might be the issue

Here is mine:
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt2
  server: mqtt://10.144.1.103:1883
  keepalive: 60
  version: 4
  user: ****
  password: ****
  reject_unauthorized: false
serial:
  port: /dev/ttyACM0
  adapter: ember
advanced:
  network_key:
    - 169
    - 39
    - 201
    ...
  channel: 15
  log_output:
    - console
    - file
  log_level: debug
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  homeassistant_status_topic: homeassistant/status
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
frontend:
  port: 8080
devices: devices.yaml
groups: groups.yaml
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
device_options:
  legacy: false
```

hytrek, 2025-01-04:
Hi there... deepst applogies... I am no programmer and relatively new to HA. I tried upgrading to the 2.0.0-1 vom 1.48 or so... I was reading the info to possible make changes to the configuration.yaml and after digging for it I found the yaml and below is what is in there. I anybody able to help me make the required changes as the installed version 2.0.0-1 does not start... I would highly appreciate if someone would be taking my hand and grandpa wise instruct me step by step what to do... again. I am a dummie and require slow little steps - MANY THANKS IN ADVANCE

After the installing the update zigbee2mqtt was not running and when attempting to start it "502: Bad Gatewy" is displayed.
here is my full config.yaml... I shortened the password and the list of devices

`homeassistant:
  enabled: true
advanced:
  homeassistant_status_topic: homeassistant/status
  network_key:
    - 60
    - 86
    - 81
    - 40
    - 167
    - 116
    - 169
    - 20
    - 76
    - 134
    - 130
    - 183
    - 105
    - 145
    - 92
    - 133
  pan_id: 7749
  ext_pan_id:
    - 85
    - 136
    - 98
    - 63
    - 1
    - 87
    - 204
    - 201
  log_level: debug
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: secret
  keepalive: 60
  reject_unauthorized: true
  version: 4
frontend:
  enabled: true
  port: 8099
serial:
  port: tcp://192.168.2.218:6638
  baudrate: 115200
device_options: {}
devices:
  'my device list to folow!
    no_occupancy_since: []
version: 4`

krzysiek2788, 2025-01-04:
> Here is mine:
>
> ```
> homeassistant: true
> permit_join: false
> mqtt:
>   base_topic: zigbee2mqtt2
>   server: mqtt://10.144.1.103:1883
>   keepalive: 60
>   version: 4
>   user: ****
>   password: **** [...]

I would remove:
` adapter: ember`
and try it.

adorobis, 2025-01-04:
> I would remove:
> ` adapter: ember`
> and try it.

Nope, still the same. And my understanding from the upgrade post is that it is now mandatory (although I've had it before as well).

adorobis, 2025-01-04:
Btw, I have a second instance with deconz adapter (ConbeeII) and have exactly the same issue there.

krzysiek2788, 2025-01-04:
> > I would remove:
> > ` adapter: ember`
> > and try it.
> 
> Nope, still the same. And my understanding from the upgrade post is that it is now mandatory (although I've had it before as well).

nope for 2.0 it isn't 2.0 is able to autodiscover coordinator you are using.

adorobis, 2025-01-04:
Quick update and success:
after removing the following line from `advanced` section:
`homeassistant_status_topic: homeassistant/status`
the system moved on. Later it turned out that it won't work without the `adapter: ember` line so I've put it back in.
So now it works.

Theoreet, 2025-01-04:
Here's my config  
```
`external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.14
  user: MQTT
  password: MQTT
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  adapter: zstack
advanced:
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  homeassistant_discovery_topic: homeassistant
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  output: attribute_and_json
availability: false
`
```

but when tried to start Z2M getting this error 

```

`[15:34:50] INFO: Preparing to start...
[15:34:50] INFO: Socat not enabled
[15:34:50] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
TypeError: Cannot create property 'discovery_topic' on boolean 'true'
    at setValue (/app/lib/util/settingsMigration.ts:55:33)
    at transferValue (/app/lib/util/settingsMigration.ts:180:13)
    at Object.migrateIfNecessary (/app/lib/util/settingsMigration.ts:535:60)
    at start (/app/index.js:136:27)
    at start (/app/index.js:136:27)`
```

frantathefranta, 2025-01-04:
The issue is this:
```yaml
homeassistant: true
```
while it should look like this:
```yaml
homeassistant:
  enabled: true
  discovery_topic: homeassistant
```

Theoreet, 2025-01-04:
> The issue is this:
> 
> ```yaml
> homeassistant: true
> ```
> 
> while it should look like this:
> 
> ```yaml
> homeassistant:
>   enabled: true
>   discovery_topic: homeassistant
> ```

seems the fix for me...THX a lot

gregg098, 2025-01-04:
> The issue is this:
> 
> ```yaml
> homeassistant: true
> ```
> 
> while it should look like this:
> 
> ```yaml
> homeassistant:
>   enabled: true
>   discovery_topic: homeassistant
> ```

This was the fix for me as well.

adorobis, 2025-01-04:
> The issue is this:
> 
> ```yaml
> homeassistant: true
> ```
> 
> while it should look like this:
> 
> ```yaml
> homeassistant:
>   enabled: true
>   discovery_topic: homeassistant
> ```

For me it was changed automatically when I upgraded to 2.0.0. But still I had to remove below line from `advanced:` section:
```yaml
  homeassistant_status_topic: homeassistant/status
```

derekoharrow, 2025-01-04:
> The issue is this:
> 
> ```yaml
> homeassistant: true
> ```
> 
> while it should look like this:
> 
> ```yaml
> homeassistant:
>   enabled: true
>   discovery_topic: homeassistant
> ```

This has fixed it for me too - not sure why my config worked with the pre-release v2 version, but it works now.

closed 2025-01-04 by derekoharrow

Phara0h, 2025-01-05:
I'm still having the issue here is my config. Woke up to all my things not working

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.109:1883
  user: hhhhhhhh
  password: hhhhhhhh
  keepalive: 30
  version: 5
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2469925-if00
  adapter: deconz
external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
  discovery_topic: homeassistant
permit_join: false
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  availability_timeout: 300
  rtscts: false
  report: true
  last_seen: ISO_8601
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}

```

michalpulda, 2025-01-10:
> For me, I had to manually edit the config file (the GUI didn't work). Not sure if anyone has already mentioned it. I used the "File editor" add-on but you could also use samba and notepad++ or something.

Same here! I was trying repeatedly to change the config from the add-on configuration tab, however when I opened the configuration.yaml file using the File Editor, it was indeed different. Editing the configuration.yaml file from the File Editor fixed the error. Thanks for pointing this out!

drizuid, 2025-01-14:
just wanted to say, i yolo upgraded after who knows how long and I read nothing. I encountered the issues here. My fixes were to do this

> The issue is this:
> 
> ```yaml
> homeassistant: true
> ```
> 
> while it should look like this:
> 
> ```yaml
> homeassistant:
>   enabled: true
>   discovery_topic: homeassistant
> ```

and changing `auto` to `zstack` though i initially ignored the log message `z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.`, because there is no usb adapter to discover... but it still matters 

```
serial:
  port: tcp://192.168.55.28:6638
  adapter: zstack
```

HTH and thanks for the answers in this thread :)