Koenkk/zigbee2mqtt #25472: After update to Z2M 2.0.0 no more sensor.xxx_action for TS0044 devices

sanderlv, 2025-01-04:
### What happened?

After update today to latest 2.0.0 I lost 4 action entities for 4 devices. only battery is now presented.

The action is in Z2M:
[image]

But not in HA anymore:
[image]

I discovered when opening node-red:
[image]

4 devices do not work anymore now.

Please also look into: https://community.home-assistant.io/t/tuya-ts0044-first-action-ignored/475940

### What did you expect to happen?

to see the same previously available and working entities.

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20221226

### Adapter

 zStack3x0 (SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coördinator)

### Setup

Addon on HAOS

blind-oracle, 2025-01-04:
The same goes for IKEA remotes sadly, they don't work anymore

sanderlv, 2025-01-04:
It might be an "action" problem:

https://github.com/Koenkk/zigbee2mqtt/issues/25460

https://github.com/Koenkk/zigbee2mqtt/issues/25461

and there is some guidance here:
https://github.com/Koenkk/zigbee2mqtt/issues/25461#issuecomment-2570458519

I could not find any "action" word in the release notes though...

sanderlv, 2025-01-04:
Ok, found this: https://github.com/Koenkk/zigbee2mqtt/discussions/24198

All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml.
```
homeassistant:
  legacy_action_sensor: true
```

Since I want to be using no-legacy stuff I tried to change to "https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended"

But Is cannot seem to succeed. What entity or device (in HA) presents this "trigger"? I have only 1 battery entity in the device...

sanderlv, 2025-01-04:
Ok, for me it works now.

I do no use HA anymore but read MQTT directly in NodeRed:

topic: zigbee2mqtt/4-button-kantoor/action

All is fine now.

closed 2025-01-04 by sanderlv

CrazyBavarian, 2025-01-04:
@sanderlv  i also use nodered could you just tell me what exactly did you change in nodered to get ,,action"?

sanderlv, 2025-01-04:
Use the mqtt in node and listen to mqtt directly

ulli-us, 2025-01-07:
I attached a sample flow. 
MQTT-in node: change to the name of your device   i.e. zigbee2mqtt/<device name>
Change node: Set msg.payload to msg.payload.action value.
The resulting message has the same payload as the original event node.

[click_action_sample_flow.json](https://github.com/user-attachments/files/18339264/click_action_sample_flow.json)