Koenkk/zigbee2mqtt #25473: Error: Adapter 'auto' does not exists

skipperkm, 2025-01-04:
### What happened?

Hi,
Have a healthy and successful New Year in advance.
After the update from 1.42.0-2 to 2.0.0-1 the coordinator is no longer recognized. 
The add-on does not start.

serial:
  port: tcp://172.16.98.91:6638
  adapter: auto
  baudrate: 115200

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20240710

### Adapter

cod.m ZigBee Coordinator 1.0 (CZC-1.0)

### Setup

Homeassistant OS AddOn Proxmox VM

### Debug log

Migration notes written in /config/zigbee2mqtt/migration-1-to-2.log
Migration notes written in /config/zigbee2mqtt/migration-2-to-3.log
Migration notes written in /config/zigbee2mqtt/migration-3-to-4.log
[2025-01-04 09:34:33] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 09:34:33] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 09:34:33] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 09:34:33] error: 	z2m: Exiting...
[2025-01-04 09:34:33] error: 	z2m: Error: Adapter 'auto' does not exists, possible options: deconz, ember, ezsp, zstack, zboss, zigate
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:80:19)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

EinfachHans, 2025-01-04:
You should read the breaking changes docs before updating:
https://github.com/Koenkk/zigbee2mqtt/discussions/24198

First point under **General**:

> Improved adapter discovery; when upgrading from Zigbee2MQTT 1.x.x, this might result in USB adapter discovery error (No valid USB adapter found), see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 how to fix this.

z-master42, 2025-01-04:
The solution is in your logs:

Adapter 'auto' does not exists, possible options: deconz, ember, ezsp, zstack, zboss, zigate

skipperkm, 2025-01-04:
Do all devices then have to be retrained?

z-master42, 2025-01-04:
Just specifiy your adapter.

skipperkm, 2025-01-04:
Hi,

thank you, its working, have a nice day
:-)

z-master42, 2025-01-04:
Please close the issue.

closed 2025-01-04 by skipperkm