Koenkk/zigbee2mqtt #25476: Version 2.0.0-1 won't start after update

lubracator, 2025-01-04:
### What happened?

After the update to v2.0.0-1 Zigbee2MQTT won't start anymore

### What did you expect to happen?

I expected the add on to start after the update from 1.42.0, which was working fine

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

latest

### Adapter

Nabu Casa Home Assistant Connect ZBT-1

### Setup

Docker container in UNRAID server

### Debug log

[11:07:02] INFO: Preparing to start...
[11:07:02] INFO: Socat not enabled
[11:07:03] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 11:07:04] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 11:07:04] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 11:07:04] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 11:07:04] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Nabu_Casa_Home_Assistant_Connect_ZBT-1_serialnr-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-04 11:07:04] info: 	zh:zstack:znp: Serialport opened
[2025-01-04 11:07:04] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-04 11:07:05] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-04 11:07:24] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 11:07:24] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 11:07:24] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 11:07:24] error: 	z2m: Exiting...
[2025-01-04 11:07:24] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

m0wlheld, 2025-01-04:
> [2025-01-04 11:07:24] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions

2.0 is a breaking change and requires preparation. Did you do this?

lubracator, 2025-01-04:
It's not my instance, it's my father's. I asked him what steps he took. He just updated when prompted, he didn't do any prepartion. I'm just helping to troubleshoot, but I'm fairly new to home assistant. And I can't pinpoint the issue he's having.

m0wlheld, 2025-01-04:
If you are using the Home Assistant Zigbee2MQTT add-on, you should ask the [Home Assistant community](https://community.home-assistant.io/) for assistance. 

The required actions when using Zigbee2MQTT standalone are present in the release notes.

Photogad, 2025-01-05:
You need to add adapter: ezsp to configuration.yaml under serial section. This fixed it for me.

roobbb, 2025-01-05:
Hello.

As far as I know "ezsp" is only necessary when you have a Silicon Labs based device and it is stated on this page as deprecated https://www.zigbee2mqtt.io/guide/adapters/emberznet.html.
The device in the logs above looks more like a zstack one (Texas Instruments).

Kind regards
roobbb

Koenkk, 2025-01-05:
> Nabu Casa Home Assistant Connect ZBT-1

You need to add adapter `ezsp` or `ember` to your `configuration.yaml`, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 for more info

closed 2025-01-05 by Koenkk