Koenkk/zigbee2mqtt #25477: ZWT198/ZWT100-BH missing child lock switch in HA after 2.0 upgrade

pznamenskii, 2025-01-04:
### What happened?

child lock switch object is no longer present in HA after 2.0 upgrade.
was present before upgrade.
still exposed in z2m though:
```
{
    "backlight_mode": "medium",
    "child_lock": "LOCK",
    "current_heating_setpoint": 23.5,
    "deadzone_temperature": 1,
    "factory_reset": "OFF",
    "frost_protection": "ON",
    "linkquality": 142,
    "local_temperature": 22.1,
    "local_temperature_calibration": 0,
    "max_temperature_limit": 60,
    "preset": "auto",
    "running_state": "heat",
    "schedule_holiday": "08:00/22.0°C 23:00/16.0°C",
    "schedule_weekday": "06:00/20.0°C 08:00/16.0°C 11:30/16.0°C 12:30/16.0°C 17:00/22.0°C 22:00/16.0°C",
    "sensor": "internal",
    "system_mode": "heat",
    "working_day": "5-2"
}
```

### What did you expect to happen?

child lock switch object exposed to HA.

### How to reproduce it (minimal and precise)

update to 2.0, observe MQTT object for thermostat

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20240710

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

HA add-on, Home Assistant Blue

### Debug log

[45df7312_zigbee2mqtt_2025-01-04T10-16-34.970Z.log](https://github.com/user-attachments/files/18306427/45df7312_zigbee2mqtt_2025-01-04T10-16-34.970Z.log)

m0wlheld, 2025-01-04:
> All child lock entities are now discovered as a switch instead of lock.

See https://github.com/Koenkk/zigbee2mqtt/discussions/24198

it‘s a documented breaking change

pznamenskii, 2025-01-04:
> > All child lock entities are now discovered as a switch instead of lock.
> 
> See #24198
> 
> it‘s a documented breaking change

thanks, I was specifically looking for this switch on device's dashboard but none was to be found - it seems it took some time to appear, now I see this switch.

closed 2025-01-04 by pznamenskii