Koenkk/zigbee2mqtt #25479: Aqara LLKZMK12LM

KrOnAsK, 2025-01-04:
### What happened?

After Updating to Zigbee 2.0.0-1 the Aqara LLKZMK12LM does not publish the state action to Homeassistant anymore even though it is enabled in the device specific settings.
[image]
[image]

### What did you expect to happen?

That like before I could see the state action 

### How to reproduce it (minimal and precise)

add device to z2m and then seee that its not there anymore

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee Gateway, ZBDongle-E

### Setup

Add-on

dpgh947, 2025-01-04:
Interesting. I just opened an issue for a smart knob device that also lost its action entity.

dpgh947, 2025-01-04:
I just found info about action entities under the breaking changes (I missed the doc, it was not flagged up very well)

m0wlheld, 2025-01-04:
So, can this issue be closed (and the one about the smart knob)?

KrOnAsK, 2025-01-04:
could you link the breaking change and what to do about it? I can't seem to find it

m0wlheld, 2025-01-04:
> could you link the breaking change and what to do about it? I can't seem to find it

Release Notes: https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0

KrOnAsK, 2025-01-04:
okay so there is a new method to do that. Thanks

closed 2025-01-04 by KrOnAsK