Koenkk/zigbee2mqtt #25480: [Feature request]: Add changelog to docker releases

PossiblyMax, 2025-01-04:
### Is your feature request related to a problem? Please describe

It seems there has been some frustration with users who do not understand (or care?) about semver versioning.

For example, in Home Assistant the update window shows the docker changelog as: Updated Zigbee2MQTT to version [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0)

This is fine for folk who recognise this is a big jump and click through, but it might be clearer to publish more information on the docker image, so users can be more informed before using.

### Describe the solution you'd like

I would like to see the changelog (or some of it) on the docker release too.

At minimum any Breaking Changes and Features. Bug Fixes can't hurt but are likely less pressing. A link to the release (as current) would still be appreciated to facilitate jumping in to the full notes if necessary.

### Describe alternatives you've considered

Even just a little information if the version has breaking changes from different version numbers would be handy. Then users could compare their current version and identify if they should click through. I feel like this is more maintenance than just attaching the latest changelog though.

### Additional context

[image]

m0wlheld, 2025-01-04:
Click on the "2.0.0" link ...

PossiblyMax, 2025-01-04:
Yes, I note that in my request:

> This is fine for folk who recognise this is a big jump and click through

But it can't hurt to put the information up front.

Koenkk, 2025-01-05:
This has been added in https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/pull/690

closed 2025-01-05 by Koenkk