Koenkk/zigbee2mqtt #25481: tuya 4-button switch ts0044  not show mqtt

khoanong, 2025-01-04:
### What happened?

after upgrade , tuya 4-button switch ts0044  not show mqtt
[image]

### Zigbee2MQTT version

2.0

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Intel NUC

m0wlheld, 2025-01-04:
Did you step through the release notes and the mentioned breaking changes before the update? See https://github.com/Koenkk/zigbee2mqtt/discussions/24198 for details.

KossikCZ, 2025-01-04:
@khoanong I have the same button, see it in MQTT the same way as you do but the button can be still used in automations. What issue do you really see? You only need to change your automations a bit, since this release removes the sensors (sensor.*_action) and replaces them by actions.

khoanong, 2025-01-04:
> Did you step through the release notes and the mentioned breaking changes before the update? See #24198 for details.

need add :
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
in configuration.yaml before update ?

khoanong, 2025-01-04:
> @khoanong I have the same button, see it in MQTT the same way as you do but the button can be still used in automations. What issue do you really see? You only need to change your automations a bit, since this release removes the sensors (sensor.*_action) and replaces them by actions.

Let me check again due to just restore to 1.42.0-2

KossikCZ, 2025-01-04:
@khoanong Once you are on version 2.0.x again, simply go to your automation and use the device trigger instead. 
[image]

dpgh947, 2025-01-04:
There is a simple tickbox I just found in the z2m settings (select z2m in left menu, then settings under the cog wheel at the top, then the home assistant tab) to re-enable the action entities:

[image]

KossikCZ, 2025-01-04:
@dpgh947 That's true, of course, but that option is deprecated and will go away sooner or later. I would not rely on that any longer personally and rather use the time now to migrate over.

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking chnages, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>    ```yaml
>    homeassistant:
>      legacy_action_sensor: true
>    ```

closed 2025-01-05 by Koenkk

robinoux, 2025-01-10:
Invalid config for 'homeassistant' at configuration.yaml, line 37: 'legacy_action_sensor' is an invalid option for 'homeassistant', check: legacy_action_sensor

nz-ngauruhoe, 2025-02-02:
Just to get a feel for the impact. Using [https://github.com/moshess/homeassistant-blueprints/blob/8c49cb7e7d0e428ac3023d556ce08fe29c8f2395/mqtt/tuya-ts004f.yaml](url) blueprint. Makes it easier for the-not-so-technical user of HA. Couldn't get it to work anymore after upgrading to a RPI4. Spent very many hours to find out that removed 'action sensors' are causing this. 
With the number of people using, and enjoying, HA, the developers should perhaps be more thoughtful introducing breaking changes. The more users relying on it the more important compatibility becomes. My two cents......
ps. I have gratefully made use of the legacy actions option.......