Koenkk/zigbee2mqtt #25485: Upgrade to 2.0 issue even when 2.0.0 breaking changes steps are executed.

edegraaff, 2025-01-04:
### What happened?

Read 2.0.0 breaking changes 
https://github.com/Koenkk/zigbee2mqtt/discussions/24198
Checked configu file against these items and corrected:
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
I have a pi with zigbee2mqtt installed so: Bare-metal installations (no Docker/Home Assistant add-on)

since  pnpm is now used instead of npm, the [installation](https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/installation/01_linux.md#installing) instructions and update.sh script has been updated accordingly. For existing installations you need to execute npm install -g pnpm before updating.
executed : sudo npm install -g pnpm > ok 

executed :
mv data/configuration.yaml data/configuration.yaml.bak
./update.sh # (update as you usually do)
mv data/configuration.yaml.bak data/configuration.yaml

i keep this message 
error: Your local changes to the following files would be overwritten by merge:
  data/configuration.yaml

Could it be the update.sh script is not able to update the 1.x to 2.x branch and some additional commands are needed to pull the 2.x branch 

### What did you expect to happen?

After the prereq's the upgrade from 1.x to 2.x should work.

### How to reproduce it (minimal and precise)

Take 1.x distribution 
Execute the proposed changes from Read 2.0.0 breaking changes 
run update.sh 

After the procedures, make sure to put the config files in original state to be able to continue the operation. And/or execute sudo reboot. 

Regards
Eelco

### Zigbee2MQTT version

 1.42.0 commit: bc6eccf6

### Adapter firmware version

 20230507

### Adapter

zStack3x0

### Setup

Plain on raspberry pi Raspberry Pi 3 Model B Plus Rev 1.3 Linux 	Linux 6.1.21-v7+

### Debug log

Not Applicable

m0wlheld, 2025-01-04:
From the breaking changes issue #24198:

> Bare-metal installations (no Docker/Home Assistant add-on)
> 
> * For Git based installations: removed configuration.yaml from repo, this will result in the following error when updating:
> 
> ```
>     error: Your local changes to the following files would be overwritten by merge:
>       data/configuration.yaml
> ```
> 
>   To fix this, update by executing:
> 
> ```
>     mv data/configuration.yaml data/configuration.yaml.bak
>     ./update.sh # (update as you usually do)
>     mv data/configuration.yaml.bak data/configuration.yaml
> ```

Did you follow this?

edegraaff, 2025-01-04:
yes, thats what puzzeled me...

m0wlheld, 2025-01-04:
Looking at the code of [update.sh](https://github.com/Koenkk/zigbee2mqtt/blob/master/update.sh) i see that it does a `git pull` which could cause the error, *if* configuration.yaml still exists. But it shouldn't, because the previous 

```bash
mv data/configuration.yaml data/configuration.yaml.bak
```

renames `configuration.yaml` to `configuration.yaml.bak`, so there is no `configuration.yaml` any more. You can easily verify this (using `ls` or any filesystem explorer). Did the first `mv` command into any errors?

edegraaff, 2025-01-04:
mv command works without error

$ ls data/configuration.*
data/configuration.example.yaml
data/configuration.yaml.bak~
data/configuration.yaml.bak

./update.sh 
Checking Zigbee2MQTT status...
Stopping Zigbee2MQTT...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating bc6eccf6..060ae99c
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.example.yaml
Please commit your changes or stash them before you merge.
Aborting

My action: This error> the example file also exists. The file renamed ( $ mv data/configuration.example.yaml data/configuration.example.yaml.bak) then the update runs, but will fail later in other dependencys. 

See attached log: 
Installing dependencies...
npm ERR! code EUSAGE
npm ERR!
npm ERR! The `npm ci` command can only install with an existing package-lock.jso                                                                                                                                                             n or
npm ERR! npm-shrinkwrap.json with lockfileVersion >= 1. Run an install with npm@                                                                                                                                                             5 or
npm ERR! later to generate a package-lock.json file, then try again.
npm ERR!
npm ERR! Clean install a project
npm ERR!
npm ERR! Usage:
npm ERR! npm ci
npm ERR!
npm ERR! Options:
npm ERR! [-S|--save|--no-save|--save-prod|--save-dev|--save-optional|--save-peer                                                                                                                                                             |--save-bundle]
npm ERR! [-E|--save-exact] [-g|--global] [--global-style] [--legacy-bundling]
npm ERR! [--omit <dev|optional|peer> [--omit <dev|optional|peer> ...]]
npm ERR! [--strict-peer-deps] [--no-package-lock] [--foreground-scripts]
npm ERR! [--ignore-scripts] [--no-audit] [--no-bin-links] [--no-fund] [--dry-run                                                                                                                                                             ]
npm ERR! [-w|--workspace <workspace-name> [-w|--workspace <workspace-name> ...]]
npm ERR! [-ws|--workspaces] [--include-workspace-root] [--install-links]
npm ERR!
npm ERR! aliases: clean-install, ic, install-clean, isntall-clean
npm ERR!
npm ERR! Run "npm help ci" for more info

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2025-01-04T15_22_46_758Z-debug-0.log

[zigbee2mqtterror.log](https://github.com/user-attachments/files/18307329/zigbee2mqtterror.log)

m0wlheld, 2025-01-04:
You (unintenionally) did changes to configuration.example.yaml, that would be overriden by the git pull during update.sh. 

Try „git checkout configuration.example.yaml“ to revert the changes, then run update.sh again.

edegraaff, 2025-01-04:
I think i copyed the example file to the config file in the past. The changes as described in the documentation where added to the configuration.yaml. That part is ok. The update.sh introduced a new issue with the versions of the dependancys. In this case the old version of node. 
The fix:  sudo curl -fsSL https://deb.nodesource.com/setup_20.x | sudo -E bash -
And then run the update script. 
After that copy the config file back to the original config.yaml

Restart the pi and up we are.  running : Zigbee2MQTT versie
[2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) commit: [060ae99c](https://github.com/Koenkk/zigbee2mqtt/commit/060ae99c)

Suggestion: maybe some prereq tests in the procedure before to go for the upgrade to 2.x ?

Last issue:  npm install -g pnpm fail as user, maybe i need to add a sudo in front of it. >> yes that is needed. 

I am up and running, thanks for helping me out, and maybe my observations could help to improve the upgrade procedure. 
Regards
Eelco

closed 2025-03-13 by github-actions[bot]