Koenkk/zigbee2mqtt #25486: z2m 2.0.0-1 fails to start without any error logs

Hardtack, 2025-01-04:
### What happened?

After upgrading z2m to 2.0.0-1 it started to fail to boot without any error logs.
It just stopped after showing following logs I attached

These are what I tried

- Add `adapter: zstack` field to `serial` section in `config/zigbee2mqtt/configuration.yaml` file
- Add following fields to the config file
  ```yaml
  advanced:
    homeassistant_legacy_entity_attributes: false
    homeassistant_legacy_triggers: false
    legacy_api: false
    legacy_availability_payload: false
  device_options:
    legacy: false
  ```
- Verify that serial path to coordinator is correct (both tried `/dev/ttyUSB0` and `/dev/serial/by-id/...`
- Upgrade coordinator's firmware version.

But nothing works for me. I cannot even investigate the cause since I don't have any error logs for this.

### What did you expect to happen?

I expected that Z2M start successfully or at least prints error logs of failure.

### How to reproduce it (minimal and precise)

It happens whenever I try to start Z2M

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20240710

### Adapter

zstack

### Setup

Add-on on Home Assistant OS on Raspberry Pi 4

### Debug log

```

[2025-01-04 21:35:34] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.20-48-53'
[2025-01-04 21:35:34] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 21:35:34] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 21:35:34] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 21:35:34] debug: 	z2m: sd-notify loaded
[2025-01-04 21:35:34] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 21:35:34] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":33163,\"extendedPanID\":[75,222,162,128,58,24,6,44],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-04 21:35:34] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"c8bfffc396d8ed11b8cf776162c613ac","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_c8bfffc396d8ed11b8cf776162c613ac-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-04 21:35:34] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"c8bfffc396d8ed11b8cf776162c613ac","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_c8bfffc396d8ed11b8cf776162c613ac-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-04 21:35:34] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":33163,"extendedPanID":[75,222,162,128,58,24,6,44],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-04 21:35:34] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
```

jeeck-jan, 2025-01-04:
same problem here, 
starts up, don't finish startup, watchdog restarts again and again, no error reports in log

jeeck-jan, 2025-01-04:
this is the log, no error

[14:23:42] INFO: Preparing to start...
[14:23:42] INFO: Socat not enabled
[14:23:44] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 14:23:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 14:23:49] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 14:23:49] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 14:23:49] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_becefcf6f13aec11a768a4957a0af07f-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}

eshbalt, 2025-01-04:
same problem

[2025-01-04 17:56:44] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.13-59-29'
[2025-01-04 17:56:44] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 17:56:44] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 17:56:44] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 17:56:44] debug: 	z2m: sd-notify loaded
[2025-01-04 17:56:44] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 17:56:44] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-04 17:56:44] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"505575d90513ec11b1f923c7bd930c07","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_505575d90513ec11b1f923c7bd930c07-if00-port0","vendorId":"10c4","productId":"ea60"}]

Fluimucil, 2025-01-04:
Exact same problema, exactly done the some. Hope some one comes with a solution

dcampogiani, 2025-01-05:
Same issue here.

Config:

port: >-
  /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_4c7d4d1fd61bef11aee9a4d94909ffd0-if00-port0
adapter: zstack

Specs:
Core 2025.1.0
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250103.0

Hardware:
Raspberry Pi 2

PatMei, 2025-01-05:
Following, same problem here.

Changed config as mentioned in guide, upgraded Zigbee stick (Slaesh), upgraded several times and restored backup.
Log ends every time with "Opening SerialPort ..."
No Error message in logfile:

[2025-01-05 09:25:47] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.09-23-36'
[2025-01-05 09:25:47] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 09:25:47] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 09:25:47] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 09:25:47] debug: 	z2m: sd-notify loaded
[2025-01-05 09:25:47] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 09:25:47] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6723,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 09:25:47] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"00_12_4B_00_21_CC_3D_DE","pnpId":"usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_3D_DE-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-05 09:25:47] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"00_12_4B_00_21_CC_3D_DE","pnpId":"usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_3D_DE-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-05 09:25:47] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6723,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 09:25:47] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}

Config:
data_path: /config/zigbee2mqtt
socat:
  enabled: true
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.2.11:1883
  user: z2m
  password: XXXX
serial:
  port: /dev/ttyUSB0
  adapter: zstack
external_converters: []
devices: devices.yaml
groups: groups.yaml
permit_join: false
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  homeassistant_status_topic: homeassistant/status
  log_level: debug
  pan_id: 6723
  channel: 11
  network_key:
    - 3
    - ...
  availability_blocklist: []
  availability_passlist: []
device_options:
  legacy: false
blocklist: []
passlist: []
queue: {}