Koenkk/zigbee2mqtt #25488: Version 2 incompatible with PHilips Hue remote 324131092621

gerrytb, 2025-01-04:
### What happened?

After install version 2.0.0-1 and following the steps from the documentation for the bracking change, the remote does not work. With version 1.42 it is working with Legacy=true, with the version 2.0.0 I cannot set the Legacy.

### What did you expect to happen?

It is expected that in version 2.0.0-1 the device is still suported like it is in version 1.42

### How to reproduce it (minimal and precise)

Install version 2.0.0 and the remote stops working. Home Assistant detects when a button is pressed, but no actions in triggered.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20220219

### Adapter

zStack3x0 SONOFF ZBDongle-P

### Setup

Add-on

EitschiDE, 2025-01-05:
Same problem with Hue Dimmer Switch.
There is no longer the action entity in HA and all the automations with the action entity not working.

klmi, 2025-01-05:
same here, any news @EitschiDE @gerrytb.
Anything I can provide to help?

EitschiDE, 2025-01-05:
When i press a button on the dimmer switch a payload is send to mqtt.
I can see it in the MQTT Explorer:
[image]
But there is still no action entity in HA.
Is this a problem in zigbee2mqtt or in mqtt itself?

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-05 by Koenkk

EitschiDE, 2025-01-05:
Thank you very much for the hint, I unnoticed the detail.
I have changed the automation accordingly, now it works again.
Now no need action sensors :-)

DynaM1ke, 2025-04-26:
> Thank you very much for the hint, I unnoticed the detail. I have changed the automation accordingly, now it works again. Now no need action sensors :-)

Can you please post your changes?