Koenkk/zigbee2mqtt #25489: Version 2.0.0 not full compatible with Philips 9290030675 Motion Sensor

gerrytb, 2025-01-04:
### What happened?

After installing mqtt 2.0.0-1 the motion sensor hast lost the entitiy sensor.philipsmotion_illuminance, so it is not possible to detect light or darkness.

### What did you expect to happen?

With version 2.0.0-1 all entities should be present

### How to reproduce it (minimal and precise)

Install version 2.0.0-1 and illuminance entity will not be aviable for Philips 9290030675

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20220219

### Adapter

zStack3x0 SONOFF ZBDongle-P

### Setup

Add-on

Bart-van-Gorp, 2025-01-04:
I've opened the same issue at the exact time. My issue is the lux entity of the Hue outdoor sensor. Is yours the indoor or outdoor sensor?

m0wlheld, 2025-01-04:
https://github.com/Koenkk/zigbee2mqtt/discussions/24198 - illuminance_lux has been replaced by illuminance. You need to reflect this in your Home Assistant configuration and dashboards.

gerrytb, 2025-01-04:
My device is indor, but the illuminance is not present for Home Assisntant.

In Zigbee2MQTT I can see:
> Temperature
> Temperatura de la vàlvula mesurada
> 21.57°C
> Occupancy
> Indica si el dispositiu ha detectat ocupació
> Clar
> Battery
> Remaining battery in %, can take up to 24 hours before reported
> 1%
> Illuminance
> Measured illuminance
> 16lx

But in Home assistant Illuminance is like disabled:
sensor.philipsmotion_illuminance

m0wlheld, 2025-01-04:
Enable it.

gerrytb, 2025-01-04:
Fixed, in Zigbee2MQTT I renamed the device and Home Asssistant toke the right entity.

For me the ticked can be closed.
Thank you very much

leeaash, 2025-01-05:
miss illuminace for Aqara sensors also

Larsvb0, 2025-01-05:
as per https://github.com/Koenkk/zigbee2mqtt/issues/25488 I renamed the device in MQTT and now it appears
[image]
[image]
:

Koenkk, 2025-01-05:
This is one of the breaking changes of z2m 2.0.0, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> Renamed illuminance_lux to illuminance, this breaks in two ways: illuminance_lux has been removed and the value of illuminance changes.

closed 2025-01-05 by Koenkk