Koenkk/zigbee2mqtt #25490: Hue outdoor sensor: lux sensor not available since last update

Bart-van-Gorp, 2025-01-04:
### What happened?

I've updated ZigBee to mqtt to the latest version and found out that I got an error on a specific entity in home assistant. The lux sensor of the Hue outdoor sensor doesn't work no more, while it worked in previous versions. Although it seems that the lux sensor is available in the Zigbee database, it seems that the automations show the lux entity as an unknown entity. My automations don't work no more since updating.

[image]

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20210708

### Adapter

Sonoff ZigBee 3.0

### Setup

Home assistant OS

m0wlheld, 2025-01-04:
> * Renamed illuminance_lux to illuminance, this breaks in two ways: illuminance_lux has been removed and the value of illuminance changes. This affects the following devices: 83633205, MOSZB-140, MOSZB-153, EFEKTA_iAQ3, EFEKTA_PWS_Max, EFEKTA_PWS_MaxPro, EFEKTA_eON29wz, EFEKTA_eFlower_Pro, EFEKTA_iAQ, E2134, POK004, POK011, ROB_200-070-0, HK-SENSOR-4IN1-A, AU-A1ZBPIRS, RADON TriTech ZB, TS0601_illuminance_temperature_humidity_sensor_1, TS0202_3, S-LUX-ZB, TS0222_light, THE01860A, TS0222, LCZ030, MIR-HE200-TY, XFY-CGQ-ZIGB, TS0601_smart_human_presence_sensor_1, Y1_IN, ZY-M100-S_1, ZY-M100-S_2, WZ-M100, ZG-205ZL, ZG-204ZM, YXZBRB58, ZY-M100-24G, ZY-M100-24GV2, ZY-M100-24GV3, MTG075-ZB-RL, RT_ZCZ03Z, ZY-M100-S_3, ZG-205Z, 8750001213, HSE2927E, ZB003-X, 07502L, JS-MC-SENSOR-ZB, KK-ES-J01W, KMPCIL_RES005, RTCGQ11LM, RTCGQ12LM, b-parasite, 7A-SS-ZABC-H0, 5AA-SS-ZA-H0, RTCGQ14LM, RTCGQ15LM, ZNCLBL01LM, GZCGQ01LM, GZCGQ11LM, 07047L, 9290012607, 9290019758, 9290030675, 9290030674, 545D6306, CCT595011, ZSS-ZK-THL, 4512770, TERNCY-PP01, MBD-S, CTM_MBD_Dim, 3RSNL02043Z, WB-MSW-ZIGBEE v.3, MS100, USM-300ZB, PIR313-E.

Breaking changes, mentioned in the release notes and in detail explained by https://github.com/Koenkk/zigbee2mqtt/discussions/24198 . Additionally, there have been at least 3 issues today, describing the same "issue".

Bart-van-Gorp, 2025-01-04:
> Breaking changes, mentioned in the release notes and in detail explained by #24198 . Additionally, there have been at least 3 issues today, describing the same "issue".

Thank you, I was able to solve this issue by editing the automation on my desktop instead of my mobile phone. On my mobile it didn't wanted to select the outdoor sensor, on my desktop it did. It's true that it has a new an better name. It all worked out.

m0wlheld, 2025-01-04:
May i kindly ask you to close this issue?

closed 2025-01-05 by Koenkk