Koenkk/zigbee2mqtt #25491: lumi.lock.v1 no longer works with update 1.40.2-1

pwet45, 2025-01-04:
### What happened?

Hello, following the update in 1.40.2-1 the module lumi.lock.v1 no longer works, zigbee2mqtt no longer receives the status of the inserted key

### What did you expect to happen?

The lumi.lock.v1 module works well with version 1.40.1-1

### How to reproduce it (minimal and precise)

By updating to 1.40.2-1

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

latest

### Adapter

SLZB-06

### Setup

Docker

pwet45, 2025-01-16:
Here is the difference between the two Z2M versions for this module

1.40.1-1

`[2025-01-16 11:14:33] debug: 	z2m: Received Zigbee message from 'Serrure Porte Entrée', type 'attributeReport', cluster 'genBasic', data '{"65328":"0x120101ffffffffff"}' from endpoint 1 with groupID 0
[2025-01-16 11:14:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Serrure Porte Entrée', payload '{"inserted":"01","last_seen":"2022-09-26T21:37:39+02:00","linkquality":98}'
[2025-01-16 11:14:34] debug: 	zh:zstack:unpi:parser: <-- [254,37,68,129,0,0,0,239,197,198,1,1,0,112,0,62,111,102,0,0,17,25,79,2,0,86,7,0,0,8,8,239,0,1,190,0,0,72,197,198,29,89]
[2025-01-16 11:14:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,37,68,129,0,0,0,239,197,198,1,1,0,112,0,62,111,102,0,0,17,25,79,2,0,86,7,0,0,8,8,239,0,1,190,0,0,72,197,198,29,89]
[2025-01-16 11:14:34] debug: 	zh:zstack:unpi:parser: --> parsed 37 - 2 - 4 - 129 - [0,0,0,239,197,198,1,1,0,112,0,62,111,102,0,0,17,25,79,2,0,86,7,0,0,8,8,239,0,1,190,0,0,72,197,198,29] - 89
[2025-01-16 11:14:34] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":50885,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":112,"securityuse":0,"timestamp":6713150,"transseqnumber":0,"len":17,"data":{"type":"Buffer","data":[25,79,2,0,86,7,0,0,8,8,239,0,1,190,0,0,72]}}
[2025-01-16 11:14:34] debug: 	zh:controller: Received payload: clusterID=61184, address=50885, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=112, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":79,"commandIdentifier":2},"payload":{"seq":22016,"dpValues":[{"dp":7,"datatype":0,"data":{"type":"Buffer","data":[8,239,0,1,190,0,0,72]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2025-01-16 11:14:34] debug: 	zh:zstack:unpi:parser: --- parseNext []`

2.0.0-2

```
[2025-01-16 10:09:01] debug: 	z2m: Received Zigbee message from 'Serrure Porte Entrée', type 'attributeReport', cluster 'genBasic', data '{"65328":"1297320366682669055"}' from endpoint 1 with groupID 0
[2025-01-16 10:09:01] error: 	z2m: Exception while calling fromZigbee converter: data.substr is not a function}
[2025-01-16 10:09:01] debug: 	z2m: TypeError: data.substr is not a function
   at Object.convert (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/lumi.ts:2852:36)
   at Receive.onDeviceMessage (/app/lib/extension/receive.ts:170:51)
   at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23)
   at EventEmitter.emit (node:events:530:35)
   at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:130:22)
   at Controller.<anonymous> (/app/lib/zigbee.ts:125:27)
   at processTicksAndRejections (node:internal/process/task_queues:105:5)
[2025-01-16 10:09:02] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,197,198,1,1,0,112,0,144,226,191,0,0,13,25,34,2,0,69,1,2,0,4,0,2,83,22,197,198,29,154]
[2025-01-16 10:09:02] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,197,198,1,1,0,112,0,144,226,191,0,0,13,25,34,2,0,69,1,2,0,4,0,2,83,22,197,198,29,154]
[2025-01-16 10:09:02] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,197,198,1,1,0,112,0,144,226,191,0,0,13,25,34,2,0,69,1,2,0,4,0,2,83,22,197,198,29] - 154
[2025-01-16 10:09:02] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":50885,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":112,"securityuse":0,"timestamp":12575376,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[25,34,2,0,69,1,2,0,4,0,2,83,22]}}
[2025-01-16 10:09:02] debug: 	zh:controller: Received payload: clusterID=61184, address=50885, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=112, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":34,"commandIdentifier":2},"payload":{"seq":17664,"dpValues":[{"dp":1,"datatype":2,"data":{"type":"Buffer","data":[0,2,83,22]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
```

Help me solve this problem Thank you in advance

Koenkk, 2025-01-17:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

pwet45, 2025-02-02:
Thank you very much for your help, it works very well with version 2.1.0-1, you can close the ticket

closed 2025-02-03 by Koenkk