Koenkk/zigbee2mqtt #25492: Version 2.0.0 breaks ROB_200-007-0 (Robb Zigbee 8 button wall switch)

rutgerh, 2025-01-04:
### What happened?

After upgrading to 2.0.0 the entities are gone in HA. (Also updated HA to 2025.1.0
The actions are visible in the Exposes. Also visible in MQTT is seeing the action.

Reconfigure / deleting and re-adding also didn't helped.

### What did you expect to happen?

Show / publish entities in HA 

### How to reproduce it (minimal and precise)

Upgrade to 2.0.0 and / or 2025.1.0.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

HAOS on esxi host from OVA

rutgerh, 2025-01-04:
Debug
[2025-01-04 141946.002 DEBUG (MainTh.txt](https://github.com/user-attachments/files/18306951/2025-01-04.141946.002.DEBUG.MainTh.txt)

m0wlheld, 2025-01-04:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0 - read the linked "Breaking Chances" issue. It's the one marked in red, because it's important.

rutgerh, 2025-01-04:
I did checked the breaking changes before upgrading, did checked them again but unable to find what you are referring to in the breaking changes post.

m0wlheld, 2025-01-04:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0, then (first link) https://github.com/Koenkk/zigbee2mqtt/discussions/24198. From the section "Breaking Changes" -> "Home Assistant"

> ## Breaking Changes
> ### Home Assistant
> * Removed duplicate sensor/select for select/number/button entities, the full list of removed entities per model can be found [here](https://gist.github.com/Koenkk/7499edc879b7398213438c6e837ceb2c).
> * Only breaking when legacy is enabled:
> 
>     All click sensors have been removed (homeassistant.legacy_triggers setting). This means all sensor.*_click entities are removed. Use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead.
>     All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml.
> 
>     homeassistant:
>       legacy_action_sensor: true
> 
>     Entity attributes (homeassistant.legacy_entity_attributes setting) has been removed. This means that entities discovered by Zigbee2MQTT will no longer have entity attributes (Home Assistant entity attributes are accessed via e.g. states.binary_sensor.my_sensor.attributes).

Your device is explicitely mentioned in the "full list of removed entities per model can be found [here](https://gist.github.com/Koenkk/7499edc879b7398213438c6e837ceb2c)"

rutgerh, 2025-01-04:
Still I cannot find the ROB_200-007-0, but now I get it.
The action sensor is [deprecated ](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-action-sensor-deprecated) So I will check that out.

For now the workaround is to enable Home Assistant legacy action sensors.
Zigbee2MQTT > Cog symbol at top > Settings > Home Assistant integration > Check "Home Assistant legacy action sensors"

closed 2025-01-04 by rutgerh