Koenkk/zigbee2mqtt #25493: Since the update to 2.0.0 my Aqara motion sensors aren't showing illuminance in HA anymore

Fridolin34, 2025-01-04:
### What happened?

I dared to update, but now all automations relying on the brightness in the room aren't working anymore.

btw: for a Hue sensor, the attribute was only renamed, so the ':lux' was missing. This was easy to repair. But for all Aqaras the attribute is simply not shown in HA anymore. In Z2M it is still there.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20201128

### Adapter

CC2531

### Setup

HA on docker

Tinuz97, 2025-01-04:
I did not test it yet (also in the middle of the upgrade) ,but i did read that you have to enable that entity again, hope it helps.

m0wlheld, 2025-01-04:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0 - read the linked "Breaking Chances" issue. It's the one marked in red, because it's important. 

If you have disabled auto activation in the system options of the Home Assistant MQTT integration, you need to activate them manually.

Fridolin34, 2025-01-04:
Thanks for pointing  me to "enable the entity" ;)
Finally I figured out, what you were trying to tell me, went to the entity in HA, found that next to "motion detected" there was another attribute and it said I could enable this "entity". I did this for every of my sensors and after a couple of seconds, all values were back.

I still do not know, why this was needed and reading the release notes didn't help me either. I did not disable any auto activation (at least not intentionally), so I was clueless.

closed 2025-01-04 by Fridolin34