Koenkk/zigbee2mqtt #25495: After updating from 1.42.0-2 no re-start

hytrek, 2025-01-04:
### What happened?

Hi there... deepest applogies to start with... I am no programmer and relatively new to HA and have no clue yet. I tried upgrading to the 2.0.0-1 vom 1.42.0-2... I was reading the info to possible make changes to the configuration.yaml and my config looks very different than the suggested changes ... after digging for it I found the yaml and below is what is in there. If anybody would be kind enough and able to help me make the required changes I would be very happy - as the installed version 2.0.0-1 does not start... I would highly appreciate if someone would be taking my hand and grandpa wise instruct me step by step what to do... again. I am a dummie and require slow little steps - I read the errors in the log file but I am not able to understand what to change... where ... MANY THANKS IN ADVANCE

After the installing the update zigbee2mqtt was not running and when attempting to start it "502: Bad Gatewy" is displayed.
here is my full config.yaml... I shortened the password and the list of devices

homeassistant:
enabled: true
advanced:
homeassistant_status_topic: homeassistant/status
network_key:
- 60
- 86
- 81
- 40
- 167
- 116
- 169
- 20
- 76
- 134
- 130
- 183
- 105
- 145
- 92
- 133
pan_id: 7749
ext_pan_id:
- 85
- 136
- 98
- 63
- 1
- 87
- 204
- 201
log_level: debug
mqtt:
server: mqtt://core-mosquitto:1883
user: addons
password: myPW
keepalive: 60
reject_unauthorized: true
version: 4
frontend:
enabled: true
port: 8099
serial:
port: tcp://192.168.2.218:6638
baudrate: 115200
device_options: {}
devices:
'0x00158d0006bb71fa':
...

: []
version: 4

Here is the log entry from HA:
[14:09:54] INFO: Preparing to start...
[14:09:54] INFO: Socat not enabled
[14:09:54] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 14:09:55] debug: z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-04.14-09-38'
[2025-01-04 14:09:55] info: z2m: Logging to console, file (filename: log.log)
[2025-01-04 14:09:55] debug: z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-04 14:09:55] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 14:09:55] debug: z2m: sd-notify loaded
[2025-01-04 14:09:55] info: z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 14:09:55] debug: z2m: Using zigbee-herdsman with settings: '"{"network":{"panID":7749,"extendedPanID":[85,136,98,63,1,87,204,201],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"path":"tcp://192.168.2.218:6638"},"adapter":{"disableLED":false}}"'
[2025-01-04 14:09:55] error: z2m: Error while starting zigbee-herdsman
[2025-01-04 14:09:55] error: z2m: Failed to start zigbee-herdsman
[2025-01-04 14:09:55] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 14:09:55] error: z2m: Exiting...
[2025-01-04 14:09:55] error: z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
at Zigbee.start (/app/lib/zigbee.ts:69:47)
at Controller.start (/app/lib/controller.ts:142:13)
at start (/app/index.js:161:5)

### Zigbee2MQTT version

1.42.0-2 to 2.00.

### Adapter firmware version

20240315	v2.5.8

### Adapter

SLZB-06P7

### Setup

HAS OS on Proxmox

m0wlheld, 2025-01-04:
1. please edit this posting, as you are exposing security relevant data of your Zigbee network
2. please read as suggested by the error message you copied:

> [85,136,98,63,1,87,204,201],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/conf [...]

Also check the release notes at https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0 regarding "breaking changes" (which you are facing)

hytrek, 2025-01-04:
Thanks @m0wlheld ... 
As written, I read the error messages but was totally unable to underdtand what I should do... I would require some "let me take you hand and guide you" support...

peros550, 2025-01-04:
how can we downgrade the version back to previous stable? I am running it as HA addon.

hytrek, 2025-01-04:
@peros550 when I installed the update I was asked to make a backup which I did and I was able to "go Back" after the update did not work for me. So have you chekced you Backups if there is any?

m0wlheld, 2025-01-04:
> I am running it as HA addon.

Check for a thread in the [Home Assistant community](https://community.home-assistant.io/new), I am sure there will be at least one.

hytrek, 2025-01-04:
@m0wlheld btw. love ur username :-)
Well I was thinking I am right here... This is the thread for this right? Anyway... I was hoping for some hand here

m0wlheld, 2025-01-04:
> Thanks @m0wlheld ... As written, I read the error messages but was totally unable to understand what I should do... I would require some "let me take you hand and guide you" support...SORRY And thanks for helping

I am sorry, but this is not the Home Assistant Add-On repository, you should look for an appropriate thread in the [Home Assistant](https://community.home-assistant.io/new) forum. 

In general, you need access to the `configuration.yaml` file of Zigbee2MQTT and add some entries mentioned by https://github.com/Koenkk/zigbee2mqtt/discussions/24364

hytrek, 2025-01-04:
@m0wlheld I found the access of the yaml and in my first post I coupied it (see above). But where shall I add what? That might be easy for someone that is experienced... so not me at this time... Would you mind helping me telling: wite this there and that there?

m0wlheld, 2025-01-04:
> @m0wlheld btw. love ur username :-) 

Danke.

> Well I was thinking I am right here... This is the thread for this right? Anyway... I was hoping for some hand here

This is the issue tracker for Zigbee2MQTT, a software that is packed into an Home Assistant add-on, but can also be used standalone. The required steps prior to upgrading Zigbee2MQTT 1.x to 2.0 have been enumerated in the release notes and the lines issues there.

The steps are basically identical for the Home Assistant add-on, but access to the configuration may be different. Assistance may be available at https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

hytrek, 2025-01-04:
@m0wlheld well than no hint for me? You suggest I copy my ticket to hassio-zigbee2mqtt github?

peros550, 2025-01-04:
> @peros550 when I installed the update I was asked to make a backup which I did and I was able to "go Back" after the update did not work for me. So have you chekced you Backups if there is any?

strangely, I followed your idea, found the backup, now it says I am on 1.42 but the addon can't start. I 'll look for it more online. 
many thanks

hytrek, 2025-01-04:
> https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/guide/configuration/adapter-settings.md

But it worked with that yml on the 1.42 so I am stuck what to add

m0wlheld, 2025-01-04:
> > https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/guide/configuration/adapter-settings.md
> 
> But it worked with that yml on the 1.42 so I am stuck what to add

Post the current content of the "serial:" section. Please use code formatting ([image]), so indendation stays intact.

hytrek, 2025-01-04:
> > > https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/guide/configuration/adapter-settings.md
> > 
> > 
> > But it worked with that yml on the 1.42 so I am stuck what to add
> 
> Post the current content of the "serial:" section. Please use code formatting

`
serial:
  port: tcp://192.168.2.218:6638
  baudrate: 115200`

hytrek, 2025-01-04:
`serial:
  port: tcp://192.168.2.218:6638
  baudrate: 115200`

m0wlheld, 2025-01-04:
try:

```yaml
serial:
    port: tcp://192.168.2.218:6638
    baudrate: 115200
    adapter: zstack
```

hytrek, 2025-01-04:
Adding adapter: zstack in serial worked! THANKS to @m0wlheld

closed 2025-01-04 by hytrek