Koenkk/zigbee2mqtt #25496: IKEA E2001/E2002 Aktion Exposes is missing

Goldi5, 2025-01-04:
### What happened?

Hello,

since the update to 2.0.0 and HA 2025.1.0 the Exposes action is missing in IkeaE2001/E2002. All 3 other Exposes are also displayed correctly. The switching behavior is also displayed in Z2M, as well as in the MQTT broker. Unfortunately, HA no longer sees the action sensor and therefore cannot accept any switching commands. Deleting and reconnecting the device has not brought any improvement. Everything worked very well before. 

Many thanks in advance. 

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.4

### Adapter

EZSP with ember

### Setup

Add-on on Home Assistant OS on Intel NUC

m0wlheld, 2025-01-04:
Yes. Expected behaviour, required actions are mentioned in the releasen notes. See https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0

tivoo, 2025-01-04:
Same issue here. Not able to find how to resolve this.

> Yes. Expected behaviour, required actions are mentioned in the releasen notes. See https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0

I don't see anything mentioned when I search for Action through the release notes. ANy link to point out the specific release notes you mention?

m10x, 2025-01-04:
Does someone have a working blueprint?

dpgh947, 2025-01-04:
Go to z2m screen (from lefthand menu), select settings under the cog wheel at the top, then the home assistant tab, there is a tickbox there to re-enable the legacy action entities:

[image]

tivoo, 2025-01-04:
> Go to z2m screen (from lefthand menu), select settings under the cog wheel at the top, then the home assistant tab, there is a tickbox there to re-enable the legacy action entities: [...]

Is not recommended though. Does anyone have a working blueprint? I use: https://community.home-assistant.io/t/zha-deconz-zigbee2mqtt-ikea-e2001-e2002-styrbar-remote-control-universal-blueprint-redux/734435

dpgh947, 2025-01-04:
I don't care if it is not recommended. It's there, it works, and is a quick and easy fix without more work for now, especially if you use node red.

Goldi5, 2025-01-04:
Thank you very much for your answers. The short-term solution helped and I had skimmed the changelog, but not that intensively. 

I will take a closer look at the topic of Home Assistant event entity and MQTT device trigger.

https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-event-entity-experimental

Thanks again.

closed 2025-01-04 by Goldi5

m10x, 2025-01-04:
Here is an updated blueprint for the E2001/E2002 https://github.com/lsismeiro/awesome-ha-blueprints/blob/b4219cab2f7769c23ac6cca803a3aabf972cdf37/blueprints/controllers/ikea_e2001_e2002/ikea_e2001_e2002.yaml

However, It says "No matching device found" for me.

hytrek, 2025-01-04:
re-enable the legacy action entities is poosible but does not do the trick for me. I can now select the action bt still it does not switch (in my case the lamps) on or off. Even adding a new blueprint using the now re-enabled legacy option does not work...

tivoo, 2025-01-05:
> Here is an updated blueprint for the E2001/E2002 https://github.com/lsismeiro/awesome-ha-blueprints/blob/b4219cab2f7769c23ac6cca803a3aabf972cdf37/blueprints/controllers/ikea_e2001_e2002/ikea_e2001_e2002.yaml
> 
> However, It says "No matching device found" for me.

Same for me unfortunately.

legantois, 2025-01-05:
WTH ?

I suffer so hazardous behavior with this V2.0.0 

What is the easiest way to roll back to the "old Working perfectly"  version ?

TheFtg, 2025-01-12:
For me, I reverted to version 1.42.0-2. The restoration also worked perfectly: well done guys ! You have to go to manual backups, find the right version and install it. 

Now, to go to version 2, it's not clear to me. Should I wait for zigbee2mqtt fix ? Do I need a new version of the blueprint?