Koenkk/zigbee2mqtt #25500: Home Assistant no longer acts on button presses on Hue dimmer after upgrade to 2.0.0

unikitty37, 2025-01-04:
### What happened?

After upgrading to 2.0.0, my Home Assistant automations no longer fire when I press a button on any Hue dimmer (version 1). This worked before the upgrade.

The Zigbee2MQTT log shows it is receiving the button presses:

```
[2025-01-04 14:37:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch', payload '{"action":"on_press","action_duration":null,"battery":58,"brightness":255,"counter":1,"linkquality":126,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
[2025-01-04 14:37:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch/action', payload 'on_press'
[2025-01-04 14:37:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office Hue dimmer switch","ieee_address":"0x0017880106730369"},"type":"device_announce"}'
[2025-01-04 14:37:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch', payload '{"action":"on_press_release","action_duration":null,"battery":58,"brightness":255,"counter":1,"linkquality":123,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
[2025-01-04 14:37:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch/action', payload 'on_press_release'
[2025-01-04 14:37:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch', payload '{"action":"on_press","action_duration":null,"battery":58,"brightness":255,"counter":1,"linkquality":123,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
[2025-01-04 14:37:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch/action', payload 'on_press'
[2025-01-04 14:37:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch', payload '{"action":"on_press_release","action_duration":null,"battery":58,"brightness":255,"counter":1,"linkquality":123,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
[2025-01-04 14:37:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Hue dimmer switch/action', payload 'on_press_release'
```

but nothing is seen by HA — the status here does not change when I press the button.

[image]

The birth message topic is `homeassistant/status`, per the release notes, and I have applied the config changes and restarted HA. I *can* control the Zigbee Hue lightbulbs from the HA interface, so Zigbee2MQTT seems to be talking to HA.

I noticed that my VALLHORN motion sensors had the "illuminance" sensor marked as disabled in HA and I had to re-enable them to get light level detection working, as I was using the now-removed "-lux" versions. But there doesn't seem to be anything disabled on the dimmer apart from "Linkquality", "Update state", and a binary sensor to show whether an update's needed — so there's nothing I can enable that might get this working again.

### What did you expect to happen?

The update did not break this, or the steps necessary to fix it were listed in the Breaking Changes section of the release notes.

### How to reproduce it (minimal and precise)

I'm not sure how to write this, TBH — I'm assuming that it's something specific to my setup, or it would probably have been spotted :)

* Have several version 1 Hue dimmers on Zigbee2MQTT version 1.4 that Home Assistant can see and trigger automations from; HA should already have the birth message topic set correctly for 2.0.0. Do *not* make any of the other config changes suggested in the release notes.
* Update to Zigbee2MQTT version 2.0.0
* Make the changes to configuration.yaml as recommended in the release notes.
* Restart Zigbee2MQTT and Home Assistant.

Sadly, these steps are probably not very helpful :(

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

20210708

### Adapter

Sonoff Universal Zigbee 3.0 (https://www.amazon.co.uk/gp/product/B09KXTCMSC)

### Setup

Add-on on Home Assistant on a Home Assistant Green

### Debug log

[log.log](https://github.com/user-attachments/files/18307246/log.log)

tivoo, 2025-01-04:
Same issue with my STYRBAR remote control on Z2M. It seems like the action entity is not available in Home Assistant anymore.

vincentbattez, 2025-01-04:
the `.action` sensor was deprecated feature (see https://github.com/Koenkk/zigbee2mqtt/pull/24494 and https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-action-sensor-deprecated)

I found this answer: 
https://community.home-assistant.io/t/using-the-new-action-events-in-zigbee2mqtt-2-0/821709/12?u=vincentbattez

now we must use:
- [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)
- or [Home Assistant event entity](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-event-entity-experimental)

unikitty37, 2025-01-04:
@vincentbattez Thanks — I saw that. However, the instructions are all for YAML-wallopers. Home Assistant has made great strides in making it so you can configure it from the GUI and not have to edit YAML, so this is a major step backwards.

MQTT device triggers require forgetting the nice, readable names you gave the devices, and instead referring to them by meaningless hex strings. Not usable.

Event entities are marked as "experimental", which means breaking changes could drop at any time, or the whole thing could just get binned.

This makes Home Assistant considerably more of a pain to work with. It's a huge backwards step.

I would love to see instructions on how to roll the addon back to the most recent version of v1; I can then turn off updates for it and keep my easy-to-use and easy-to-edit setup until things improve. But I can't find them anywhere.

vincentbattez, 2025-01-04:
If you use like me `zigbee2mqtt` and `Node-RED` you can change all your button like this:

## Option 1: [Home Assistant event entity](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-event-entity-experimental) (experimental)

### In `zigbee2mqtt`:

- check `Home Assistant experimental event entities`
[image]

### `in Node-RED`
- Before:
[image]

- After:
[image]

## Option 2: [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)

- Before
[image]

- After
[image]
[image]

m0wlheld, 2025-01-04:
> @vincentbattez Thanks — I saw that. However, the instructions are all for YAML-wallopers. Home Assistant has made great strides in making it so you can configure it from the GUI and not have to edit YAML, so this is a major step backwards.

Yeah, maybe. You should address this to the developers of the Zigbee2MQTT add-on, their repository is [here](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt).

lutzbellmann, 2025-01-04:
same issue for me! I am not a programmer and I can't control my lights anymore now! How can I roll-back the update?

m0wlheld, 2025-01-04:
> same issue for me! I am not a programmer and I can't control my lights anymore now! How can I roll-back the update?

There should be instructions at Home Assistant, maybe ask for assistance at the [Home Assistant forum](https://community.home-assistant.io) or the Home Assistant Zigbee2MQTT Add-On repository at https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

lutzbellmann, 2025-01-04:
> There should be instructions at Home Assistant, maybe ask for assistance at the [Home Assistant forum](https://community.home-assistant.io) or the Home Assistant Zigbee2MQTT Add-On repository at [...]

Many thanks, I should think before writing. Under one of the links above is a description. You can keep the version 2 and activate "Homeassistant legacy action sensors" under the setting Z2M and then "Home Assistant Integration". But who knows for how long this will be working.

sandervandegeijn, 2025-01-04:
Same here, can't pinpoint it exactly, but it seems it's using friendly names instead of the addresses in the 2.0 upgrade. Trying to rollback from my restic backups now.

WillCodeForCats, 2025-01-04:
This blueprint I made for myself to duplicate how the Hue remote would function with a Hue hub works with Zigbee2MQTT legacy options disabled.

https://gist.github.com/WillCodeForCats/1573e52ee3cf45f7e269635b163fb2d1

Look at the trigger section for examples on how to use MQTT platform triggers for the Hue dimmer switch.

> MQTT device triggers require forgetting the nice, readable names you gave the devices, and instead referring to them by meaningless hex strings. Not usable.

I agree on that point, so at least when using blueprints that aspect is hidden for all of the triggers.

sandervandegeijn, 2025-01-04:
Okay, what works for me for now:

- remove the friendly_name from the configuration.json of zigbee2mqtt
- remove the device from home assistant manually (otherwise the friendly name still is present in the device id)
- restart zigbee2mqtt

m0wlheld, 2025-01-04:
> In general I'm using the addresses instead of the friendly names and just set the home assistant name in zigbee2mqtt. That way if a device changes name all of my automations will still work.

Not sure about the HA add-on, but with plain Zigbee2MQTT (which is adressed by this repository) there is an option to set a "Home Assistant Name" per device. This is not the friendly name but used as display name by Home Assistant. It is a per-device configuration, resulting in the following structure:

```yaml
devices:
  '0xcafebabe':
    homeassistant:
      name: "Display Name"
```

see https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#generic-device-options

sandervandegeijn, 2025-01-04:
Yip I know, I suspect I had some friendly names from the past when that option wasn't there. Maybe it was inconsistent and now it is, but damn this is a lot of work to fix.

m0wlheld, 2025-01-04:
@unikitty37 Is your issue addressed and ready to close?

unikitty37, 2025-01-05:
@m0wlheld I resolved this by rolling back to 1.14-0.2 and keeping at that version — but none of the other solutions is really a solution for me, as there isn't support in the Home Assistant GUI for configuring triggers that way; you have to edit the XML manually.

I think the _real_ issue is the removal of the click sensors, which causes a great deal of work for anyone using Home Assistant, especially those configuring it via the GUI. I've looked at #24198 and the wiki and I still can't find out _why_ these were removed, apart from someone having declared them as "legacy" (legacy they may be, but they work and are easier to use!)

Is there another open issue covering that removal? If so, then this should be OK to close in favour of that.

m0wlheld, 2025-01-05:
> Help me understand how can they remove and depreciate a thing without fixing the buttons first? I have like 10 switches and all stopped working. I could literally see the LQI, battery and even Voltage, but as a button it stopped working, there were no alternative new entities, they just removed the action without a replacement, why?
> 
> https://www.zigbee2mqtt.io/devices/WXKG01LM.html#deprecated-click-event I removed click for action and now the page still says use action and they removed it all together, why?

You are complaining in the wrong place, post your question at the Home Assistant Community forum or the HASSIO Zigbee2MQTT add-on repository.

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-05 by Koenkk

jeroenbaas, 2025-01-05:
> If you use like me `zigbee2mqtt` and `Node-RED` you can change all your button like this:
> 
(https://github.com/Koenkk/zigbee2mqtt/issues/25500#issuecomment-2571344098)

The issue leaves anyone who has buttons after auto-updates and who's been using HA a few years probably confused. I have a 6 button with double/single/triple/hold/release states which means reconfiguring a lot of flow, but it seems @vincentbattez 2nd solution is quite close - the only thing I'd add would be a template node right after the mqtt to take the payload.action and forward that as payload, that way you have a drop-in replacement and no need to change the flow downstream.

[image]
In the example, the orphan node is the old button action, the two below it are the new nodes. The template is just and empty template node with {{payload.action}} in it. That's all. The rest of the flow is just exactly as it was before.

AF360, 2025-01-06:
> ### What happened?
> After upgrading to 2.0.0, my Home Assistant automations no longer fire when I press a button on any Hue dimmer (version 1). This worked before the upgrade.
>
> The Zigbee2MQTT log [...]

@unikitty37 
Double check your automations. I was running into the same issue with my Hue Dimmer switches and noticed that in my automation I was triggering for action "on-press" and apparently this is now no longer on-press but on_press. Once I changed that in the automation it worked again right away.

elgeniskogen, 2025-01-06:
> > If you use like me `zigbee2mqtt` and `Node-RED` you can change all your button like this:
> 
> ([#25500 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25500#issuecomment-2571344098))
> 
> The issue leaves anyone who has buttons after auto-updates and who's been using HA a few years probably confused. I have a 6 button with double/single/triple/hold/release states which means reconfiguring a lot of flows, but it seems @vincentbattez 2nd solution is quite close - the only thing I'd add would be a template node right after the mqtt to take the payload.action and forward that as payload, that way you have a drop-in replacement and no need to change the flow downstream.
> 
> ![image](https://private-user-images.githubusercontent.com/487600/400256961-b9ad58fe-e7e4-4682-8048-fc8b6ebbd24e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzMwNzEsIm5iZiI6MTczNjE3Mjc3MSwicGF0aCI6Ii80ODc2MDAvNDAwMjU2OTYxLWI5YWQ1OGZlLWU3ZTQtNDY4Mi04MDQ4LWZjOGI2ZWJiZDI0ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQxNDEyNTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYzE4NWY3YmIwZGVkNTFkMGIwODRkZTkzMjBmNjA3ZDRkMjU4Mzg5YjRkNGQ0NTBmNzM5MWZkNTM4YTcwN2MwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.JOEWtpPVtoEpNW9k4zvwnl0WLzzuquL0aes1FFU3Zaw) In the example, the orphan node is the old button action, the two below it are the new nodes. The template is just an empty template node with {{payload.action}} in it. That's all. The rest of the flow is just exactly as it was before.

Would adding "action" to the topic do the same? 

[image]

JohnnyM84, 2025-01-06:
> {{payload.action}}

Is this way a single double and tripple click possible? because for me with version 2.0 the counters stayed 2 and not 1,2 or 3.

jeroenbaas, 2025-01-06:
> > {{payload.action}}
> 
> Is this way a single double and tripple click possible? because for me with version 2.0 the counters stayed 2 and not 1,2 or 3.

I don't know what you mean by counters; my buttons expose single and double clicks through the action as a string. For instance, double clicking button_1 on my 6 button device gives me action: button_1_double. double clicking my single button aqara device just gives action "double" etc. Not keeping any counters or anything. Easiest is to press the button and look at the zigbee2mqtt log at the messages.

elgeniskogen, 2025-01-06:
> > Would adding "action" to the topic do the same?
> 
> I guess only if you have the "Home Assistant legacy action sensors" setting checked in MQTT, at which state there was already no reason to change any of your node red flows as you had backwards compatibility? at least my assumption is that the zigbee2mqtt/{device}/action topic only exists by the grace of that setting, but I could be wrong.

I have turned off "Home Assistant legacy action sensors"  (never had it on) and still get the action in the MQTT topic. But isn't that exactly what you attract with the {{payload.action}} ?!

[image]

jeroenbaas, 2025-01-06:
I think you need a full reboot of HA to see the effect of it, at least it was when doing it the other way for me (not in a position to reboot now).

look at these log lines, all caused by a single button press:
[image]
your solution tries to access row 3 there (the _topic_ z2m/device/action)
the template solution access the other topic: z2m/device, and from the _payload_ takes the .action property and forward that as the new payload. which has the same effect without needing that backwards-compatible topic.

elgeniskogen, 2025-01-06:
> I think you need a full reboot of HA to see the effect of it, at least it was when doing it the other way for me (not in a position to reboot now). look at these log lines, all caused by a single button press: ![image](https://private-user-images.githubusercontent.com/487600/400469056-0e802e82-f52d-410b-ae1b-80a249791860.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzgzNjQsIm5iZiI6MTczNjE3ODA2NCwicGF0aCI6Ii80ODc2MDAvNDAwNDY5MDU2LTBlODAyZTgyLWY1MmQtNDEwYi1hZTFiLTgwYTI0OTc5MTg2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQxNTQxMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZDljODg0MTkyNGI0ZGRkNzdiNTgzMGFhNjE4YTAzODAzOWVlZDE2MThjOTViOWNkMDdkNDViZjYzMGU2N2Q3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zfcO_esPe4U-nxWAyBlgMc2NKQBuji_fR-z8p11vJk0) your solution tries to access row 3 there (the _topic_ z2m/device/action) the template solution accesses the other topic (line 1): z2m/device, and from the _payload_ takes the .action property and forward that as the new payload. which has the same effect without needing that backwards-compatible topic.
> 
> EDIT: I stand corrected. The z2m/device/action topic is still there. I have no idea what the "Home Assistant legacy action sensors" check does if that action is still there as a topic when unchecked. makes no sense to me.

Strange. Could it be for the integration to know it should make an event.device entity or a sensor.device_action entity in Home Assistant? My MQTT looks like this in MQTT explorer: 
[image]

JohnnyM84, 2025-01-06:
> I don't know what you mean by counters; my buttons expose single and double clicks through the action as a string. For instance, double clicking button_1 on my 6 button device gives me action: [...]

I have this hue dimmer switch:
https://www.zigbee2mqtt.io/devices/324131092621.html

And it reports no double or triple, only on, off, up and down and the hold of these 4. 
for a double or triple it uses a counter that is not there anymore in v2.
I still see it in mqtt but value is always 2 and not 1 or 3:
This is the mqtt message:
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dimmer Switch Woonkamer', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":191,"counter":1,"duration":0,"last_seen":"2025-01-06T17:22:51+01:00","linkquality":69,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'

So I thought to use mqtt instead with v2 but the counter does not report good values so I cannot have a double of triple click. For now I downgraded to v1.42

MaoZ1993, 2025-01-07:
[WXKG01LM](https://www.zigbee2mqtt.io/devices/WXKG01LM.html#xiaomi-wxkg01lm) 

Mi wireless switch lost the action entity as well after 2.0.0 update. 
I have tried to enable or disable Home Assistant legacy action sensors and reboot the entire HA, nothing works.

I think this is also related to this thread

jeroenbaas, 2025-01-07:
@JohnnyM84 
> "counter":1,

seems to suggest it is not 2? I don't know the button you have but I suggest clearing the log, fiddle with it and keep an eye on your logs to figure out how it gets set. What tools are you using to read and act on the message? Node-red?

JohnnyM84, 2025-01-07:
> @JohnnyM84
> 
> > "counter":1,
> 
> seems to suggest it is not 2? I don't know the button you have but I suggest clearing the log, fiddle with it and keep an eye on your logs to figure out how it gets set. What tools are you using to read and act on the message? Node-red?

this was al log of version 1.42, i'm using indeed node-red and i know how it works.
But with version 2.0 there is no counter anymore in the message.
So i thought i do it with mqtt, because thats recommended in the breaking changes documentation, but there the counter is always 2.
With version 1.42 i get it working in nodered with subscribing to the topic of the dimmer switch and then 2 switch nodes:
[image]
[image]

jeroenbaas, 2025-01-07:
Maybe open another issue since this one is closed by the owner and your issue seems different from the one reported here.

JohnnyM84, 2025-01-07:
> Maybe open another issue since this one is closed by the owner and your issue seems different from the one reported here.

i have already one: https://github.com/Koenkk/zigbee2mqtt/issues/25559
but no reaction yet, and this problem looked the same, but was eventually not.

louisdmueller, 2025-04-29:
If you use the remote for simple things like turning off/on a light, this automation is enough:
```yaml
triggers:
  - trigger: template
    value_template: >-
      {{ state_attr("event.remote_action", "event_type") ==
      "press_release" }}
conditions:
  - condition: template
    value_template: "{{ state_attr("event.remote_action", "button") == "on" }}"
actions:
  - action: light.toggle
    data: {}
    target:
      device_id: 3f923fd1028bc95959da74b15351b348
```