Koenkk/zigbee2mqtt #25502: IKEA E1743 (Dimmer Switch) does not expose action in HA

thogens, 2025-01-04:
### What happened?

After updating to new Zigbee2Mqtt 2.0 version, the old click event got removed (I knew that).
Within Zigbee2Mqtt Addon, I can see the action, but those are not exposed to any entity.
I also notice that in the logs of Zigbee2Mqtt, there is no payload published.
What can I do?

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

Sonoff P

### Setup

HomeAssistant OS on Synology

alvoram, 2025-01-04:
Go to the "Home Assistant" tab of the Zigbee2Mqtt settings page, and enable "Home Assistant legacy action sensors"

Zigbee2MQTT > Pres Cog At Top > Settings > Home Assistant Tab > Enable "Home Assistant legacy action sensors"

thogens, 2025-01-04:
Works perfectly. Thanks!

closed 2025-01-04 by thogens

alvoram, 2025-01-04:
You're welcome, just sharing what others have already shared when I found myself in your shoes, as they seem to be inundated with the same issue / request at the moment, and struggling to reply to all. :)

kopaszsop, 2025-01-29:
> Go to the "Home Assistant" tab of the Zigbee2Mqtt settings page, and enable "Home Assistant legacy action sensors"
> 
> Zigbee2MQTT > Pres Cog At Top > Settings > Home Assistant Tab > Enable "Home Assistant legacy action sensors"

sadly not for me.
i have tried all the variatons of the checkboxes. at least have "event" domain in HA with on/off/up/down/release of the switch but have to check all my automations and scenes because of this change.
or is it a workaround for it?