Koenkk/zigbee2mqtt #25505: IKEA E1812 - TRADFRI shortcut button stopped working

ojkippen, 2025-01-04:
### What happened?

IKEA E1812 - TRADFRI shortcut button stopped working after upgrade to 2.0. 

### How to reproduce it (minimal and precise)

Forcing leave and then join doesn't fix the problem. Join fails.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

With Homeassitant 2025.1

alvoram, 2025-01-04:
Go to the "Home Assistant" tab of the Zigbee2Mqtt settings page, and enable "Home Assistant legacy action sensors"

Zigbee2MQTT > Pres Cog At Top > Settings > Home Assistant Tab > Enable "Home Assistant legacy action sensors"

ojkippen, 2025-01-04:
That helped. Working again. Thanks

alvoram, 2025-01-04:
You're welcome. I had same issue, and saw the answer elsewhere, but it seems so many people are having this issue that the people are struggling to keep up with responding, so thought I'd help a little. 

For anybody else having a similar issue, the .action sensor has been deprecated, in the long term, you need to change all of your automations to use instead the MQTT device trigger, but for now, to keep things working, you can just enable .action again using the legacy setting mentioned above. 

https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-action-sensor-deprecated

Rabiat1963, 2025-01-04:
@alvoram This also brought back the action button on an Ikea and an Aqara push button. Saved me a lot!

alvoram, 2025-01-04:
@Rabiat1963 Awesome

It works on the sonoff devices that previously used .action too, it bought my Sonoff SNZB-01P buttons back to life. 

I 'think' it will work on anything that is missing .action since the update.

thomasaarholt, 2025-01-05:
> in the long term, you need to change all of your automations to use instead the MQTT device trigger

@alvoram could you expand on what that means for e.g. an ikea switch? I spent a day without functioning light switches (which was not great for my partner 😅), so am hesitant to try Z2M again without a clear path forward. I’d really appreciate any help!

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-05 by Koenkk

alvoram, 2025-01-05:
@thomasaarholt I'm no expert, like I said up top, I'm just re-sharing what I'd seen elsewhere. But I think it just means you need to change the triggers used in each automation from the button .action trigger to the MQTT device trigger. @Koenkk has shared the correct resource, but here it is again... https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended

I haven't done it myself yet, but I did notice that the guide seems to imply you need to physically press all of the buttons on your device for MQTT to discover and expose them under mqtt device triggers. 

Also, I'm not sure how this is going to affect the Styrbar blueprint I use, I 'think' it breaks it, and I'll need to either find another blueprint, or just create my own automation for each remote. 

When I get around to actually trying it myself, I'll report back, but for now I'm just going to use 'Legacy' until I'm forced to change, as it's working well, and changing seems like a hell of a lot of work.