Koenkk/zigbee2mqtt #25507: V2 update not possible bare-metal config

Jeepmg, 2025-01-04:
### What happened?

 **git pull --no-rebase**

Updating bc6eccf6..060ae99c
error: Your local changes to the following files would be overwritten by merge:
	data/configuration.example.yaml
Please commit your changes or stash them before you merge.

Aborting

### What did you expect to happen?

to have a uptade.sh running the V2 update 

to know how to upgrade 

do I do git pull --rebase ?

### Zigbee2MQTT version

1.42

### Adapter firmware version

202407

### Adapter

zstack

### Setup

bare-metal

m0wlheld, 2025-01-04:
You (unintentionally) did changes to file `data/configuration.example.yaml` that would be overriden by `git pull` executed by `update.sh`. Revert those changes running `git checkout data/configuration.example.yaml`, then run `update.sh` again.

aperezva, 2025-01-04:
Same error, and I tried to do the instructions to saove it, and I have the same error in my both Lxc in proxmox

Jeepmg, 2025-01-04:
OK for me. thank you for your halp.
However I noticed that my groups.yalm file only contains the header, the list of devices has disappeared. Should I put it back? It works without it.

m0wlheld, 2025-01-04:
It‘s intended 

> Removed configuring group members through configuration.yaml (groups.devices setting). This will not impact current group members; however, you will no longer be able to add or remove devices from a group through the configuration.yaml.

dsiebel, 2025-01-04:
See #25517.

closed 2025-03-13 by github-actions[bot]