Koenkk/zigbee2mqtt #25509: BHT-002/BHT-006 sometimes sends invalid data

stefanpratsch, 2025-01-04:
### What happened?

I am using a "BHT-002/BHT-006" and from time to time I get crazy invalid data points in my reporting.

Zigbee-Modell: TS0601
Zigbee-Manufacture: _TZE204_aoclfnxz
Manufacture: Moes
Modell: BHT-002/BHT-006

I checked the logs and I happens for the fields current_heating_setpoint and also local_temperature.

Here one example:

`"current_heating_setpoint":361321898,`

Full log block:

`[2024-12-15 16:05:21] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Thermostat Wohnzimmer', payload '{"child_lock":null,"current_heating_setpoint":361321898,"deadzone_temperature":null,"heat":"OFF","linkquality":47,"local_temperature":21.5,"local_temperature_calibration":-3,"max_temperature_limit":null,"min_temperature_limit":0,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"IN","system_mode":"heat"}'`

### How to reproduce it (minimal and precise)

No way to directly reproduce the behaviour. The indalid data packages are sometimes  received multiple times a day, but it also happens, that one day does not contains invalid data.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

ZBDongle-E

### Setup

Run on Raspberry PI

stefanpratsch, 2025-03-06:
I hope somebody has an idea

srichard95, 2025-04-21:
I am still experiencing this issue, i hope somebody has any idea whats going on

VerbalP, 2025-04-27:
Same issue here. Happens with two thermostats

[image]

closed 2025-09-04 by github-actions[bot]