Koenkk/zigbee2mqtt #25510: z2m 2.0 failed to load external converter which used to work

ehwwong, 2025-01-04:
### What happened?

After updating to version 2,0, my external converter for a tuya blind motor failed to load. 

### What did you expect to happen?

When can i load external converter again? 

### How to reproduce it (minimal and precise)

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
// Since a lot of TuYa devices use the same modelID, but use different datapoints
// it's necessary to provide a fingerprint instead of a zigbeeModel
fingerprint: [
{
// The model ID from: Device with modelID 'TS0601' is not supported
// You may need to add \u0000 at the end of the name in some cases
modelID: 'TS0601',
// The manufacturer name from: Device with modelID 'TS0601' is not supported.
manufacturerName: '_TZE200_xtrnjaoz',
},
],
model: 'TS0601_blinds',
vendor: 'TuYa',
description: 'Blinds motor',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
options: [exposes.options.invert_cover()],
exposes: [
e.cover_position().setAccess('position', ea.STATE_SET),
e.enum('reverse_direction', ea.STATE_SET, ['true', 'false'])
.withDescription('Reverse the motor direction'),
e.binary('motor_fault', ea.STATE, true, false),
],
meta: {
tuyaDatapoints: [
[1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
[9, 'position', tuya.valueConverter.coverPositionInverted],
[8, 'position', tuya.valueConverter.raw],
[11, 'reverse_direction', tuya.valueConverterBasic.lookup({'true': tuya.enum(0), 'false': tuya.enum(1)})],
[10, 'motor_fault', tuya.valueConverter.trueFalse1],
],
},
};

module.exports = definition;

### Zigbee2MQTT version

2.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

EmberZNet

### Setup

Add-on on home assistant

ehwwong, 2025-01-04:
I like to add that somehow my external converter js files are still in the /zigbee2mqtt directory and no new directory has been created.. 

should i create a directory /zigbee2mqtt/data/external_converters ?

any help is apppreciated. for now i have rolled back to my backup

m0wlheld, 2025-01-04:
Luckily, this i covered in the [Release Notes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198) at section "External converters and extensions".

dane-io, 2025-01-05:
@m0wlheld I just checked the release notes and that fix did not work for me. A directory called external_converters (in Home Assistant, this looks like zigbee2mqtt/external_converters) instead of data/external_converters needed to be made.

Not sure if this is only a Home Assistant issue, but it should probably be noted since a significant amount of Z2M users are running it as a Home Assistant add-on.

ehwwong, 2025-01-05:
i am running HAOS and z2m as a add-on.. will be willing to do some test if point at the correct direction. 

should i manually create a folder call zigbee2mqtt/external_converters ?

dane-io, 2025-01-05:
@ehwwong I think so, that's what I did and was successful. I attempted to use the MQTT method described here:

[https://www.zigbee2mqtt.io/advanced/more/external_converters.html#save-converter](https://www.zigbee2mqtt.io/advanced/more/external_converters.html#save-converter)

But I kept getting "invalid payload" errors from Zigbee2MQTT. Adding that folder manually, adding my converters in there manually, and restarting Zigbee2MQTT worked for me.

ehwwong, 2025-01-05:
thanks.. maybe i just stick to my back up until there is official word

dane-io, 2025-01-05:
@ehwwong that's fine, everything seems to be working properly on my end. What I did is almost identical to the official recommendation in the release notes linked previously. I'm assuming that the /zigbee2mqtt folder is named /data in a standalone installation and that's the difference. Shouldn't hurt to try

Mikescotland, 2025-01-05:
Had similar problem.
Remove external converters from z2m configuration.yaml and if using z2m addon in HA crwatw directory zigbee2mqtt/external_converters (not data as in the guide). Restart z2m.

steigerbalett, 2025-01-05:
This also works for me:
### Zigbee2MQTT as Home Assistant Add-on

1. Create directory: /homeassistant/zigbee2mqtt/external_converters/
2. Move custom converters to the new directory
3. Remove: `fz.legacy.ptvo_switch_buttons, ` in the cusomconverter *.js files (located at `fromZigbee: `)
4. Restart Z2M

kds69, 2025-01-07:
> @m0wlheld I just checked the release notes and that fix did not work for me. A directory called external_converters (in Home Assistant, this looks like zigbee2mqtt/external_converters) instead of [...]

agree, tested in all ways, only a directory created as `addons_config/45df7312_zigbee2mqtt/zigbee2mqtt/external_converters` works for me

m0wlheld, 2025-01-07:
That's because you are referring to the Home Assistant Zigbee2MQTT Add-On for Home Assistant OS (AKA HASSIO). This repository is for "stock" Zigbee2MQTT, which is encapsulated by the add-on and embedded somewhere into the filesystem.

The add-on repository is over here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

ehwwong, 2025-01-17:
I was pretty busy the last 2 weeks and didnt had time to tinkle until today. I am happy to report that this solution works for me. 

1. Create directory: /homeassistant/zigbee2mqtt/external_converters/
2. Move custom converters to the new directory
3. Restart Z2M

Thanks for everyone who had provided suggestions and help.

closed 2025-01-17 by ehwwong