Koenkk/zigbee2mqtt #25511: Mi control switch doesn't work correctly after upgrade V 2.0

masciotti, 2025-01-04:
### What happened?

After 2.0 upgrade the device work in Z2M 
info 2025-01-04 17:52:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tasto Xiaomi 2', payload '{"action":"double","battery":null,"linkquality":252,"power_outage_count":null,"voltage":null}'
info 2025-01-04 17:52:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tasto Xiaomi 2/action', payload 'double'

But in HA only battery appears as entity.
Tasto Xiaomi 2 Battery
sensor.tasto_xiaomi_2_battery
MQTT

### Zigbee2MQTT version

2.0

### Adapter firmware version

7.4.3 (GA)

### Adapter

ember

### Setup

Pi

m0wlheld, 2025-01-04:
Duplicate of #25505, #25500 and may others. Please read the release notes and the references [Breaking Changes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

rogervdh, 2025-01-04:
I have the same problem and I don't use the deprecated sensors. The release notes are mentionning: "All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead.". I was already using the MQQ device triggers, but it doesn't work since Z2M 2.0.0. Please see below the triggering part of an automation:

```
alias: Lichtschakelaar | Keuken
description: ""
triggers:
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 1_single
    discovery_id: 0x3425b4fffef55a1e action_1_single
    id: links-boven-enkel
    alias: Links boven, enkel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 2_single
    discovery_id: 0x3425b4fffef55a1e action_2_single
    id: rechts-boven-enkel
    alias: Rechts boven, enkel drukken
    trigger: device
```

closed 2025-03-13 by github-actions[bot]