Koenkk/zigbee2mqtt #25512: Failed to send with status=BUSY

plaiddk, 2025-01-04:
### What happened?

After updating zigbee2mqtt to newest version my hue dimmer automation throws an error.

It gives this error:
z2m: Publish 'set' 'state' to 'Lille stue' failed: 'Error: Command 12 genOnOff.off({}) failed (~x~> [ZCL GROUP groupId=12] Failed to send with status=BUSY.)'

I have several areas attached to the off_press button (10 in total) and I have 2 areas attached to the on_press button.

The on_press button switches to light areas on. This works fine.
The off_press button switches them off, but it only works on like 1 or 2 of the areas. The rest gets the above error message.

### What did you expect to happen?

I expected the off_press to work as the on_press. It should switch on light for all areas

### How to reproduce it (minimal and precise)

Add more areas to the automation

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.4.0

### Adapter

ember

### Setup

Add-on on Homeassistant on a virtualbox

kennethslot, 2025-03-08:
I seems to have same issue, i have just move from the Skyconnect to SLZB-06m and startet to get these errors on groups

bjornfor, 2025-03-31:
Related: https://github.com/Koenkk/zigbee2mqtt/issues/24249 (people having same symptom, which was fixed with firmware upgrade).