Koenkk/zigbee2mqtt #25513: [New device support]: _TYST12_54rrtbbg

emimor123, 2025-01-04:
### Link

https://es.aliexpress.com/item/1005006124802072.html?spm=a2g0o.order_list.order_list_main.17.7f3e194dhIDxjy&gatewayAdapt=glo2esp

### Database entry

{"id":34,"type":"EndDevice","ieeeAddr":"0x44e2f8fffeb224a5","nwkAddr":40273,"manufId":0,"manufName":"_TYST12_54rrtbbg","powerSource":"Battery","modelId":"TY0A01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"^\u0000\f/\u0013_\u0000\f/\u0012","65506":28,"appVersion":88,"modelId":"TY0A01","manufacturerName":"_TYST12_54rrtbbg","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x00124b0009d6cafe","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":88,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736001808446}

### Zigbee2MQTT version

2.0.0_1

### Comments

I try to create an external converter for this electronic lock but no status is seen and no action can be performed.

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const getMsg = {
    cluster: 'genBasic',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        return {
            msgData: msg,
        };
    },
};

const tuyaLock = {
    fingerprint: [{modelID: 'TY0A01', manufacturerName: '_TYST12_54rrtbbg'}],
    model: 'TY0A01',
    vendor: 'Tuya',
    description: 'Tuya based lock with various unlock methods',
    fromZigbee: [tuya.fz.datapoints, getMsg],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.lock(),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
        e.text('alarm', ea.STATE).withDescription('Alarm'),
        e.binary('open_close', ea.STATE_SET, 'ON', 'OFF').withDescription('Open or Close'),
    ],
    meta: {
        tuyaDatapoints: [
            [8, 'open_close', tuya.valueConverter.trueFalse],
            [9, 'alarm', tuya.valueConverter.trueFalse],
            [10, 'battery', tuya.valueConverter.batteryPercentage],
        ],
    },
};

module.exports = tuyaLock;
```

### What does/doesn't work with the external definition?

Only the value of "Linkquality" can be seen, the rest of the values ​​appear unavailable and all actions give an error: " z2m: Publish 'set' 'open_close' to '0x44e2f8fffeb224a5' failed: 'TypeError: Cannot read properties of undefined (reading 'to')' "

[image]

closed 2025-03-13 by github-actions[bot]

OlegVekhov, 2025-09-10:
Same lock here... Have you got some success? @emimor123

emimor123, 2025-09-14:
I installed "Xtend Tuya" from HACS "https://github.com/azerty9971/xtend_tuya." Several lock controls and sensors appeared that weren't there before (you have to enable them; they're disabled by default). I haven't gotten it working yet, but I think it's a step forward. @OlegVekhov try it.
I'm adding a log in case anyone dares...
@Koenkk can something be done with this information? Sorry for bothering you.
Thanks in advance.

"documentation": "https://www.home-assistant.io/integrations/tuya",
    "integration_type": "hub",
    "iot_class": "cloud_push",
    "loggers": [
      "tuya_sharing"
    ],
    "requirements": [
      "tuya-device-sharing-sdk==0.2.1"
    ],
    "is_built_in": true,
    "overwrites_built_in": false
  },
  "setup_times": {
    "null": {
      "setup": 1.907299999004408e-05
    },
    "39ea3facb9b8a317f063f125927cebad": {
      "wait_import_platforms": -2.697615543000012,
      "wait_base_component": -0.0007118850000011889,
      "config_entry_setup": 6.088845086999996
    }
  },
  "data": {
    "endpoint": "https://apigw.tuyaeu.com",
    "terminal_id": "1751217761381d0UZp1",
    "mqtt_connected": true,
    "disabled_by": null,
    "disabled_polling": false,
    "id": "bf554b842899937724ves0",
    "name": "SmartLock-zigbee",
    "category": "ms",
    "product_id": "54rrtbbg",
    "product_name": "SmartLock-zigbee",
    "online": true,
    "sub": true,
    "time_zone": "+01:00",
    "active_time": "2025-02-23T16:28:47+00:00",
    "create_time": "2025-02-23T16:28:47+00:00",
    "update_time": "2025-09-14T11:24:04+00:00",
    "function": {
      "doorbell_volume": {
        "type": "Enum",
        "value": {
          "range": [
            "low",
            "high"
          ]
        }
      },
      "key_tone": {
        "type": "Enum",
        "value": {
          "range": [
            "0",
            "1",
            "2",
            "3",
            "4",
            "5",
            "6",
            "7",
            "8",
            "9"
          ]
        }
[... 275 lines cut ...]
          "disabled_by": null,
          "entity_category": null,
          "device_class": null,
          "original_device_class": null,
          "icon": null,
          "original_icon": null,
          "unit_of_measurement": null,
          "state": {
            "entity_id": "switch.smartlock_zigbee_lock_assistant",
            "state": "off",
            "attributes": {
              "friendly_name": "SmartLock-zigbee Lock assistant"
            },
            "last_changed": "2025-09-14T18:36:36.363673+00:00",
            "last_reported": "2025-09-14T18:36:36.363673+00:00",
            "last_updated": "2025-09-14T18:36:36.363673+00:00"
          }
        }
      ]
    },
    "set_up": true,
    "support_local": true
  },
  "issues": []
}