Koenkk/zigbee2mqtt #25514: Fail to start service after update to the 2.0.0-1 Version

1NoOneKnowsMe1, 2025-01-04:
### What happened?

After updating to the new edition, the service fails to start.

### What did you expect to happen?

The service will operate.

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20240710

### Adapter

ESP32-D0WD-V3

### Setup

Mini PC

### Debug log

[zigbee2mqtt_log.log](https://github.com/user-attachments/files/18307703/zigbee2mqtt_log.log)
[image]

Brax84, 2025-01-04:
Same problem here, seems to be a boot loop problem at loading

[19:26:59] INFO: Preparing to start...
[19:27:00] INFO: Socat not enabled
[19:27:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[19:27:52] INFO: Preparing to start...
[19:27:53] INFO: Socat not enabled
[19:27:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[19:33:09] INFO: Preparing to start...
[19:33:12] INFO: Socat not enabled
[19:33:17] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[19:34:19] INFO: Preparing to start...
[19:34:20] INFO: Socat not enabled
[19:34:22] INFO: Starting Zigbee2MQTT...

Raspberry pi4 
Core 2025.1.0
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250103.0

kemotf, 2025-01-04:
I have a similar problem after migration to 2.0.0.
zigbe2mqtt Container restart evry minute, logs from docker shows error: error in configuration, frontend must be an object
After switching to 1.42.0 - evrything works fine.

GwinGit, 2025-01-04:
These seem to be three unrelated errors, with the similarity that zigbee2mqtt is restarted after each error.

Regarding @1NoOneKnowsMe1's Problem:
It's now required to set the `adapter` type under serial, see also #24364 .

So the `serial` portion of your config should be
```yaml
serial:
  port: tcp://10.0.0.14:6638
  adapter: zstack
  baudrate: 115200
```

1NoOneKnowsMe1, 2025-01-04:
Thank you, that resolved the issue.

1NoOneKnowsMe1, 2025-01-05:
after several hours crash again
[lgs2-zigbee.log](https://github.com/user-attachments/files/18309171/lgs2-zigbee.log)

Koenkk, 2025-01-05:
> [2025-01-04 19:21:54] [31merror[39m: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.

You have to specific the adapter and port, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 on how to fix this.

closed 2025-01-05 by Koenkk

1NoOneKnowsMe1, 2025-01-05:
@Koenkk 
Hi, thank you for your quick and professional reply. I read the instructions and tried to configure it as shown in the attached screenshot, but the service still isn't starting.
i am using UZG-01

[image]  
[Log3.log](https://github.com/user-attachments/files/18310614/Log3.log)

1NoOneKnowsMe1, 2025-01-05:
Thank you for your help, @Koenkk . I've read the manual and configured it as instructed. The adapter successfully connected and read the devices, but it crashes at some point.
[image]
[Zigbee Crash.log](https://github.com/user-attachments/files/18311253/Zigbee.Crash.log)