Koenkk/zigbee2mqtt #25515: Philips Hue motion sensor missing Illuminance condition after 2.0.0 upgrade

Larsvb0, 2025-01-04:
### What happened?

Philips Hue motion sensor missing Illuminance condition after 2.0.0 upgrade

### What did you expect to happen?

TO use the Illuminance condition

### How to reproduce it (minimal and precise)

[image]
[image]

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

 7.4.2 [GA]

### Adapter

EZSP

### Setup

HA Supervisor in Hyper-V

m0wlheld, 2025-01-04:
Same reason as #25489: The entity id (not the display name) changed from illuminance_lux to illuminance.

fabiomsouto, 2025-01-04:
Same here. The entity ID did not change, I can't find it anywhere.

m0wlheld, 2025-01-04:
No, the former entity id ending with _illuminance_lux has been removed in favour of a new entity ending in _illuminance only. 

You should see this new entity at the device you, but maybe you have to activate it in Home Assistant. 

However, the name change is documented in the release notes.

Larsvb0, 2025-01-05:
as per https://github.com/Koenkk/zigbee2mqtt/issues/25488 I renamed the device in MQTT and now it appears:
[image]
[image]

closed 2025-01-05 by Larsvb0

askmurphy, 2025-03-19:
I have the same pir's, this trick to solvevit doesn't work here...