Koenkk/zigbee2mqtt #25516: Migrated to 2.0.0, failed to start, exited with code 139 - from addon

andreabressancern, 2025-01-04:
### What happened?

**I'm using HA supervised installed as docker on raspberry pi 4 running on bookworm. 
I have zigbee2mqtt installed as addon but after updating to 2.0.0 the container exit with with code 139. Log in the corresponding box ** 

** I cannot roll back to the previous version (don't find the backup for it, maybe skipped).
I've done minimal changes in the configuration (as mentioned in the release note. Any help is welcome. Hereafter the config I'm using:**

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
  restartdelay: 1
  initialdelay: 1
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: ************
  password: **********
  version: 5
  include_device_information: true
serial:
  port: /dev/ttyACM0
  adapter: ember
  external_converters: []
  devices: devices.yaml
  groups: groups.yaml
  homeassistant: true
  permit_join: false
  advanced:
    homeassistant_legacy_entity_attributes: false
    homeassistant_legacy_triggers: false
    legacy_api: false
    legacy_availability_payload: false
    log_level: debug
    pan_id: 6754
    channel: 15
    network_key:
      - 1
      - 3
      - 5
      - 7
      - 9
      - 11
      - 13
      - 15
      - 0
      - 2
      - 4
      - 6
      - 8
      - 10
      - 12
      - 13
    availability_blocklist: []
    availability_passlist: []
  device_options:
    legacy: false
  blocklist: []
  passlist: []
  queue: {}
  frontend:
    port: 8099
  experimental:
    new_api: true
  availability: false
  ota:
    update_check_interval: 10
  disable_automatic_update_check: false
zigbee_herdsman_debug: false
advanced:
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  last_seen: disable
  elapsed: false
  report: true
  legacy_api: true
  output: json

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20250103.0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus v2

### Setup

add-on on HA

### Debug log

docker logs addon_45df7312_zigbee2mqtt
[18:02:45] INFO: Preparing to start...
[18:02:45] INFO: Socat not enabled
[18:02:47] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 18:02:55] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 18:02:55] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 18:02:55] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 18:02:55] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20230803083122","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230803083122-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-01-04 18:02:55] info: 	zh:ember: Using default stack config.
[2025-01-04 18:02:55] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-04 18:02:55] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-04 18:02:55] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-04 18:02:55] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.

alexls74, 2025-01-05:
The same with Sonoff P Dongle

[10:16:20] INFO: Preparing to start...
[10:16:20] INFO: Socat not enabled
[10:16:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 10:16:28] info: z2m: Logging to console, file (filename: log.log)
[2025-01-05 10:16:28] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 10:16:28] info: z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 10:16:29] info: zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"e2f92ea33a1aec1189d274e5f01c6278","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_e2f92ea33a1aec1189d274e5f01c6278-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-05 10:16:29] info: zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}

andreabressancern, 2025-01-07:
Solved by 2.0.0-2. Can be closed.
Thank you

closed 2025-01-07 by Koenkk