Koenkk/zigbee2mqtt #25517: Error in Proxmox

aperezva, 2025-01-04:
### What happened?

Bare-metal installations (no Docker/Home Assistant add-on)
For Git based installations: removed configuration.yaml from repo, this will result in the following error when updating:
error: Your local changes to the following files would be overwritten by merge:
  data/configuration.yaml
To fix this, update by executing:
mv data/configuration.yaml data/configuration.yaml.bak
./update.sh # (update as you usually do)
mv data/configuration.yaml.bak data/configuration.yaml

I have tried to do this intructions 3 times, and it´s not woirking.

### What did you expect to happen?

Work as usual to update version in proxmox

### How to reproduce it (minimal and precise)

trying again update and trying again the instructions and the same error

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

 20221226

### Adapter

Slzb-06

### Setup

Proxmox

smarteshome2020, 2025-01-04:
It look that i have the same Problem in Proxmox. The update is aborting.

root@zigbee2mqtt:/opt/zigbee2mqtt# ./update.sh 
Checking Zigbee2MQTT status...
Stopping Zigbee2MQTT...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating a353acc8..3c46cfb2
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.example.yaml
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting
root@zigbee2mqtt:/opt/zigbee2mqtt#

dsiebel, 2025-01-04:
The issue is that the whole `data/` directory has been added to gitignore (finally!). This causes the conflict with your local changes.
Copy the file to a safe location, `git restore` the config (should be deleted), run the update, recover the config from the safe location.
There is a related discussion on the Proxmox community-scripts repo.

smarteshome2020, 2025-01-04:
> The issue is that the whole `data/` directory has been added to gitignore (finally!). This causes the conflict with your local changes. Copy the file to a safe location, `git restore` the config [...]

git restore /opt/zigbee2mqtt/data/ din't delete the configuration.yaml. Should i delete it manuell and update without any configuration.yaml?

KrX3D, 2025-01-04:
i had the same problem i solved it by:

git stash
git stash drop

after that it worked when i removed data-backup

i tried it than again and got some errors so i did

npm install
npm ci
npm install -g pnpm

and then the update.sh worked again

smarteshome2020, 2025-01-04:
> npm install -g pnpm

Thanks, afterthe npm commands it looks good :-)

> root@zigbee2mqtt:~# npm install -g pnpm

changed 1 package in 2s

1 package is looking for funding
  run `npm fund` for details
root@zigbee2mqtt:~# npm fund
root

root@zigbee2mqtt:~# cd /opt/zigbee2mqtt/
root@zigbee2mqtt:/opt/zigbee2mqtt# ./update.sh 
Checking Zigbee2MQTT status...
Stopping Zigbee2MQTT...
Updating...
Already up to date.
Installing dependencies...
Lockfile is up to date, resolution step is skipped
Packages: +368
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Progress: resolved 368, reused 0, downloaded 368, added 368, done
node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram: Running install script, done in 2.7s
node_modules/.pnpm/sd-notify@2.8.0/node_modules/sd-notify: Running install script, failed in 921ms (skipped as optional)
node_modules/.pnpm/esbuild@0.21.5/node_modules/esbuild: Running postinstall script, done in 121ms
node_modules/.pnpm/@serialport+bindings-cpp@12.0.1/node_modules/@serialport/bindings-cpp: Running install script, done in 222ms

dependencies:
+ ajv 8.17.1
+ bind-decorator 1.0.11
+ debounce 2.2.0
+ express-static-gzip 2.2.0
+ fast-deep-equal 3.1.3
+ finalhandler 1.3.1
+ git-last-commit 1.0.1
+ humanize-duration 3.32.1
+ js-yaml 4.1.0
+ json-stable-stringify-without-jsonify 1.0.1
+ jszip 3.10.1
+ moment 2.30.1
+ mqtt 5.10.3
+ object-assign-deep 0.4.0
+ rimraf 6.0.1
+ semver 7.6.3
+ source-map-support 0.5.21
+ throttleit 2.1.0
+ winston 3.17.0
+ winston-syslog 2.7.1
+ winston-transport 4.9.0
+ ws 8.18.0
+ zigbee-herdsman 3.2.1
+ zigbee-herdsman-converters 21.12.0
+ zigbee2mqtt-frontend 0.9.4

optionalDependencies:
+ sd-notify 2.8.0

devDependencies:
+ @eslint/core 0.9.1
+ @eslint/js 9.17.0
+ @ianvs/prettier-plugin-sort-imports 4.4.0
+ @types/eslint__js 8.42.3
+ @types/finalhandler 1.2.3
+ @types/humanize-duration 3.27.4
+ @types/js-yaml 4.0.9
+ @types/node 22.10.2
+ @types/object-assign-deep 0.4.3
+ @types/readable-stream 4.0.18
+ @types/sd-notify 2.8.2
+ @types/serve-static 1.15.7
+ @types/ws 8.5.13
+ @vitest/coverage-v8 2.1.8
+ eslint 9.17.0
+ eslint-config-prettier 9.1.0
+ prettier 3.4.2
+ tmp 0.2.3
+ typescript 5.7.2
+ typescript-eslint 8.18.2
+ vitest 2.1.8

Done in 10.9s
Building...

> zigbee2mqtt@2.0.0-dev build /opt/zigbee2mqtt
> tsc && node index.js writehash

Starting Zigbee2MQTT...
Done!
root@zigbee2mqtt:/opt/zigbee2mqtt#

aperezva, 2025-01-04:
i´m a little confuse,

Please could you put together all the process I have to follow?.

Thanks

Koenkk, 2025-01-05:
I updated https://github.com/Koenkk/zigbee2mqtt/discussions/24198, search for `Bare-metal installations (no Docker/Home Assistant add-on)`, hopefully this helps.

AGitDev, 2025-01-07:
> I updated #24198, search for `Bare-metal installations (no Docker/Home Assistant add-on)`, hopefully this helps.

Would it be possible to update your instructions to include the commands to cover the subsequent issues after following the [bare-metal installation instructions](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)?:  
```
ERROR: Backup directory exists. May be previous restoring was failed?
Save 'data-backup' and 'data' dirs to safe location to make possibility to restore config later.
Manually delete 'data-backup' dir and try again.
```
then the subsequent [npm errors](https://github.com/Koenkk/zigbee2mqtt/issues/25517#issuecomment-2571395097).

deflao, 2025-01-11:
Hello,
I have the same problem so here is what I did
I ran ./update.sh, then problem occurs.
so I do git stash
            git stash drop
then npm install -g pnpm
then git pull --no-rebase
then ./update.sh again. this time it went through with the update. 
But now I can't access zigbee2mqtt user interface so I did pnpm run clean. Still can't access it and come to find out the file configuration.yaml disappear from /data folder.
Any idea how I can restore my configuration.yaml file. Thank you.

AGitDev, 2025-01-14:
> Hello, I have the same problem so here is what I did I ran ./update.sh, then problem occurs. so I do git stash git stash drop then npm install -g pnpm then git pull --no-rebase then ./update.sh [...]

I think you needed to backup the configuration.yaml before upgrading.

zupancicmarko, 2025-02-09:
This worked perfectly for me with zigbee2mqtt running in Proxmox using tteck's script (debian).

`sudo systemctl stop zigbee2mqtt
apt update
apt upgrade
git checkout data/configuration.example.yaml
git checkout package-lock.json
mv data/configuration.yaml data/configuration.yaml.bak
git stash
git stash drop
npm install
npm audit fix
npm ci
npm install -g pnpm
git pull --no-rebase
./update.sh
mv data/configuration.yaml.bak data/configuration.yaml
sudo systemctl start zigbee2mqtt`

AGitDev, 2025-02-15:
> This worked perfectly for me with zigbee2mqtt running in Proxmox using tteck's script (debian).
>
> ```
> sudo systemctl stop zigbee2mqtt
> apt update
> apt upgrade
> git checkout [...]

Thanks. That solved it for me.

Schmakus, 2025-03-06:
Hi,
I have the following error after ./update.sh:

`root@zigbee2mqtt:/opt/zigbee2mqtt# ./update.sh
Checking Zigbee2MQTT status...
Updating...
Already up to date.
Installing dependencies...
 ERROR  Failed to switch pnpm to v10.4.1. Looks like pnpm CLI is missing at "/root/.local/share/pnpm/.tools/pnpm/10.4.1/bin" or is incorrect
spawnSync /root/.local/share/pnpm/.tools/pnpm/10.4.1/bin/pnpm ENOENT
Building...
 ERROR  Failed to switch pnpm to v10.4.1. Looks like pnpm CLI is missing at "/root/.local/share/pnpm/.tools/pnpm/10.4.1/bin" or is incorrect
spawnSync /root/.local/share/pnpm/.tools/pnpm/10.4.1/bin/pnpm ENOENT
Done!
root@zigbee2mqtt:/opt/zigbee2mqtt# 
`

> This worked perfectly for me with zigbee2mqtt running in Proxmox using tteck's script (debian).
>
> ```
> sudo systemctl stop zigbee2mqtt
> apt update
> apt upgrade
> git checkout [...]

thargy, 2025-05-02:
@Schmakus I'm also getting this same error now. I run 4 z2m instances, one local in HA, 2 in proxmox, 1 one Rasberry Pi. Since the update to 2, I have endless nightmares updating the 3 instance not in HA.

closed 2025-07-09 by github-actions[bot]