Koenkk/zigbee2mqtt #25520: V2 Upgrade / Aqara E1 switches (aka WXKG16LM and WXKG17LM) lost "action" entities (e is undefined)

musapinar, 2025-01-04:
### What happened?

Following a V2 upgrade, Aqara E1 switches no longer work.
The sensor's "_action" suffixes no longer exist.
[image]
[image]
[image]

### What did you expect to happen?

Being able to trigger automations binded on button clicks (actions)

### How to reproduce it (minimal and precise)

- Install z2m v1
- Pair Aqara E1 switches
- Create an automation to handle a button click
- Verify the automation works
- Upgrade z2m to v2
- The automation no longer works

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

20240710

### Setup

z2m addon on haos (oracle vm)

lukas947, 2025-01-04:
I have the same problem with Philips hue dimmers, so it's affecting at least two hw platforms.

Honzik1234, 2025-01-04:
SONOFF SNZB-01P Zigbee Wireless Switch same problem.

Rainz3, 2025-01-04:
1 Sonoff SNZB-01 and 2 Aqara WXKG11LM action has stopped showing in HA. I can still see the action in the exposes section and it respons to presses. Everything else is showing in HA.

apiontek, 2025-01-04:
I'm also experiencing this and I would rather fix it than roll back, but I'm struggling to find how to fix it. I'm also struggling to understand why switch actions would be disabled by default when they seem to be the primary way one would want to use the hardware...

Rainz3, 2025-01-04:
> I'm also experiencing this and I would rather fix it than roll back, but I'm struggling to find how to fix it. I'm also struggling to understand why switch actions would be disabled by default when they seem to be the primary way one would want to use the hardware...
> 
> Edit: I see we are now supposed to use some sort of [device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) thing. Something new to spend some time this weekend figuring out I suppose!

I used some instructions here https://community.home-assistant.io/t/aqara-buttons-all-in-one-zigbee2mqtt/268717/44

`Create a new empty automation.
In the When section, click on Add trigger and choose Device. Select your device and the desired trigger (e.g., “single_left” action), then set the Trigger ID (e.g., single_left) by clicking Edit ID in the options menu (three dots). If the action you’re looking for doesn’t appear, perform the action on your device, and it will show up in the list (you may also need to refresh the page in the browser).
Repeat step 2 for each trigger type you need (double, hold, etc…)
1
1
1225×846 38 KB
In the When section, click on Add building blocks and choose If-then.
In the If section click Add condition and choose Triggered by (from Other conditions menu). Next, select the trigger ID you want, such as single_left.
In the Then section, click Add action select the desired action, e.g. toggle entity.
image
image
1225×846 41.2 KB
Repeat step 4 for each trigger ID and save your changes.
Enjoy!`

So far it is working. I was using a blueprint that worked great from that thread. Just requires me redoing every action for the 3 buttons.

musapinar, 2025-01-04:
@Rainz3 Good catch. Thanks. It worked for me. I wish they explained why they did so.

For anyone wondering : 

Example before : 
```
alias: Interrupteur pour spots cuisine
description: ""
triggers:
  - entity_id: sensor.interrupteur_cuisine_action
    to: single
    trigger: state
actions:
  - data: {}
    action: script.cuisine_toggle_lights
mode: single
```

Example after : 
```
alias: Interrupteur pour spots cuisine
description: ""
triggers:
  - trigger: device
    domain: mqtt
    device_id: 171199ef83df18f064f8af155c727df6
    type: action
    subtype: single
actions:
  - data: {}
    action: script.cuisine_toggle_lights
mode: single
```

You can find the device id in the url here : 
[image]
[image]

You can identify your automations that require a fix searching for "_action" in your yaml here : 
[image]

Honzik1234, 2025-01-04:
SONOFF SNZB-01P Zigbee Wireless Switch - the edit works
alias: Linka tlačítko
description: ""
triggers:
  - trigger: device
    domain: mqtt
    device_id: 629b3fd5632d08d851acd1df31d75da8
    type: action
    subtype: single
actions:
  - action: switch.toggle
    metadata: {}
    data: {}
    target:
      entity_id: switch.sonoff_10014ed5c8
mode: single

closed 2025-01-04 by musapinar