Koenkk/zigbee2mqtt #25522: HA Zigbee Light Group Receives "On" Signal But Not "Off" Signal

ClintElliotMalcolm, 2025-01-04:
### What happened?

I have zigbee groups throughout my house consisting of inovelli switches and hue lights. The switch is bound to the group for brightness and on/off. This has worked flawlessly for awhile now. I updated to 2.0.0 yesterday and believe I did everything correctly. I had no legacy attributes on. However, since then I have had this issue.

When I press a light switch on, the lights come on and everything is fine. In HA the group "device" also shows that it turned on. When I turn the light switch off everything seems to work, all the lights turn off and the inovelli led strip behaves properly. However, the zigbee group "device" still shows it as "on". This causes problems because I use adaptive lighting and it will turn the lights on again.

If I turn the zigbee group "device" off via HA everything works as expected. The lights turn off and HA understands that it turned off.

Strangely enough the individual light and light switch devices in HA behave as expected. HA knows that all lights and light switches within the group are off but it doesn't believe that the zigbee group is off.

### What did you expect to happen?

When a zigbee group of light/light switches are on and I press off on the light switch bound to that group; not only should the lights turn off and the switch go to the proper state, but the HA group should move to the right state. 

### How to reproduce it (minimal and precise)

* Turn on smart light switch (Inovelli) bound to zigbee group
* See lights on and group lights on in reality and in HA; can see transition from off to on
* Turn off smart light switch
* See lights off but HA group state remains "on"

### Zigbee2MQTT version

 2.0.0

### Adapter firmware version

20240710

### Adapter

Tubeszb-Cc2652P7-Poe-2023

### Setup

Add-on For Home Assistant OS on Raspberry Pi 5

### Debug log

zigbee group that I reproduced on is zigbee2mqtt/upper_guest_room_lights 
Shortly after a zigbee2mqtt restart I:

1. Pressed on the Inovelli switch
2. Confirmed lights were on physically
3. Confirmed lights and HA zigbee group device on in HA
4. Pressed off the Inovelli switch
5. Confirmed lights were off physically
6. Confirmed HA zigbee group device was still on in HA
7. Turned the HA zigbee group device off in HA to have it so that the lights wouldn't turn on again later due to Adaptive Lighting

[log.log](https://github.com/user-attachments/files/18307948/log.log)

ClintElliotMalcolm, 2025-01-06:
If anyone can please take a look at this I'd really appreciate it. The lights feel like they are gaslighting us. I'd much rather roll forward than rollback.

ClintElliotMalcolm, 2025-01-07:
Just to further clarify. This was working until 2.0.0. I have the binding right. This relief on no legacy features as far as I can tell. It just used raw zigbee bindings and the zigbee group. Everything works as intended except for the group device. I've attached pictures to demonstrate.

Binding set
[image]

Devices in group are off on zigbee2mqtt 
[image]

Turned on the switch from this screen. Lights turn on!
[image]

I flipped the switch from the previous screen again and all lights are off. As shown in a prior picture. However, the light group still shows as "on" in HA. 
[image]

The device log shows it turning on but nothing about it being off or any issues.
[image]

If I were to show you the individual light and light switch devices all would be off.

For right now I'll take the advice of the comment above and make a dumb automation to mitigate the issue. But the issue still remains.

ClintElliotMalcolm, 2025-01-07:
Issue discussion on inovelli community page: https://community.inovelli.com/t/new-zigbee2mqtt-2-0-update-breaks-group-control/18492

closed 2025-01-07 by ClintElliotMalcolm

SisyphusMD, 2025-01-21:
Sorry to resurrect this, but I’m not sure the solution fully addresses all use cases.

I had the same issue with hue/inovelli/adaptive lighting that initially led me here, and the above fix seems to assume that the inovelli switches are only controlling smart bulbs.

However, the inovelli switches can also control actual circuits. I have 2 light circuits in my kitchen, for example, that are controlled by 2 different inovelli blue switches. I can’t just remove the switches from the group because they need to be controlled by the group. This is why the prior behavior seemed to make so much logical sense, one of the endpoints for controlling, and one of the endpoints for being controlled.

I tried the above solution from this thread, leaving an empty ‘kitchen’ group to which I bound both switches’ endpoints 1 and 2. And the switches did continue to work to control both circuits, but that group was no longer controllable though home assistant, and it didn’t report its status at all, which I believe is due to being an empty group.

rccoleman, 2025-01-22:
I think I'm struggling with the same thing, where the previous direction to remove the actual switches from the group won't work because two switches in a three-way configuration also control a load.  I originally had a binding for the source endpoint 2 to the group that contains that switch along with several other switches and I just added a binding for endpoint 1 to the same group.  I can't tell if that's really helping yet, but the state seems to be correct if I use the physical switch or control the group from HA.

rccoleman, 2025-01-24:
This inconsistent state issue is happening to me pretty regularly now and I'm wondering if it's a race or ordering issue between device state reporting and evaluation of the group state.  Here's an example of where I think it went wrong, with some of my own debug inserted.  Here's the full log:

```
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29,53]
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29,53]
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,116,150,1,1,0,54,0,77,162,96,0,0,7,24,76,10,0,0,16,0,116,150,29] - 53
[2025-01-23 19:47:40] debug:    zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":38516,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":54,"securityuse":0,"timestamp":6333005,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,76,10,0,0,16,0]}}
[2025-01-23 19:47:40] debug:    zh:controller: Received payload: clusterID=6, address=38516, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=54, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":76,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-01-23 19:47:40] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-01-23 19:47:40] debug:    z2m: Received Zigbee message from 'Kit Sink', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
[2025-01-23 19:47:40] debug:    z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Sink is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Morning Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: Garage Workshop Lights, Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: DNH (Bar), Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Garage Door Open Light Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group, Device: Kit (DR), Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit (DR) is ON
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group New, Device: Kit Desk, Current State: OFF
[2025-01-23 19:47:40] debug:    z2m: Group: Kit All Lights Group New, Device: Kit Pendants, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Pendants is ON
[2025-01-23 19:47:40] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqttp/Kit Sink', payload '{"action":null,"activeEnergyReports":0,"activePowerReports":0,"autoTimerOff":null,"auxSwitchUniqueScenes":null,"bindingOffToOnSyncLevel":null,"brightness":254,"brightnessLevelForDoubleTapDown":null,"brightnessLevelForDoubleTapUp":null,"buttonDelay":"0ms","defaultLed1ColorWhenOff":null,"defaultLed1ColorWhenOn":null,"defaultLed1IntensityWhenOff":null,"defaultLed1IntensityWhenOn":null,"defaultLed2ColorWhenOff":null,"defaultLed2ColorWhenOn":null,"defaultLed2IntensityWhenOff":null,"defaultLed2IntensityWhenOn":null,"defaultLed3ColorWhenOff":null,"defaultLed3ColorWhenOn":null,"defaultLed3IntensityWhenOff":null,"defaultLed3IntensityWhenOn":null,"defaultLed4ColorWhenOff":null,"defaultLed4ColorWhenOn":null,"defaultLed4IntensityWhenOff":null,"defaultLed4IntensityWhenOn":null,"defaultLed5ColorWhenOff":null,"defaultLed5ColorWhenOn":null,"defaultLed5IntensityWhenOff":null,"defaultLed5IntensityWhenOn":null,"defaultLed6ColorWhenOff":null,"defaultLed6ColorWhenOn":null,"defaultLed6IntensityWhenOff":null,"defaultLed6IntensityWhenOn":null,"defaultLed7ColorWhenOff":null,"defaultLed7ColorWhenOn":null,"defaultLed7IntensityWhenOff":null,"defaultLed7IntensityWhenOn":null,"defaultLevelLocal":null,"defaultLevelRemote":null,"deviceBindNumber":null,"dimmingSpeedDownLocal":null,"dimmingSpeedDownRemote":null,"dimmingSpeedUpLocal":null,"dimmingSpeedUpRemote":0,"doubleTapClearNotifications":null,"doubleTapDownToParam56":null,"doubleTapUpToParam55":null,"energy":8.01,"fanControlMode":null,"fanLedLevelType":null,"firmwareUpdateInProgressIndicator":null,"highLevelForFanControlMode":null,"higherOutputInNonNeutral":null,"identify":null,"individual_led_effect":{"color":""},"internalTemperature":null,"invertSwitch":null,"ledBarScaling":null,"ledColorForFanControlMode":null,"ledColorWhenOff":null,"ledColorWhenOn":null,"ledIntensityWhenOff":null,"ledIntensityWhenOn":null,"led_effect":null,"linkquality":54,"loadLevelIndicatorTimeout":null,"localProtection":null,"lowLevelForFanControlMode":null,"maximumLevel":null,"mediumLevelForFanControlMode":null,"minimumLevel":null,"onOffLedMode":null,"outputMode":"On/Off","overheat":null,"periodicPowerAndEnergyReports":null,"power":0,"powerType":null,"quickStartLevel":null,"quickStartTime":null,"rampRateOffToOnLocal":null,"rampRateOffToOnRemote":null,"rampRateOnToOffLocal":null,"rampRateOnToO [... cut ...]
[2025-01-23 19:47:40] debug:    z2m: Group: Kit Light Group, Device: Kit Sink, Current State: ON
[2025-01-23 19:47:40] debug:    z2m: Kit Sink is ON
```
And later we see that the new state of `Kit Sink` of `OFF` is finally reported to MQTT with
`"state":"OFF",`

Z2M never reevaluates the state of that group, even after a restart of Z2M, presumably until there's another state change that would affect the members in that group.  Sometimes the state is properly reflected and the group state changes and sometimes it doesn't, leading me to think that there's a race condition.