Koenkk/zigbee2mqtt #25523: HIVE SLR2(x) climate entity

craigsbits, 2025-01-04:
### What happened?

Climate heat endpoint  'occupied_heating_setpoint' changed to standard climate 'temperature'

### What did you expect to happen?

not to change!

### How to reproduce it (minimal and precise)

N/A

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20211217

### Adapter

CC Stick 3.0 one as coordinator and another one as router

### Setup

NUC OS

### Debug log

N/A

tominozh, 2025-01-04:
Just to add a bit more info here as I am also facing this problem.

I went back to 1.4.x and got a climate entity screenshot with all attributes that were available before the update.

[image]

Now back on z2m 2.0.x, the climate entity is missing few attributes, namely the important for my case
 
- **_system_mode_** 
- **_occupied_heating_setpoint_** 
-  _**temperature_setpoint_hold_duration**_ 

 which I use to see if someone already boosted heating through Hive Thermostat so I can prevent changes from my automation.

[image]

How should this problem be corrected?

Fierys, 2025-01-05:
I have the same story with the thermostat [Moes TV01-ZB](https://www.zigbee2mqtt.io/devices/TV01-ZB.html).
**z2m 1.42.0-2
Attribute:**
```
hvac_modes:
  - "off"
  - heat
min_temp: 5
max_temp: 30
target_temp_step: 0.5
preset_modes:
  - none
  - auto
  - manual
  - holiday
current_temperature: 19.9
temperature: 23
preset_mode: manual
battery_low: false
boost_timeset_countdown: 0
child_lock: UNLOCK
comfort_temperature: 23
current_heating_setpoint: 23
eco_temperature: 20
error_status: 0
frost_protection: "OFF"
heating_stop: "OFF"
holiday_start_stop: 2021/01/01 01:01 | 2021/01/01 01:01
holiday_temperature: 17
linkquality: 33
local_temperature: 19.9
local_temperature_calibration: 0
online: "ON"
open_window: "OFF"
open_window_temperature: 18
preset: manual
schedule_friday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_monday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_saturday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_sunday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_thursday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_tuesday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
schedule_wednesday: 06:00/17 12:00/21 14:00/17 17:00/21 24:00/17
system_mode: heat
update:
  installed_version: 69
  latest_version: 69
  state: idle
update_available: false
working_day: mon_sun
friendly_name: Спальня, TRV 1
supported_features: 401
```
**z2m 2.0.0
Attribute:**
```
hvac_modes:
  - "off"
  - heat
min_temp: 5
max_temp: 30
target_temp_step: 0.5
preset_modes:
  - none
  - auto
  - manual
  - holiday
friendly_name: Спальня, TRV 1
supported_features: 401
current_temperature: 21
temperature: 23
preset_mode: manual
```
My automations are no longer working...

closed 2025-03-16 by github-actions[bot]