Koenkk/zigbee2mqtt #25525: Philips Hue TapDial entity for buttons not available

ruebezahl17, 2025-01-04:
### What happened?

After the update to 2.0 my Hue TapDial isn't working like expected: if I press a button, nothing happens. In Z2M I can see the button is pressed, but in HA not. Than I found there is the entity for the buttons no more available. The wheel is working. 

### Zigbee2MQTT version

2.0

### Adapter firmware version

v2.6.8.dev25

### Adapter

SLZB-06

### Setup

Add-in on HA running as VM on Synology

andreaseu, 2025-01-04:
I have the same problem with hue dimmer switches. 
This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings.

ruebezahl17, 2025-01-04:
How can I get it back? Without an entity for the buttons it is useless.

ruebezahl17, 2025-01-04:
There are all affected who use the action entity. In another issue I found the solution: [https://community.home-assistant.io/t/using-the-new-action-events-in-zigbee2mqtt-2-0/821709/12?u=vincentbattez](url)
Now it works like before.

ruebezahl17, 2025-01-04:
next update: I updated the blueprint (https://community.home-assistant.io/t/philips-hue-tap-dial-switch-zigbee2mqtt/461824) and it works now without the legacy action entity.

m0wlheld, 2025-01-04:
Looks like you are talking about Home Assistant or the Home Assistant Zigbee2MQTT add-on. Both are not subject of this repository.

ruebezahl17, 2025-01-05:
Yes, I'm talking about that. The search engine sends me here and here are more people with the same problem (and with the solution I found later).

m0wlheld, 2025-01-05:
> Yes, I'm talking about that. The search engine sends me here and here are more people with the same problem (and with the solution I found later).

The solution you mention in https://github.com/Koenkk/zigbee2mqtt/issues/25525#issuecomment-2571406614 (which is actually not a solution, but a way to keep the old way alive) is from the Home Assistant community forum, so "the search engine" must have sent you there, too.

I just demonstrates, that Zigbee2MQTT does better SEO than Home Assistant ;)

* [Official Zigbee2MQTT Home Assistant add-on](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues) Issue List
* [Official Home Assistant Community Forum](https://community.home-assistant.io)

ruebezahl17, 2025-01-05:
It depends on search words where it sends me ;-)
I also found out it was only a workaround and wrote the update above that it works now without the action entity. Fortunately the blueprint I use was updated to this new method. I red also how to manually change to this new one but it is not so easy to understand if you work only with the GUI in HA.

closed 2025-01-05 by Koenkk