Koenkk/zigbee2mqtt #25526: After update 2.0.0 missing the action button at agara xiaomi WXKG11LM mini switch

canedje, 2025-01-04:
### What happened?

After update to 2.0.0 Home assitant core is missing action sensor of agara xiaomi WXKG11LM mini switch.
It was there before.
The sensor is availabel in the Z2m UI
[image]
but not at Homeassistant:
[image]

### What did you expect to happen?

The sensor is available as usual and before

### How to reproduce it (minimal and precise)

It just happens

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZBDongle-E  emberZNet

### Setup

as a docker under unraid. And Homeassistant as doker under Unraid

ruebezahl17, 2025-01-04:
here is the solution (had the same problem with a Hue TapDial): [https://community.home-assistant.io/t/using-the-new-action-events-in-zigbee2mqtt-2-0/821709/12?u=vincentbattez](url)

canedje, 2025-01-04:
Thanks!!! that did the trick

closed 2025-01-04 by canedje

lasry1, 2025-01-05:
Thank you! That solved my issue too activating “Home Assistant experimental even entities” and “Home Assistant legacy action sensors” in zigbee2mqtt settings

jflefebvre06, 2025-01-05:
Same problem