Koenkk/zigbee2mqtt #25528: Tuya MTG235-ZB-RL leaves network right after successful interview

adriaandotcom, 2025-01-04:
### What happened?

I replaced my CONBEE II stick with the SLZB-06M. I upgraded z2m to 2.0.0. I'm re-pairing all my devices, and this type fails. One did succeed, but the other do not.

### What did you expect to happen?

It stayed on the network after joining.

### How to reproduce it (minimal and precise)

- Allow joining
- Pair Tuya MTG235-ZB-RL (press reset button 5 seconds)
- Watch the logs (it joins and leaves right after)

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M core v2.3.6, zigbee 20241127 (both EZSP & ember)

### Setup

Home Assistant OS on Intel NUC

### Debug log

info 2025-01-04 22:08:08z2m: Zigbee: allowing new devices to join.
info 2025-01-04 22:08:08zh:ember: [STACK STATUS] Network opened.
info 2025-01-04 22:08:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"3w024-1"}'
info 2025-01-04 22:08:23zh:controller: Interview for '***' started
info 2025-01-04 22:08:23z2m: Device '***' joined
info 2025-01-04 22:08:23z2m: Starting interview of '***'
info 2025-01-04 22:08:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"***","ieee_address":"***"},"type":"device_joined"}'
info 2025-01-04 22:08:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"***","ieee_address":"***","status":"started"},"type":"device_interview"}'
info 2025-01-04 22:08:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"***","ieee_address":"***"},"type":"device_announce"}'
warning 2025-01-04 22:08:25z2m: Device '***' left the network
info 2025-01-04 22:08:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"***","ieee_address":"***"},"type":"device_leave"}'
error 2025-01-04 22:08:28zh:controller: Interview failed for '*** with error 'Error: DatabaseEntry with ID '2' does not exist'

adriaandotcom, 2025-01-04:
I did a full reset of the system, now it seems to work.

adriaandotcom, 2025-01-05:
Re-paired my whole network without problems. Then one device I wanted to repair. That is causing the same error as above. Luckily the other devices are paired now. But I can't pair this one device anymore. It gives the same errors as in the description of this issue. I more debug info is needed, let me know.

nbrownus, 2025-01-06:
Having the exact same issue. This device was bound to another device, I wonder if despite a factory reset that the binding is still being reported when interviewed and since z2m doesn't know anything about the other device that it's triggering the problem?

adriaandotcom, 2025-01-07:
It's indeed when a device was connected before. A full restart of the system didn't help for me. When I (force) remove a device from the network, I'm unable to add it again.

adriaandotcom, 2025-01-08:
When re-pairing devices multiple times, I get them to pair. Enabling the <kbd>Permit join (All)</kbd> button **after** putting the device in pair mode seems to work more consistently.

It might be an issue in my network only, because I have 12 [Tuya MTG235-ZB-RL](https://www.zigbee2mqtt.io/devices/MTG235-ZB-RL.html) devices that send a payload every second. But I think it's an issue between zigbee2mqtt and the SLZB-06M coordinator.

kolopeter221, 2025-03-16:
coordinator  SLZB-06M.  
3 modules behaved the same as in the title.
They appeared and disappeared. it was impossible to complete the configuration.
Coordinator was set to channel 25. I changed the channel to 11 and the configuration of the first module was successful. But the rest of the modules were offline, so I restored channel 25 and searched for the first module again. With success. Later I added the remaining two without any combinations and problems

FriendlyFireDavid, 2025-04-17:
I'm reporting a similar problem.

I have a rather large network with 140 devices running for years now.
Yesterday, I wanted to install 3 new IKEA bulbs.
The first one paired normally and works.
Second pairs but then immediately leaves the network (sometimes with a notification, sometimes without).
Then I tried several new bulbs of the same type (new from the box), but all leave the network after pairing.

Funny thing is that after it pairs successfully, I can control it for a few seconds, but then it flashes and leaves the network.

I've tried to pair the 3rd Reality sensor after this, and I can pair and work normally.
I'll later try with other devices too (different from IKEA) to see if it's going to be any different.

**Zigbee2MQTT version**: 2.2.1 commit: unknown
**Coordinator**: Sonoff USB Dongle P (TI CC2652)
Coordinator type: zStack3x0
Coordinator revision: 20240710
**Setup**: Home Assistant OS on Lenovo ThinkCentre, updated to latest version

closed 2025-06-25 by github-actions[bot]